ghaerr/elks #8: elksnet fails to build

darkstar, 2015-03-04:
Trying to build elksnet results in the following:
```
root@kubuntu:/usr/src/elks/elksnet# make
make[1]: Entering directory '/usr/src/elks/elksnet/ktcp'
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o ktcp.o ktcp.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o slip.o slip.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o ip.o ip.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o icmp.o icmp.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o tcp.o tcp.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o tcp_cb.o tcp_cb.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o tcp_output.o tcp_output.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o timer.o timer.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o tcpdev.o tcpdev.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o netconf.o netconf.c
bcc -ansi -O -0 -D__KERNEL__ -I/usr/src/elks/elks/include -c -o vjhc.o vjhc.c
vjhc.c:117.19: error: need ')'
vjhc.c:119.1: error: need ';'
vjhc.c:119.1: error: need variable name
vjhc.c:117.3: error: need ';'
vjhc.c:117.21: error: ip_vjhc_compress already declared
vjhc.c:118.12: error: pkt already declared
vjhc.c:120.1: error: need variable name
vjhc.c:121.8: error: need ';'
vjhc.c:130.3: error: need variable name
vjhc.c:130.5: error: need ';'
```
...and hundreds of follow-up errors. Probably related to the funny-looking DEFUN macro (why is that needed anyway?)

darkstar, 2015-03-04:
Oh forgot to mention that I selected CSLIP for this. When deselecting CSLIP it builds

jbruchon, 2015-03-05:
I removed the ansi.h section of vjhc.h and fixed the DEFUN nonsense in vjhc.c; it compiles correctly now. Fixed in commit 1420df30300487fd95131efb964c212d0d7a653f

closed 2015-03-05 by jbruchon