ghaerr/elks #9: building floppy disks fails

darkstar, 2015-03-04:
Something to do with bash, maybe? If so, what shell is recommended to use?
```
root@kubuntu:/usr/src/elks/elkscmd# make full3
make[1]: Entering directory '/usr/src/elks/elkscmd'
/bin/sh: 1: [: missing ]
/bin/sh: 1: -e: not found
-e

 *** Do builds in dev86 and dev86/bootblocks first ***

Makefile:147: recipe for target 'boot_blocks' failed
make[1]: *** [boot_blocks] Error 1
make[1]: Leaving directory '/usr/src/elks/elkscmd'
Makefile:206: recipe for target 'full3' failed
make: *** [full3] Error 2
```
with verbose tracing:
```
root@kubuntu:/usr/src/elks/elkscmd# make --trace full3
Makefile:206: target 'full3' does not exist
if [ "0" != "0" ]; \
then echo -e "\n\n *** Only root can build disk images *** \n\n"; exit 1; fi
make _build_bootable_target \
        TARGET_FS=/usr/src/elks/elkscmd/full3 \
        TARGET_RFS=rfs \
        TARGET_BLKS=1440 \
        INSTALL_KTCP=yes
make[1]: Entering directory '/usr/src/elks/elkscmd'
Makefile:147: update target 'boot_blocks' due to: /usr/src/elks/dev86/bootblocks/minix.bin /usr/src/elks/dev86/bootblocks/minix_elks.bin
if [ ! -e /usr/src/elks/dev86/bootblocks/minix.bin || ! -e /usr/src/elks/dev86/bootblocks/minix_elks.bin ]; \
then echo -e "\n\n *** Do builds in dev86 and dev86/bootblocks first *** \n\n"; exit 1; fi
/bin/sh: 1: [: missing ]
/bin/sh: 1: -e: not found
-e

 *** Do builds in dev86 and dev86/bootblocks first ***

Makefile:147: recipe for target 'boot_blocks' failed
make[1]: *** [boot_blocks] Error 1
make[1]: Leaving directory '/usr/src/elks/elkscmd'
Makefile:206: recipe for target 'full3' failed
make: *** [full3] Error 2
```

Bootblocks from dev86 are there:
```
root@kubuntu:/usr/src/elks# ls -l /usr/src/elks/dev86/bootblocks/minix*.bin
-rw-r--r-- 1 root root 1024 Mär  4 12:56 /usr/src/elks/dev86/bootblocks/minix.bin
-rw-r--r-- 1 root root  281 Mär  4 12:56 /usr/src/elks/dev86/bootblocks/minix_elks.bin
-rw-r--r-- 1 root root 1024 Mär  4 12:56 /usr/src/elks/dev86/bootblocks/minixhd.bin

```

jbruchon, 2015-03-05:
Are you on Debian? What shell is used as /bin/sh on your system? Some systems do not use bash as the default shell. I do all my testing on a CRUX Linux 3.1 machine with /bin/sh -> bash.

darkstar, 2015-03-05:
I'm on (k)ubuntu 14.10. It appears that although my login shell is set to bash, /bin/sh points to ubuntu's own "dash" shell. And then gmake picks up /bin/sh insteat of my login shell, so it was using dash all along instead of bash. Manually removing the dash-link and pointing /bin/sh to /bin/bash has solved the issue.

jbruchon, 2015-03-05:
It would be better if we could fix the problem on dash instead of requiring /bin/sh to point to bash. It seems that dash doesn't support the [[ ... ]] construct and that this is the source of the problem. I've already committed a likely fix. Could you please pull the latest git and see if this error is gone?

darkstar, 2015-03-07:
Yes, now it works.

There's still another small issue with dash. Right at the start, when running ```./build.sh```, there's supposed to be a pause where the user has to press a key to continue. I only noticed that after switching /bin/sh to bash, because with dash, it starts building right away:
```
root@kubuntu:/usr/src/elks# ./build.sh
WARNING: Disk images can only be built logged in as 'root'

Preparing to build the ELKS kernel. This will invoke 'make menuconfig'
for you to configure the system. The defaults should be OK for many
systems, but you may want to review them.

Press a key to continue... ./build.sh: 9: read: Illegal option -n

sed '/\#\#\# Dependencies:/q' < Makefile > make.tmp
make[1]: Entering directory '/usr/src/elks/elks/Documentation'
sed '/\#\#\# Dependencies:/q' < Makefile > make.tmp
make[2]: Entering directory '/usr/src/elks/elks/Documentation/Images'
sed '/\#\#\# Dependencies:/q' < Makefile > make.tmp
...
```
looks like dash doesn't like "read -n". Not sure about how to fix this as there's no /bin/read or /usr/bin/read as fallback and dash's read is very limited in what it can do..

jbruchon, 2015-03-07:
I've implemented a workaround that sleeps for 7 seconds (configurable by editing `DELAY=7` at the top of the `build.sh` script) when the $BASH_VERSION var is missing. It won't detect `bash` in 100% of scenarios (see http://stackoverflow.com/questions/3199893/howto-detect-bash-from-shell-script for more information) but it will work in most common cases.

Tested against BusyBox-flavored `ash` in `busybox-1.21.1` and confirmed the workaround triggers. Note that BusyBox supports some `bash`isms in their port of `ash` and it might be a good idea to use the `read -n 1 -s 2>/dev/null || sleep 7` construct instead so that `/bin/sh -> BusyBox` can take advantage of `read -n`, but for now we just need the initial help text to be up long enough to be readable.

Fixed in commit 32532edd5f48c33f069dfa2d9ba33e2dd888bf0a (build.sh: fix a bashism).

closed 2015-03-07 by jbruchon

mfld-fr, 2018-05-15:
Please ignore the reference from #133 : fake one from source excerpt.