ghaerr/elks #13: fail to compile elkscmd/ktcp/vjhc.c

lynus, 2015-12-11:
Here is the output:
#bcc -ansi -O -0 -D__KERNEL__ -I/home/lynus/elks/elks/include -c -o vjhc.o vjhc.c
00427                                           if ip_vjhc_compress.off=0
***** relocation impossible..............................................^
00438                                           if ip_vjhc_compress.off=0
***** relocation impossible..............................................^
00469                                           if ip_vjhc_compress.off=0
***** relocation impossible..............................................^

00003 errors
00000 warnings
../Make.defs:122: recipe for target 'vjhc.o' failed

lynus, 2015-12-11:
It seems that it is the compiler to blame. The error raise when I use your dev86 toolchain.
But I installed bcc via `apt-get install bcc', the build success.

jbruchon, 2015-12-12:
This was a problem with new `dev86` code I pulled in from upstream (see https://github.com/lkundrak/dev86/issues/11 for more info). I have reverted the commit that caused this error in my `dev86` tree as of commit 4e8476e579885ecbf5db5c704491bb3397448431 and tested against ELKS to ensure it works correctly.

closed 2015-12-12 by jbruchon