ghaerr/elks #16: full3 image crashes on boot

fortytwopar, 2016-01-20:
I was trying to compile elks and get networking running with the full3 image, but I was not able to crash because there was an error about a double free which crashed completely. I narrowed down the issue to elks/arch/i86/mm/malloc.c in commit 4786296011e318ca250ad096ce3ccd4ae01b21fe

I am not familiar enough with the internals of malloc to fix the problem myself, as some of the optimizations look like they could be helpful when specific conditions are met.

jbruchon, 2016-01-20:
Is this happening on real hardware or in an emulator?

fortytwopar, 2016-01-20:
It's happening on hardware, both a really old 286 and an also pretty old P3 laptop. I haven't been testing on an emulator.

lithoxs, 2016-02-18:
It is a bug in malloc.c inadvertently introduced by me 2 commits ago. I already set a pull request that fixes this problem.

jbruchon, 2016-02-18:
Fixed in commit 7ae683f0aefcbe8cf4fd75d69f6e7a8988e2d128

closed 2016-02-18 by jbruchon