ghaerr/elks #26: ROM version build fails

mfld-fr, 2016-05-21:
When selecting the ROM version (CONFIG_ROMCODE), the build fails:
```
gcc -E -traditional -I/mnt/data/home/mfld/advantech/elks/elks/include -DELKS_VERSION_CODE=0x00020000 -DUTS_RELEASE=\"0.2.0\" -D__KERNEL__ -o irqtab.s irqtab.S
as86 -0  -o irqtab.o irqtab.s
00399 0010           891E       0000                    mov stashed_ds,bx
***** unbound label.........................................^
00530 0083           8B1E       0000            	mov	bx,stashed_ds	! Recover the data segment
***** unbound label................................................^
00755 0147           8E1E       0000            	mov	ds,stashed_ds		! the org DS of kernel
***** unbound label................................................^
```

The reason is that GCC does not pre-process the `#define stashed_ds` directive, that stills in the .s file:
```
#asm
	.text
   #define stashed_ds       [0]
```

This causes AS86 to fail, as there is no `stashed_ds` label, and that word has not been substituted to `[0]`.

mfld-fr, 2016-05-22:
Thanks to #27, build with option CONFIG_ROMCODE succeeded to generate `setup` and `system` object files, but now the generation of the ROM image fails at the following step:
```
tools/mkbootloader  -c 0xe000 64 Image 0xe000 -a boot/setup 0xe000 -s boot/system 0xe060  
  boot/setup: ffffffe0h Bytes (- a.out) @e000
  Wrong a.out table?
  boot/system: 0h Bytes (strip) @e060
Bereichsfehler in Nr. 0 (offs < basis)!
Makefile:171 : la recette pour la cible « Image » a échouée
```

Both `setup` and `system` object files look correct when dumping with dumpobj86.

mfld-fr, 2016-05-26:
I have a patch that works on my 64 bits system, where I replaced `%lx` by `%x`, but I am not sure it would not cause regression on a 32 bits system. `uint32_t` and such are portable with fixed size, but what about the format string ?

lithoxs, 2016-06-01:
For the case of integer sizes with 32 and 64 bits systems using gcc and linux, see:

https://usrmisc.wordpress.com/2012/12/27/integer-sizes-in-c-on-32-bit-and-64-bit-linux/

So, for the program to work on 32 and 64 bit systems, all 32 bit integers should be "int" and use "%x", and all 64 bit integers should be "long long int" and use "%llx".

mfld-fr, 2016-06-02:
Thanks Juan, so I can commit the `mkbootloader`patch that replaces `%lx` by `%x`.

jbruchon, 2016-06-02:
The correct way to `printf()` C99 integer types is covered in [this Stack Overflow answer.](http://stackoverflow.com/a/3168298/1906641) In short, `#include <inttypes.h>` and use the appropriate macros from that header for the type you're working with.

Alternatively, you could cast to `int` for the `printf()` only. That's what I do sometimes. It looks better than `uint32_t r = 1000; printf("reading %" PRIu32 " bytes\n", r);`

commit d56ab8d4 (2016-06-04): Merge pull request #32 from mfld-fr/master

Fix #26 + FR keyboard tuning
commit eee36ba2 (2016-06-04): Fix #26: ROM image: build and setup data
closed 2016-06-04 by jbruchon