ghaerr/elks #39: XT IDE hdd detection results in system crash

sparky4, 2016-08-02:
My XT has an XTIDE card in it and when elks trys to look or it. it crashes.... ><
I am reporting this issue i can provide screen photos of the crash. ^^;

mfld-fr, 2016-08-03:
Hello, thanks for the report, yes your screenshots would be appreciated to help.

jbruchon, 2016-08-05:
I'd also like to see some screenshots of this crash.

sparky4, 2016-10-07:
oh my i apologize for long response! ok i will photo take with latest version. damn it school!

sparky4, 2016-10-07:
ok photos soon going to compile and make xt boot
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
here it is!

hufon, 2016-10-08:
I've exactly the same error on my olivetti PC1! which have no hard disk at all... So this is not an hard disk issue.
[image]

hufon, 2016-10-08:
I'm using the comb_net image

hufon, 2016-10-08:
same problem with comb image...

sparky4, 2016-10-08:
i amusing comb_net also

jbruchon, 2016-10-08:
It is a kernel issue so it will affect all disk images.

sparky4, 2016-10-08:
ah~ ok sorry about my late responce

hufon, 2016-10-09:
May be you should rename the issue...

lithoxs, 2016-10-17:
After seeing the screenshots, what happen is:
1. The initialization reaches the point where the whole kernel is set up succesfully, including the floppy and hard disk driver.
2. Succesfully forks the task that mounts the root directory and succesfully mounts the root drive.
3. The above task attempts to exec the "init" task. At this point the said screenshots were taken.
    The exec syscall first tries to find the binary in the disk and this is the point when it fails. In the first
    screenshot, attempts to exec "/bin/init", and the reported error 4 is "Sector not found". Then tries to                                                  exec "/bin/sh" reporting the same error in the same sector, suggesting that it cannot access even the directory.

Now, the disk driver does not touch the hardware, every thing is done through the int 13h bios interrupt. Further, it only uses the CHS addressing, and never LBA addressing.

I understand that your hardware and disk drive bios are not standard. Also, the needs by ELKS correspond to obsolete hardware. So, does your bios implement what ELKS need?. Is it possible to have a look at the source code of your disk bios?.

hufon, 2016-10-18:
I've no hard disk on my PC1. Only one floppy drive.
It don't works also with 0.2 version.

The drive sounds bad when it fails (like a bad sector on a floppy), but the floppy drive and the floppy disk are both ok

hufon, 2016-10-18:
It's strange because we use both all different PC model... all PC compatibles (mine is without any addon card, standard 1987 bios)

lithoxs, 2016-10-18:
The error indication is when attempts to access block device 0x380 both on sparky4 and hufon machines (according to screenshots), and this number corresponds to the first floppy disk. So, that is correct.

The screenshots indicate failure when attempting to read sector 34. This corresponds to a 2.88 Mbyte floppy disk with a geometry of 80 tracks, 36 sectors and 2 heads. If this is not your floppy then the driver is failing to detect the geometry of your floppy disk. Is this your case?

hufon, 2016-10-19:
I've a dd 720Ko floppy disk 80 track, 9 sector, 2 heads. My Floppy drive is dd only and my floppy controller can only handle dd floppy.
So it seems that the floppy driver fails to detect the geometry of my floppy disk.

jbruchon, 2016-11-25:
I just noticed that the "pre-set floppy types" menuconfig option is set up to ignore manual input and assume 2.88M drives. @sparky4 what type of floppy drive do you have? 720K, 1.44M?

sparky4, 2016-12-06:
!! on my XT i believe it is a 720k one

lithoxs, 2016-12-07:
Looking for the problem I found another bug in the disk driver. I posted the fix for it. Probably this will not solve this issue. Now I suspect the problem begins when mounting the root filesystem. When mounting root, ELKS will print the following:

fd: probing disc in /dev/fd0
fd: /dev/fd0 probably has 18 sectors and 80 cylinders
.....
VFS: Mounted root (minix filesystem)

The number of sectors and cylinders reported may vary, It also may print an error indication.
What prints in your system?

lithoxs, 2016-12-09:
I just posted a pull request that might solve this issue. The problem is produced by a weird behavior of the int 13 BIOS with 720K floppies. Please make tests to confirm that the problem is solved.

hufon, 2016-12-09:
Ok, the problem is solved for me...

hufon, 2016-12-10:
I've another problem (not related) with my CF XTIDE adapter i'll open a new bug

sparky4, 2016-12-17:
ok i will test soon my AC unit got mold and it is hard to test i will pull latest version and test

sparky4, 2016-12-19:
it is defiantly fixed ^^ <3

closed 2016-12-19 by sparky4