ghaerr/elks #45: Nothing loads with my XT-IDE CF Adapter

hufon, 2016-12-10:
When i try to load elks with my XT-IDE CF Adapter connected.
I choose to boot from floppy (by typing "A"). 
The kernel begins to load

Loadling elks
............................................................

And nothing more... no error logs.
When i disconnect the card, everything works fine.

lithoxs, 2016-12-14:
When the PC boots, the hard disk BIOS replaces the floppy handler in the int13. Then, there is some incompatibility between your floppy hardware and the XT-IDE diskette handler.

Further, the message " Loading ELKS kernel" is printed by the boot manager, not the kernel. After the series of dots, it should print "Starting ...".  The last message does not appear, narrowing the source of the problem.

The source code for the boot manager is in dev86 package (https://github.com/jbruchon/dev86). Your problem is narrowed to directory bootblocks, file minix_elks.c, function "code" in that package.

First, try modifying options in the BIOS setup for the XT-IDE card.

Whereas this is not a kernel issue, I think here is the only place where the issue can be solved.

jbruchon, 2017-02-24:
This issue is a couple of months old and no further reports have been posted, so I'm closing it. Re-open it if there is still an issue and you're fairly sure it is fixable in either the ELKS or dev86 code bases.

closed 2017-02-24 by jbruchon