ghaerr/elks #55: select() in userspace crashes the caller

mfld-fr, 2017-02-12:
Version: latest master (commit 661ac7079d6e28b899a515429ab2e1c78e357db1).

The following code causes the process to crash:
```
int main ()
	{
	fd_set fdset;
	FD_ZERO (&fdset);
	FD_SET  (0 /*fd*/, &fdset);
	res = select (0 /*fd*/ + 1, &fdset, NULL, NULL, NULL);
```

Result is:
```
panic: wrong waitpt
apparent call stack:
(...)
```

mfld-fr, 2017-02-12:
As reported by mail by @georgp24 and verified on my side, the crash does not occur when patching with `/fs/select.c` of the 0.2.0 release (commit 3c302cad12d34de58aba1b166da06c9173a43364).

mfld-fr, 2017-02-14:
From what I understand, the problem is coming from calling twice `wait_clear()` at the end of `do_select()`:
```
    wait_clear(&select_poll);
    current->pollhash = 0;
    current->state = TASK_RUNNING;
    wait_clear(&select_poll);
    return count;
}
```
Rather that rolling back all the changes since 0.2.0, it seems to me that removing one of the two calls would be enough to solve the problem.

mfld-fr, 2017-02-14:
Double call to `wait_clear()` was introduced by @lithoxs  in 037b4ec5eb82c475472d2b068b1c139206e3664e.

lithoxs, 2017-02-14:
I agree. My fault not to remove the first wait_clear(). Please submit pull request.

mfld-fr, 2017-02-14:
Thank you @lithoxs for the confirmation !
I am going to cycle a bit with the patch before requesting the pull.

mfld-fr, 2017-02-14:
Cycled 1000 times with the `/bin/eth` test program : PASSED with 2000 packets visible in the QEMU dump.

The commit is now part of the pull request jbruchon#54, that is still open.

closed 2017-02-14 by jbruchon