ghaerr/elks #67: [ktcp] ARP sequencing error

mfld-fr, 2017-02-26:
The current implementation of the ARP module makes the assumption that the next incoming packet after sending an ARP request will be the expected ARP reply. It works in QEMU, but in a real & more loaded network, some other packets can come between the request and the reply. Therefore the implementation of ARP should be kept asynchronous as the IP module.

georgp24, 2017-02-26:
At the moment I do not expect a loaded network on ELKS. But I do not object to modify this.

mfld-fr, 2017-02-26:
In my own roadmap, networking in QEMU is nice but a preliminary step, before networking on my real SBC, that is connected to my internet box, and thus "bombed" at a high rate with many packets that come at any time. So this improvement is a must have.

georgp24, 2017-02-26:
Rereading my post I agree that it indicates that I volunteered to do that. But if you need that I would prefer if you have a go at it.

The ARP request is not triggered by the socket interface but from ip.c resulting from a send packet by an application which is currently in progress. So making this asynchronous is not straightforward.

mfld-fr, 2017-02-27:
Hello Georg, no problem, I started working on it.

jbruchon, 2020-02-23:
Is this bug still present?

georgp24, 2020-02-23:
I cannot test it on real hardware and I expected Marc to fix this.

mfld-fr, 2020-02-24:
Yes, still present.

Mellvik, 2020-02-24:
I can't help fix this, but I can do the physical hardware testing. Let me know.

—Mellvik

ghaerr, 2020-07-21:
This three-year-old issue finally fixed and tested with PR #667. Closing issue.

closed 2020-07-21 by ghaerr

mfld-fr, 2020-07-23:
Nice, thank you !