ghaerr/elks #73: [inet] Hashed stream in case of full TCP window

mfld-fr, 2017-03-02:
While testing the telnet client, I observe that, when the server sends much data, the client reads and displays only a part of this data, then blocks. One has to press a key to make the client to display next data. In Wireshark, these blocking times looks like to happen when the server reaches the TCP window limit advertised by ELKS, that is currently 2 * slip_mtu = 2 * 256 = 512 bytes.

georgp24, 2017-03-02:
I observed the same with urlget, however, after a short while transfer continues, no need to press a key.

I had already mentioned that using slip_mtu should be modified for ethernet. Slip_mtu is used in tcp_output.c. One could test if the interface is ethernet, as being done e.g. in ip.c, and then use a different mtu.

mfld-fr, 2017-03-02:
Having small TCP MTU & window size causes more (and smaller) packets to transfer data, but is not the root cause of the problem. These parameters are initially negotiated between the client and the server at TCP connection phase, then the window size is adjusted continuously during the streaming phase. We should keep small values for these parameters, because ELKS has no much room for additional socket buffers.

mfld-fr, 2017-03-06:
To help understanding how it works and thus how it should be implemented: http://www.tcpipguide.com/free/t_TCPSlidingWindowAcknowledgmentSystemForDataTranspo.htm

ghaerr, 2020-08-25:
> Having small TCP MSS (not MTU) & window size causes more (and smaller) packets to transfer data, but is not the root cause of the problem.

Actually, the root cause of this problem is the implementation of the PTY driver in ELKS.

Should a network process present more data (that is, queue more data into the chqueue queues) than free queue space available, the PTY driver will block the process.

Should more network packet data arrive into ktcp than are buffers available, ktcp will drop the packet. This currently causes an unrecoverable retransmission cycle, but I believe that can be fixed to recover from a dropped receive packet.

The former is the problem.

> While testing the telnet client, I observe that, when the server sends much data, the client reads and displays only a part of this data, then blocks. One has to press a key to make the client to display next data. 

After the PTY process is blocked, which is actually telnetd, nothing happens as telnetd isn't blocked in its normal select() wait. When a character is typed, the TTYP process ends up doing a wake_up which allows telnetd to resume processing normally.

PR #707 fixes this problem by increasing the size of the output queue from 128 to 256 (it was 64 in 2018). Since the current TCP/IP stack has no notion of input nor output window handling, it is currently advertises a window size such that the output queue won't be likely filled.

> In Wireshark, these blocking times looks like to happen when the server reaches the TCP window limit advertised by ELKS.

Yes - the ELKS TCP window limit needs to be less than the PTY output queue size in the current implementation.

Longer term solutions to this problem are a redesign of the PTYP/TTYP character queue handling, which require processes (rather than the kernel as in Linux) to move PTY data, or possibly some assumptions between tty master and slave devices and kernel wakeups, although I'm not sure that will work.

Adding TCP/IP sliding window handling will of course permit larger sized packets to be transferred across the network, but given the already existing ktcp problems with packet buffering, which also use a character queuing mechanism, larger sliding windows will just mean requiring even larger PTY and ktcp buffers. Given that /etc/tcpdev only allows 100 bytes max to be written from ktcp to the kernel, I would say its probably not worth it for this memory-constrained system.

Closing this issue, fixed in #707.

closed 2020-08-25 by ghaerr

ghaerr, 2020-10-09:
The root cause of this problem was confirmed to be the PTY driver sleeping the telnetd process in the ttyp_write routine needlessly when the queue was full, and is permanently fixed in #781.