ghaerr/elks #79: [eth] Document more Ethernet

mfld-fr, 2017-03-04:
Following Derek's problem report on the mailing list (http://www.spinics.net/lists/linux-8086/msg00770.html), document more the Ethernet stuff in the existing "ELKS networking" document.

georgp24, 2017-03-04:
Derek wrote: "sudo ./qemu.sh". It never works for me to run qemu.sh with sudo. 

Maybe you can discuss this in the qemu.html file.

mfld-fr, 2017-03-04:
Oh really ? Starting `qemu.sh` works well with `sudo` on my side, and I wonder why you cannot run the emulator as `root`, the most privileged user ?

One should notice that, if QEMU runs with the help of KVM, you should add yourself to the `kvm` group. But this does not explain the problem you mentioned.

jbruchon, 2017-03-04:
I actually have this in my `/etc/rc` because /dev/kvm default permissions are `root:root` and `0600` and the KVM module doesn't auto-load:
```
modprobe kvm_amd
test -e /dev/kvm && chmod 660 /dev/kvm && chgrp kvm /dev/kvm
```
I'm not endorsing this as the solution, just showing what I've done to make sure the `kvm` group thing actually works as expected.

mfld-fr, 2017-03-04:
Back to Derek's misuse : Georg has now documented how to select the right kernel option in the new `networking_guide.txt` documentation file (see jbruchon#82), and I think it responds to the need stated by this issue.

closed 2017-03-04 by mfld-fr