ghaerr/elks #99: nano-X does not compile

jbruchon, 2017-03-18:
```
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devdraw.o engine/devdraw.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devmouse.o engine/devmouse.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devkbd.o engine/devkbd.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devclip.o engine/devclip.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devrgn.o engine/devrgn.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devpal1.o engine/devpal1.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devpal2.o engine/devpal2.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o engine/devpal4.o engine/devpal4.c
bcc -0 -ansi -Dconst= -Dvolatile= -DELKS=1 -DUNIX=1 -DDEBUG=1 -I/home/daivox/projects/github/elks/nano-X/src -I/usr/bcc/include -I/home/georg/elks-dev/OS-build/elks/include -O -DNONETWORK=1 -DERASEMOVE=1 -DUPDATEREGIONS=1   -c -o drivers/scr_bios.o drivers/scr_bios.c
drivers/scr_bios.c:40: error: Cannot open include file
```

georgp24, 2017-03-18:
I guess the parameter: ` -I/home/georg/elks-dev/OS-build/elks/include`
is the problem. You do not have "home/georg" on your PC and it does not point into the right direction anyway.

I will look into it.

closed 2017-03-18 by jbruchon

mfld-fr, 2018-04-17:
Actually fixed in #100.