ghaerr/elks #103: Shell script will not read parameters passed

georgp24, 2017-03-25:
The following one line script will not echo the parameter passed:
`echo $1 `

So if you run this script line with e.g. `test.sh 123` it will not echo `123`

You cannot use $1 with other commands as well.

mfld-fr, 2017-11-10:
While compiling with GCC-IA16, saw a big warning about arguments in execve. Have to look deeper at that, because it seems related to this issue.

mfld-fr, 2018-04-17:
Just tested it on `full3` image as root under QEMU with default configuration:
```
echo echo `$*` > test.sh
./test.sh a b c
```
Result is:
```
a b c
```
So it works well with the `ash` shell that is linked from `sh` in that configuration.
I also tested with `sash` shell, that works well either.
Therefore closing this issue as non-reproducible.

closed 2018-04-17 by mfld-fr