ghaerr/elks #124: Console i/o

Mellvik, 2017-05-03:
When using BIOS console I/O, the system softhangs on input of a char < (ascii)6 (such as ctrl-C). The problem goes away when the **cmp al,#6** instruction  in kbd_timer() is removed. (bioscon.c)

Mellvik, 2017-05-04:
Major improvement!! Virtual consoles now work, but: text written to an invisible console, does not show. The cursor is in the right position, that's it (tested on physical 286). Fuctions keys otherwise not tested. BTW (possibly unrelated, and has existed for a while): ^Z hangs the system. Maybe all the job control stuff should be eliminated?

mfld-fr, 2018-05-03:
This issue deals with several mixed points that would be addressed one by one:
- [ ] Control-C (or any char < 6) in BIOS console hangs the system : TBC
- [ ] Control-Z in BIOS console hangs the system : TBC
- [ ] Text written to hidden virtual console is not visible: TBC

mfld-fr, 2018-05-17:
Just tested on latest master, Control-C and Control-Z do not hang the system any more. But Control-C in `ash` still causes the spurious display of `HOME not set` (already tracked by #125).

mfld-fr, 2018-05-17:
Problem confirmed with the virtual consoles with the BIOS driver. After enabling `getty` for slots 2 and 3 in `/etc/inittab`, cursor is displayed but no text visible when switching to VC 3, and no cursor neither text visible at all in VC2. Can login in VC3, but not in VC2.

mfld-fr, 2018-05-20:
Found the cause of the VC2 problem: in `qemu.sh` script, the requested VGA card is the Cirrus one (`-vga cirrus`), but the VGA BIOS is not more consistent since QEMU 2.2. Solved that problem with option `-vga std`. Now have the same behaviour as reported by @Mellvik.

mfld-fr, 2018-05-20:
Hidden text problem fixed in #216.

PR #216 (2018-05-20): Virtual consoles tuning
- fixes #124 : initial text not visible in virtual consoles 2 and 3 with BIOS driver
- consoles 2 and 3 activated in `/etc/inittab`
- extra: automatic CPU model for IBM XT/AT configuration
commit 31d26e88 (2018-05-20): Merge pull request #216 from mfld-fr/next

- fixes #124 : initial text not visible in virtual consoles 2 and 3 with BIOS driver
- consoles 2 and 3 activated in /etc/inittab
- (extra) automatic CPU model for IBM XT/AT configuration
commit d25164a0 (2018-05-20): Fix initial text hidden in virtual consoles

Fixes #124
closed 2018-05-20 by mfld-fr