ghaerr/elks #166: GCC-IA16 guidelines

mfld-fr, 2017-11-03:
@lithoxs : some times before, you announced that you started to compile the ELKS kernel with GCC-IA6 (http://www.spinics.net/lists/linux-8086/msg00817.html).

As I can see in your PR flow, you still maintain the build with GCC-IA16, and it looks like, after tons of (more or less serious) posts to discuss that topic, that this compiler is today the best alternative to BCC.

Not only because of the work you already performed, but also because this GCC fork is still alive & able to be mainstreamed: https://github.com/tkchia/gcc-ia16

Even Alan is keeping an eye on this option: https://github.com/crtc-demos/gcc-ia16/issues/4

It would be nice if you could provide some guidelines to use that compiler in the /Documentation folder, so that we could be more to use it to improve the ELKS code.

lithoxs, 2017-11-03:
There are two compilers for 8086 based on GCC:
 ia16-unknown-elks-gcc, the original compiler by Rask, with a few improvements by me, and
 ia-elf-gcc, based on the work by Rask, with many improvements.

For the first, I posted here http://www.spinics.net/lists/linux-8086/msg00674.html instructions to build the compiler. It tested the building in Debian squezze 32 bits and Ubuntu 13.1 64 bits, producing a working cross-compiler. Trying to build in Debian jessie 64 bits fails.

This compiler can build applications for ELKS, but only some applications work well. The C library is based in newlib and is quite bloaty, and the linker likes to place everything in a library to the final binary. So, applications size are much larger than those produced by BCC.

The kernel can be built with this compiler. To avoid rewriting everything in assembly, I wrote an utility to convert ATT 8086 assembly to AS86, and rewrite the rule for compilation: first compiles with the S option, producing an assembly file, then convert it to as86 with the above utility, assemble it with AS86, and remove the garbage. Everything is already in the Makefile, but you need the conversion utility. It was necessary to write one function of libgcc.

The second compiler is newer, faster and produces smaller code. It also relies on newlib and has the same linker as before. So, the problems of first compiler in this regard are the same. As distributed, the compiler cannot produce ELKS binaries. To build the kernel, the Makefile needs further changes, incompatibles with  ia16-unknown-elks-gcc.

lithoxs, 2017-11-03:
INSTRUCTIONS TO COMPILE THE KERNEL USING ia16-unknown-elks-gcc.

1. Build the toolchain. Follow instruction from http://www.spinics.net/lists/linux-8086/msg00674.html .
2. Build the conversion utility. Name it att2as86
3. Make sure the above programs are in your PATH. I placed toolchain and att2as86 in $HOME/cross/bin
4. In file Makefile-rules, find a line with:
#USEIA16   = y
and uncomment it.
5. Build as usual.

lithoxs, 2017-11-03:
Now, the conversion utility.
/*
Copyright (C) 2014 Juan Perez

This program is free software; you can redistribute it and/or
modify it under the terms of the GNU General Public License
as published by the Free Software Foundation; either version 2
of the License, or (at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

See the GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program; if not, write to the Free Software
Foundation, Inc., 59 Temple Place - Suite 330, Boston, MA  02111-1307, USA.

*/

/* gnu as to as86 source code converter */

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <ctype.h>

#define MAX_LINE	4096
#define MAX_OPERANDS	5

static char buffer[MAX_LINE];
static char *operands[MAX_OPERANDS];
static int sects[] = {'T', 'D', 'B', 'R',};
static char *sectmsg[] = {".text", ".data", ".bss", ".rom", ".undef",};

typedef struct {
	char	*att;
	char	*as86;
	int	type;
} instruction;
static int opcode = 0;
static int lineno = 1;
static char *imm_one = "#1";

static instruction mnemonics[] = {
	{".align",  ".align", 12},
	{".arch",   ".", 0},
	{".ascii",  ".ascii", 12},
	{".asciz",  ".asciz", 12},
	{".att_syntax",  ".", 0},
	{".byte",   ".byte", 12},
	{".bss",    ".bss", 12},
	{".code16", ".", 0},
	{".comm",   ".comm", 9},
	{".data",   ".data", 12},
	{".extern", ".globl", 12},
	{".global", "export", 12},
	{".hword",  ".word", 12},
	{".ident",  ".", 0},
	{".local",   ".", 0,},
	{".long",   ".long", 12,},
	{".p2align",".align", 7},
	{".section",".sect", 6},
	{".single", ".single", 12},
	{".size",   ".", 0},
	{".skip",   ".space", 8},
	{".string", ".asciz", 12},
	{".text",   ".text", 1},
	{".type",   ".", 0},
[... 475 lines cut ...]
    }
    i = 0;
    while(nops--) {
	if(strlen(operands[i]))
	    printf("%s", operands[i]);
	if(nops)
	    printf(",");
	i++;
    };
    printf("\n");
    lineno++;
    return chr;
}

int main (int argc, char **argv)
{
/*char buffer[MAX_LINE];
char *operands[MAX_OPERANDS];*/
    int chr;

    do {
	chr = line();
    } while(chr != EOF);
    return 0;
}

lithoxs, 2017-11-03:
Things to do to switch compiler.

1. Decide if we will be using ia16-gcc only for the kernel or also for applications.
2. If also for applications, solve the problems with the library and linker. My suggestion is to port elks-libc to replace newlib. I suspect newlib is hopeless.
3. Rewrite all assembly from as86 to att.
4. Modify Makefiles to use the new toolchain. In particular, find a way for the linker to include only used modules, not everything as now.
5. Remove BCC-isms from the kernel code.

lithoxs, 2017-11-03:
So, now, what do you prefer: ia16-unknown-elks-gcc or ia-elf-gcc?
As a curiosity or permanent?
Only for kernel or also for applications?
I have patches to do the same for OpenWatcom as now for  ia16-unknown-elks-gcc,
except for a conversion utility from intel syntax to as86 syntax. Any interest on them?
I will submit PR to Documentation based on your response.

mfld-fr, 2017-11-03:
Thank you very much for this big amount of information. Mmm... at a first glance I would prefer to spend time on the second compiler, because it begins to support far pointers in the latest fork available on GitHub. But let me try the two options before giving my opinion...

lithoxs, 2017-11-04:
For the second compiler the final bit of steps is to apply the next patch before building the kernel:

diff -Nur elks.git/elks/Makefile-rules elks/Makefile-rules
--- elks.git/elks/Makefile-rules	2017-04-26 18:53:09.282883431 -0500
+++ elks/Makefile-rules	2017-05-24 18:25:28.009053605 -0500
@@ -181,7 +181,7 @@
 ifeq ($(MK_CPU), 8086)
 ifeq ($(USEIA16), y)
     CPU_AS	= -0
-    CPU_CC	= -mtune=i8086 -fno-inline -fdata-sections -ffunction-sections -mseparate-code-segment -Wl,--gc-sections
+    CPU_CC	= -mtune=i8086 -fno-inline -fdata-sections -fleading-underscore -ffunction-sections -mseparate-code-segment -Wl,--gc-sections
     CPU_LD	= -0
 else
     CPU_AS	= -0
@@ -193,7 +193,7 @@
 ifeq ($(MK_CPU), 80186)
 ifeq ($(USEIA16), y)
     CPU_AS	= -1
-    CPU_CC	= -mtune=i80186 -fno-inline -fdata-sections -ffunction-sections -mseparate-code-segment -Wl,--gc-sections
+    CPU_CC	= -mtune=i80186 -fno-inline -fdata-sections -fleading-underscore -ffunction-sections -mseparate-code-segment -Wl,--gc-sections
     CPU_LD	= -0
 else
     CPU_AS	= -1
@@ -205,7 +205,7 @@
 ifeq ($(MK_CPU), 80286)
 ifeq ($(USEIA16), y)
     CPU_AS	= -2
-    CPU_CC	= -mtune=i80286 -fno-inline -fdata-sections -ffunction-sections -mseparate-code-segment -Wl,--gc-sections
+    CPU_CC	= -mtune=i80286 -fno-inline -fdata-sections -fleading-underscore -ffunction-sections -mseparate-code-segment -Wl,--gc-sections
     CPU_LD	= -0
 else
     CPU_AS	= -2
@@ -268,7 +268,7 @@
 
 AS		= as86
 ASFLAGS 	= $(CPU_AS) $(ARCH_AS)
-CC		= ia16-unknown-elks-gcc
+CC		= ia16-elf-gcc
 CFLAGS		= $(CPU_CC) $(ARCH_CC) -Os $(CFLBASE) -Wall
 LD		= ld86
 LDFLAGS 	= $(CPU_LD) -s

mfld-fr, 2017-11-04:
Tip: surround the pasted code with MD quotes:
'''
Like this
'''

mfld-fr, 2017-11-04:
Could you please quote the text of your conversion utility for correct rendering ?
```
Like this
```

mfld-fr, 2017-11-06:
Thanks Jody, it is going to help now because I need it after succesful build of latest gcc-ia16.

mfld-fr, 2017-11-06:
I begin to like this compiler (fork version 6.3.0 from https://github.com/tkchia/gcc-ia16) 😁 👍 
```
ia16-elf-gcc -mtune=i8086 -fno-inline -fdata-sections -ffunction-sections -mseparate-code-segment -Wl,--gc-sections -fleading-underscore  -Os -I/mnt/data/home/mfld/advantech/elks/elks/include -DELKS_VERSION_CODE=0x00020000 -DUTS_RELEASE=\"0.2.0\" -D__KERNEL__ -Wall -S -o printk.s printk.c
printk.c: In function ‘numout’:
printk.c:80:7: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     i = 10;
       ^
printk.c:83:4: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
  i = 8;
    ^
printk.c:87:4: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
  i = 11;
    ^
printk.c: In function ‘printk’:
printk.c:200:5: error: address of register variable ‘p’ requested
     va_start(p, fmt);
     ^~~~~~~~
printk.c:202:5: error: address of register variable ‘p’ requested
     va_end(p);
     ^~~~~~
```

lithoxs, 2017-11-07:
I fixed most problems with ia16-elf-gcc, see latest PR. But there is a problem with commit dated Nov/03 at 15:47. I'll check that problem later

mfld-fr, 2017-11-07:
Yes, the `mkromfs` host tool is to be compiled with the host compiler, not the cross one. We should rework the Makefile globals to handle both cases: host native GCC and target cross GCC-IA16.

lithoxs, 2017-11-07:
I don't know how mkromfs is compiled, but for everything else, this is the way it works now. There is no need to rework any Makefile.

mfld-fr, 2017-11-08:
At this point, I think the safest option is to keep the current build chain as designed by Juan, i.e. to keep `dev86` (at least `bin86` as named by many distros) to assemble and link, and to replace BCC by the latest GCC-IA16 as C front-end. This option is enough to fulfill our current needs (C99 compiler, better code checking and optimization).

It would be a good milestone if all the current kernel code cleaner and upgraded to C99 standard, with no more ugly things (like using a pointer as a scalar to being able to optimize). So let us work to make the whole current code to compile and work with GCC-IA16, so that we could later drop the first and go further.

Jody could ever create a new integration branch dedicated to that, in order to preserve the current master branch and to wait for that milestone before switching definitively on the new build chain.

lithoxs, 2017-11-08:
I agree with keeping the current build toolchain as the safest option, as a first step to a complete migration.
However, it is not necessary for everybody to build the latest compilers. For example, I use the stock bcc package from Debian and it works fine. I have the Mentor ia16-elf-gcc compiler and it offers more than is needed in this transition step.
We can attract more users and developers if we keep easy getting required tools. Compiling the compiler can get complicated: there are requisite development libraries and tools, with certain range of version, and even then you might require to implement workarounds.
And it is a bad idea to use features that needs a certain compiler. One bad day the developer slips with a banana peel and the requisite compiler branch will get abandoned, without anybody to fix the pending bugs.

So, I think is better to include in Documents ways to get the requisite tools, from prebuilt packages and up to your scripts for those willing to spend the effort and time.

lithoxs, 2017-11-08:
There are a lot of pending issues to be solved before abandoning BCC.

1. We need extensive testing of kernels compiled with ia16-gcc, with different configuration options and under different load levels. How will be able to say this step is complete if we don't have a test suite?

2. The conversion utility needs more work. The data and bss sizes are very different for the kernel compiled with BCC and ia16-gcc.

3. There is a big mess with headers files that needs to be cleaned up. The earlier developers dumped the BCC compiler include files in the directory include/linuxmt and include/arch. Then, contaminated some compiler include files with definitions that should be somewhere else. For example, stdarg.h, stddef.h, stdlib.h, limits.h should be accessed by the compiler from the standard locations (/usr/lib/bcc/include in BCC case).
Then, we have to eliminate some include files and some of their contents moved to other include files.

4. Carefully review the code to identify volatile variables.

mfld-fr, 2017-11-23:
GCC is now integrated in the ELKS mainstream, and the provided guidelines implemented in configuration and build scripts. I think we could close that issue and focus now on the first migration step as discussed above.

closed 2017-11-24 by mfld-fr