ghaerr/elks #192: build.sh fails. pushd is not found

sparky4, 2018-03-14:
[12:43:40][~/p/elks]
[sparky4@lappy4]$ sudo ./build.sh 
-e 
WARNING: Disk images can only be built if you have root permissions
./build.sh: 38: ./build.sh: pushd: not found
./build.sh: 39: .: Can't open ./env.sh

sparky4, 2018-03-14:
also

bcc -s -ansi -o ash builtins.o cd.o dirent.o error.o eval.o exec.o expand.o input.o jobs.o mail.o main.o memalloc.o miscbltin.o mystring.o nodes.o options.o parser.o redir.o show.o signames.o syntax.o trap.o output.o var.o init.o bltin/echo.o bltin/expr.o bltin/regexp.o bltin/operators.o
make[2]: Leaving directory '/home/sparky4/p/elks/elkscmd/ash'
make[2]: Entering directory '/home/sparky4/p/elks/elkscmd/bc'
flex -I -8 scan.l
make[2]: flex: Command not found
Makefile:80: recipe for target 'scan.c' failed
make[2]: *** [scan.c] Error 127
make[2]: Leaving directory '/home/sparky4/p/elks/elkscmd/bc'
Makefile:52: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/home/sparky4/p/elks/elkscmd'
Make.defs:192: recipe for target 'me' failed
make: *** [me] Error 2

when i try to make in elkscmd

sparky4, 2018-03-14:
i changed
#!/bin/sh
to
#!/bin/bash

and it fixed it!

sparky4, 2018-03-14:
flex -I -8 scan.l
make[2]: flex: Command not found
Makefile:80: recipe for target 'scan.c' failed
make[2]: *** [scan.c] Error 127
make[2]: Leaving directory '/home/sparky4/p/elks/elkscmd/bc'
Makefile:52: recipe for target 'all' failed
make[1]: *** [all] Error 1
make[1]: Leaving directory '/home/sparky4/p/elks/elkscmd'
Make.defs:192: recipe for target 'me' failed
make: *** [me] Error 2
Build script has terminated with error 7

flex..

sparky4, 2018-03-14:
make[2]: *** No rule to make target 'min_rfs'.  Stop.
Makefile:119: recipe for target '_populate_target' failed
make[1]: *** [_populate_target] Error 1
make[1]: Leaving directory '/home/sparky4/p/elks/elkscmd'
Makefile:209: recipe for target 'root' failed
make: *** [root] Error 2
Build script has terminated with error 8

min_rfs? sigh...

mfld-fr, 2018-04-11:
I am able to reproduce the reported failure with the latest version.
I guess it is coming from the import of the /dev86 `bootblocks` in commit 0deea11.
Let me fix the Makefile...

mfld-fr, 2018-04-11:
@sparky4 : please update your local clone and test again.

mfld-fr, 2018-04-12:
@jbruchon : thanks for the quick pull (as usual). I think you could also close this issue.

closed 2018-04-12 by jbruchon