ghaerr/elks #199: Move HD image creation back into main build

mfld-fr, 2018-04-23:
Next step after PR #197:

The standalone script `build_hd_image.sh` from @georgp24 generates an HD image by expanding the FD1440 one. Generate this HD image directly in the main build process.

mfld-fr, 2020-02-28:
HD image (**without MBR**) now included in configuration. Bootable as 'big' floppy image, but not as true HD image (see #348). Still some work to add an MBR into that HD image.

ghaerr, 2020-03-01:
Hello @mfld-fr,

How are you thinking this should be completed, since everyone seems to want this now?

How about the following, which would help for development and testing: When CONFIG_IMG_HD is set in config, the build creates two output files (for the time being), a 'flat or floppy' fd.bin, which is built using the size and CHS configured, using the same method now in image/Makefile, and also an hd.bin image, which would be the result of assembling a dev86 'bootblocks/mbr.s' ported to the new asm format and taking in the config CHS parameters, which would then be prepended to fd.bin, creating an MBR hard disk image. Initially, this would only be done for MINIX bootable images, leaving FAT to worry about later.

This would allow testing of both flat and MBR disk images, since there seems to be some booting issues presently with both. I'm trying to understand what exactly needs to get done to move HD boot and image creation forward.

mfld-fr, 2020-03-02:
@ghaerr : I like the idea to generate both FD and HD images in the same run. Moreover, if the FD image has always the `fd.bin` image, one would not have to edit the `qemu.sh` script whatever the image format is.

To complete the current HD generation, I think the easiest and most consistent would be to reuse @georgp24's stuff from his standalone script and inject it into the current image Makefile.

georgp24, 2020-03-02:
@mfld-fr : The menuconfig system currently supports generating a hard disk, but this is not partitioned nor formatted. So if you select that, you get no disk to boot from.

Therefore I suggest to modify menuconfig so you can generate both a floppy image plus a hard disk to be used as DISK2 in qemu.sh. With other words, not just either a floppy image or a hard disk image but floppy image and hard disk image to select and configure separately.

ghaerr, 2020-03-02:
Hello @mfld-fr,

>  I like the idea to generate both FD and HD images in the same run. Moreover, if the FD image has always the `fd.bin` name, one would not have to edit the `qemu.sh` script whatever the image format is.

Yes, a name change would make things more constistent, and then when building HD images in images/Makefile, both a flat and MBR version created.

 
> To complete the current HD generation, I think the easiest and most consistent would be to reuse @georgp24's stuff from his standalone script and inject it into the current image Makefile.

Using any components of that script won't work, since it's dependent on using a loopback driver, and MINIX filesystem support, neither of which are present on OSX. My question was more about the means of creating an MBR image from a flat image: The MBR image is just the concatenation of the appropriately-assembled 'mbr.s' from the config settings and the 'fd.bin' flat image to create 'hd.bin', correct? The fd.bin image doesn't need any modifications, only the addition of the MBR, which could be done with the 'cat' command. So the only real changes are porting 'mbr.s', and image/Makefile, correct?

mfld-fr, 2020-03-02:
Not sure... when the MBR pass control to the VBR, it provides some information on the current boot partition (see https://en.wikipedia.org/wiki/Master_boot_record#MBR_to_VBR_interface). The current VBR (= ELKS boot sector) does not use that information,  and does not offset the CHS according to that partition.

tkchia, 2020-03-09:
Hello @ghaerr, hello @mfld-fr,

> when the MBR pass control to the VBR, it provides some information on the current boot partition (see https://en.wikipedia.org/wiki/Master_boot_record#MBR_to_VBR_interface).

Thanks @mfld-fr --- I had not known about this `%ds:%si` or `%ds:%bp` pointer.

Yes --- the current bootloader code assumes that the filesystem (Minix or FAT) starts at sector 0, so this will need to change.  The FAT bootloader can use the `bpb_hidd_sec` field --- which is part of Microsoft's BPB --- but the Minix bootloader will need some other way to figure out the start of the partition.

Thank you!

tkchia, 2020-03-09:
I guess one thing the Minix FS bootloader can do is to use the `%ds:%si` or `%ds:%bp` pointer to read the partition table entry, to figure out where it was booted from.  (This means the bootloader must figure out which of these two is the correct pointer...)

An alternative would be to prepend a new "partition start" field to the ELKS parameter block (`struct elks_disk_parms`), similar to the "hidden sectors" field in the FAT BPB.  However I am still a little wary of simply adding a 4-byte (or even 3-byte) field to a structure which may become part of a binary protocol.

Thank you!

mfld-fr, 2020-03-09:
@tkchia : this alternative conflicts with your proposal to have a partition-independent ELKS image that the user could install on any existing partition, doesn't it ?

ghaerr, 2020-03-09:
> @tkchia : this alternative conflicts with your proposal to have a partition-independent ELKS image that the user could install on any existing partition, doesn't it ?

I think the 'flat' boot images should not have a portion in the boot block, either Minix EPB nor FAT bpb_hidd_sec (which I'm not sure is touched when partitions are managed anyways) to determine offset. Instead, some mechanism of how the either a 'standard?' MBR or an ELKS MBR boot conveys the partition offset from the MBR table should be used. I have not yet read the documentation mentioned earlier due to other pressing issues (but will).

ghaerr, 2020-03-09:
> This means the bootloader must figure out which of these two is the correct pointer...)

Since the Wiki article indicates that DS:SI always points to the active  partition tablntrye, and DS:BP only optionally does, it seems using DS:SI is the correct pointer... (This could be checked looking at the many `dev86` MBR loaders).

tkchia, 2020-03-10:
Hello @mfld-fr,

> this alternative conflicts with your proposal to have a partition-independent ELKS image that the user could install on any existing partition, doesn't it ?

It kind of does, but interestingly, but having a "hidden sectors" field means we do not need to make any assumptions about `%ds:%si` or `%ds:%bp`.  It is quite messy, indeed. :-|

Hello @ghaerr,

> Since the Wiki article indicates that DS:SI always points to the active partition tablntrye, and DS:BP only optionally does, it seems using DS:SI is the correct pointer...

Unfortunately, "[t]he Windows Vista/7 MBRs no longer provide this DS:SI pointer."  A [disassembly of the MBR code in modern Windows versions](https://thestarman.pcministry.com/asm/mbr/W7MBR.htm) by Daniel Sedory also seems to corroborate this: the MBR sets `%ds:%bp` to point to the table entry to be booted, but not `%ds:%si`.

Thank you!

ghaerr, 2020-04-03:
Hello @mfld-fr,

PR #506, which is completed and ready for commit, completes this request which was opened two years ago and discussed this month. 

A recap of how it works: #506 adds the ability for any ELKS HD (or floppy) image to be booted from an MBR partition, and also adds the ability to create an MBR-partitioned HD image by prepending the new MBR boot sector 'bootblocks/mbr.S' to an ELKS 'flat' image, and using 'setboot' (or another partition tool) to define the partition. By default, 'setboot' will create partition 1 using the defined CHS for the HD volume to set the partition size.

The MBR to VBR interface in 'boot_sect.S' uses DS:SI or DS:BP to point to the booted partition record, of which the starting sector offset is used to load 'linux', and then the BIOS hd/floppy driver uses the offset to determine the /dev/bda partition to set ROOT_DEV. Currently, the MBR passes a magic number in AX to allow use of the same VBR boot sector for floppy and hard disk images. The magic number could be removed, but then we'd need to have separate boot sectors for hard drive images.

When selected in config, the extra images produced include `hd32mbr-minix.bin` and `hd32mbr-fat.bin` for testing. Converting any custom ELKS disk image to MBR is done with two commands (see images/Makefile::hd32mbr-minix'.

Is there anything else you'd like to see in getting the HD image creation back into the main build? PR #506 pretty much completes all work that I know of for HD image creation at this point.

mfld-fr, 2020-04-04:
Hello @ghaerr, it looks like this issue is now fixed by your latest changes. Thanks.

closed 2020-04-04 by mfld-fr