ghaerr/elks #207: 386 config fails to build

Mellvik, 2018-05-07:
Setting processor type to 386 in the config cause the build to fail in bootsect.s (as86 -3):

	00557                                           	mov	sectors, ax
	***** illegal immediate mode...........................................^
	***** mismatched size………………………………………………^
	… and so on …

mfld-fr, 2018-05-16:
Hi, could you explain what do you expect from that setting ? Today it only allows 386 detection in `setup.S`, resulting in a nice but useless 'CPU 80386` at ELKS start-up prompt in `mm/init.c`.

Moreover, that setting badly triggers 32 bits segment for `as86` in the build script. As 32 bits is not in ELKS scope, I intend to fix that bug by cleaning all the stuff linked to CPU model above 386.

Mellvik, 2018-05-16:
Why is there a 386 config option in the first place, when it does nothing meaningful and cause the build to fail?

For anyone interested in experimenting with ELKS (there should be many more of us!), testing different config options when things don’t work, is natural and should be encouraged. 

I’d go for simplicity. Your description suggests that anything 32 bit should go - it’s not in scope, and thus not meaningful. Further, someone mentioned 286 protected mode a while back. Maybe that should go too, unless someone has a really good reason to keep it.

Recent progress has been very encouraging. I’m inclined to start fixing telnetd and add DHCP if no one else have that on the agenda.

—Helge

mfld-fr, 2018-05-16:
Well... as I understood for now, ELKS started many years ago by forking from his big brother, that required at least a 386 for paging. I have the feeling that the guys who downsized Linux had some expectations to run ELKS on a wide range of hardware, and were a bit 'conservative' with many pieces of code from that epic period. Maybe @jbruchon could better explain the development strategy at that earlier time.

Except having fun to run a Linux-like OS on antic 80's & 90's PCs, my opinion is that today the only place where ELKS could still evolve is the embedded world, with x86 clones or compatible soft cores in FPGA personal projects, that still reuse the huge legacy of the PC/XT/AT platform despite the ARM big pressure.

So I agree to drop the 16 bits protected mode that is not needed in that context, in addition to be painful or a blocking point for features where you have to work with many segments, and finally has demonstrated over 3 decades it was not suitable for a general purpose OS.

But before dropping that mode, we should first get @jbruchon agreement, and check on the mailing-list.

mfld-fr, 2018-05-16:
Fixed in #215

Mellvik, 2018-05-16:
BTW - and this is unrelated to #207:

I mentioned before - a few weeks back, that the memcmp function seems suspicious.
It is used in the eth.c test program - and reports unequal when the compared addresses are indeed equal. I wrote a test program and get the same symptoms. Rarely does memcmp give the expected result.

It may be my test, but still: Can someone with a better handle on x86 assembly take a look at memcmp.s ??

—Helge

mfld-fr, 2018-05-16:
In order to help us to track efficiently the bugs, please create another issue for that problem, and more generally, one issue per problem.

PR #215 (2018-05-17): Bunch of non-critical commits
- add a precursor of the early `printk()` (normal one is not available until console initialization)
- more intuitive configuration layout for hardware description and kernel options
- remove support of 80386 and later (fixes #207)
- move documentation folder to top one
- select() behaviour compliant to POSIX after a timeout (fixes #213)
commit 317f048a (2018-05-17): Remove support of 386 and later

ELKS was not using the features of 386 and later
except for CPU model detection and display. Selecting
386 or 486 was also crashing the build.

Fixes #207 (build crash)
commit 53e0057a (2018-05-17): Merge pull request #215 from mfld-fr/next

- add a precursor of the early printk() (normal one is not available until console initialization)
- more intuitive configuration layout for hardware description and kernel options
- remove support of 80386 and later (fixes #207)
- move documentation folder to top one
- select() behaviour compliant to POSIX after a timeout (fixes #213)
closed 2018-05-17 by mfld-fr