ghaerr/elks #221: Create a new integration branch to host the toolchain migration

mfld-fr, 2018-06-11:
@jbruchon : as the toolchain migration would make ELKS unstable for quite a long time until we complete and mature the changes, I think it would be safe to fork the master branch into another one dedicated to that migration.

If you agree, could you then please create another branch from master, and also set up the master and the migration branches as protected ?

mfld-fr, 2018-06-16:
Hello @jbruchon, I expect you are in a better shape and will be fully back with us soon. Have you recovered enough to process this request ? Looking forward to get your news.

mfld-fr, 2018-06-24:
Looks like @jbruchon is not available these times. Will stabilize in my own branch before pushing to upstream.

jbruchon, 2018-06-24:
I'm around, just busy with other things.

jbruchon, 2018-06-24:
How's this? https://github.com/jbruchon/elks/tree/switch-to-gcc

mfld-fr, 2018-06-24:
Looks fine, thanks, just missing the protection on both branches.

jbruchon, 2018-06-24:
What do you mean by "protected?"

jbruchon, 2018-06-24:
Never mind, I got it.

mfld-fr, 2019-03-13:
That working branch has been merged to main, could you please delete it now ? Thanks.

jbruchon, 2019-03-14:
Deleted.

closed 2019-03-14 by jbruchon