ghaerr/elks #225: GNU branch

mfld-fr, 2018-06-28:
Current version in `switch-to-gnu` branch is now able to login and execute some commands. This issue tracks the remaining points to address to reach an acceptable quality level :

- [ ] Fixes the GCC warnings (1988 warnings left !)
- [ ] Remove the unit test programs in the default configuration
- [ ] Rework the heap + stack space allocation in execve()
- [ ] Reactivate the heap + stack size parameter (previous ld86 -H option)
- [ ] Wipe out the remaining DEV86 stuff (in tools)
- [ ] Optimize the string functions (currently only in C)
- [ ] Split the large objects file in LIBC to optimize the executable sizes
- [ ] Understand the data sizes changes with & without -Os option
- [ ] Remove duplicated code between kernel and LIBC
- [ ] Restore the compression programs (need a GCC primitive)

jbruchon, 2018-06-29:
I have long thought that removing the redundancy between the kernel and C library would be a great idea. I don't know what the best way to implement this would be though. The libc bits are statically linked into every program such that they're all stand-alone binaries and there is no support for shared libraries and dynamic loading which would make the task a breeze. How much of libc ends up in the kernel? It seems like the best route would be to add shared library support but that implies things like `-fPIC` which will make the code larger and slower. There seems to be no good solution I can think of.

What's this one? "Optimize the string functions (currently only in C)"

mfld-fr, 2018-07-03:
I am thinking about a lower cost solution to share code between the applets than implementing the "shared object file" feature : moving the LIBC code for large functions like printf() to a standalone executable with only a code segment, restore the old "executable sticky bit" to make this executable to stay in memory, and using the interrupt vector 81h as a trampoline for such functions (same technique as for DOS extension modules).

The "*Optimize the string functions (currently only in C)*" is to restore the assembler implementation of some functions like memcpy(), that I temporarily disabled during the move to GCC.

mfld-fr, 2019-01-28:
The `switch-to-gnu` branch is currently blocked because the MINIX `bootblock` is still built with `dev86`. Currently reworking that block... also an opportunity to simplify by directly loading the kernel from that block without the intermediate `minix_elks.bin` helper.

mfld-fr, 2019-03-03:
Now unblocked after commit 9e038b8. Still a draft of a simplified boot process from the MINIX filesystem, with some limitations (only FD1440 supported, kernel size limited to 64K), but works well under QEMU (faster boot as intermediate 'minix_elks' boot helper was dropped).

mfld-fr, 2019-03-13:
Submitted as #240.

closed 2019-03-13 by mfld-fr