ghaerr/elks #226: Compression of user land executables

mfld-fr, 2018-11-20:
While testing ELKS with kernel & root FS in ROM, I quickly come to exhaust the ROM space, because even with GCC-IA16 performance in code reduction (`-Os` option), and until the availability of the optimized LIBC, each user land executable still sizes several KB.

On the mailing list (https://www.spinics.net/lists/linux-8086/msg00768.html), @EtchedPixels suggested to use compression as an option, with an Exomizer-like for low decompression cost.

This issue tracks the integration of such on-the-fly decompression in ELKS. Generic algorithm is currently developed in another project (https://www.github.com/mfld-fr/compress).

tkchia, 2018-11-26:
Hello @mfld-fr ,

On a tangentially related note --- and in case you are not already aware --- I recently implemented an alternate calling convention named `regparmcall` in IA-16 GCC, where the first few arguments to a function can go into registers rather than onto the stack.  This can also potentially help save some code space (and time) for userland programs.  However, some extra work is probably needed to add [multilib support](https://gcc.gnu.org/onlinedocs/gccint/Target-Fragment.html) to `elks-libc` so that everything works.

Thank you!

mfld-fr, 2018-11-30:
Hello @tkchia, thanks for the hint, I agree it is worth to experiment.

But for the moment, I still try the compression option, because first results show that we can gain several KB at very low cost (see the benchmark in https://github.com/mfld-fr/compress/blob/master/README.txt).

Mellvik, 2018-11-30:
This does indeed sound interesting, but keep in mind that the target hardware may have limited resources to accommodate compression/decompression. Try ZIP on a (physical) 286 with or without FPU…

—Helge

mfld-fr, 2018-11-30:
@Mellvik : true, and so taken as a design constraint : decompression has to be possible with a little piece of 8086 code added to the kernel setup, or as a kernel routine.

tkchia, 2018-11-30:
Hello @mfld-fr ,

> > On a tangentially related note --- and in case you are not already aware --- I recently implemented an alternate calling convention named `regparmcall` in IA-16 GCC

> Hello @tkchia, thanks for the hint, I agree it is worth to experiment.
> But for the moment, I still try the compression option

Well, thanks --- that was just a heads-up.  (Hopefully later on we can try to do both, i.e. use an alternate calling convention _and_ apply compression to userland code.)

> data: kernel data only
> code: kernel code only
> ash: shell (mixed code & data)

If you would like to also compress the _kernel_, I think it might be feasible to use a more powerful compression method for that, since the kernel only needs to be expanded once per system bootup.  The [FreeDOS kernel](https://github.com/FDOS/kernel) sources already use UPX (plus some pre- and post-processing) to compress the binary kernel as part of the build process, so that might be a useful reference point to start with.  Perhaps I will see if I can get something similar to work for the ELKS kernel.

A less powerful Exomizer-like compressor can still be useful for userland programs which are run frequently.

Thank you!

tkchia, 2018-12-01:
Hello @mfld-fr ,

For your test setup, may I know if you were using both the "Build kernel as ROM-bootable" (`CONFIG_ROMCODE`) and "ROM file system" (`CONFIG_ROMFS_FS`) options during `make menuconfig`?

Thank you!

mfld-fr, 2018-12-02:
Hello @tkchia, yes, I use these two options to have all in ROM at boot, plus minimal configuration to make it runnable inside EMU86. See the whole configuration in `config_emu86`.

mfld-fr, 2019-01-19:
Note: ELKS master is not working any more in EMU86 (see https://github.com/mfld-fr/dev86/issues/22).

mfld-fr, 2019-01-19:
Just fixed by e8b31472c90e49db9b3684ce38e1ab53f16ed1c4.

tkchia, 2019-02-17:
Hello @mfld-fr ,

I am [toying with the possibility](https://gist.github.com/tkchia/3dc6e3a64d20ef51d4e183bff3105c16) of compressing the kernel data section, which will be going to RAM.

The next step is probably to write and add a decompressor stub to the kernel text section --- the stub will decompress the data section in place, before handing over to the actual kernel code.  One drawback of this approach is that the compressed data will be copied twice without any processing: the bootloader will copy the initial data section contents from ROM to RAM, then the kernel stub will need to copy the contents again to a "safe" higher address in RAM, before expanding it.

Thank you!

mfld-fr, 2019-02-17:
Hello @tkchia, amazing to see you contributing to this "little compression
challenge".

To avoid such drawback, I think the best solution is to place the
decompression routine in the setup code, not in the kernel one, and to have
setup loaded before the kernel.

Moreover, the kernel code segment is almost full today with all the
features selected in the configuration. Any new kernel routine should be
carefully considered in term of code size.

tkchia, 2019-02-17:
Hello @mfld-fr ,

> Hello @tkchia, amazing to see you contributing to this "little compression challenge".

Thanks!  At the moment I am just trying some things out, and trying to learn a bit more about Exomizer --- I have yet to come up with an actual working system.

> To avoid such drawback, I think the best solution is to place the decompression routine in the setup code, not in the kernel one, and to have setup loaded before the kernel.

I think that will be a nice solution too, and (as you mentioned) it will not allow more space in the kernel text segment for other things.  My main worry is that the `setup.S` code already has 3 separate ways to load the kernel, for 3 types of configurations ---

  * `CONFIG_ROMCODE`,
  * `! CONFIG_ROMCODE` with `REL_SYS`, and
  * `! CONFIG_ROMCODE` with `! REL_SYS`

--- and the decompression will need to be properly tested for at least these 3 cases.

There is another (smaller) problem: we will need to somehow signal to the bootloader whether the kernel text and/or data are compressed, and how.  This probably means adding some fields to the ELKS `a.out` header...

Thank you!

mfld-fr, 2019-02-17:
@tkchia : Yes, you are right, the setup module has became messy, and it should be reworked to ease maintenance. The relocation code is especially too much complex and I wonder why @lithoxs kept two different relocation methods. I also realized that situation while currently rewriting the boot loader for the Minix filesystem for the `switch_to_gcc` branch (see #225).

Also agree the need to amend the executable header format, to tell the loader that either the code or the data sections are compressed, and what is the compression method. This is not really a problem, as we have no more compatibility concern with the legacy `a.out` variants for the kernel, since it moved to `binutils-ia16` and ELKS-specific linker scripts.

jbruchon, 2019-02-17:
Just so everyone knows, I still read everything that goes by, even though I often have nothing to add.

mfld-fr, 2019-02-17:
@jbruchon : yeah, the temple guardian is keep watching us in case we would lack respect to our glorious ancestors :stuck_out_tongue:

jbruchon, 2019-02-17:
Hey, I'm just committed to the project. That was a git joke. I'll see myself out.

mfld-fr, 2019-02-17:
@jbruchon : a **git** joke => excellent :rofl: !

tkchia, 2019-02-22:
Hello @mfld-fr and @jbruchon,

I have [uploaded](https://gist.github.com/tkchia/3dc6e3a64d20ef51d4e183bff3105c16#file-exodecr86-s) a quick IA-16 implementation of the [Exomizer](https://bitbucket.org/magli143/exomizer/wiki/Home) v3.0.2 _backward_ decompression algorithm (this mostly corresponds to `exo_decrunch(`_in_`,` _out_`)` in the reference code).

I guess the next problem is how to fit everything together. 😐

Thank you!

mfld-fr, 2019-02-23:
@tkchia : the decompression routine looks small enough to be inserted into both `setup` and `system`. What are the compression ratio compared to the ones listed in https://github.com/mfld-fr/compress ?

tkchia, 2019-02-23:
Hello @mfld-fr,

> What are the compression ratio compared to the ones listed in https://github.com/mfld-fr/compress ?

Not bad, I think --- when I used Exomizer to compress the data section of the `system` binary I have, it reduced it to 66% of the original size (from 3,684 bytes to 2,426).

Thank you!

mfld-fr, 2019-02-24:
@tkchia : you did not use the same data segment as in my benchmark :smile: . Therefore I also tested Exomizer with my samples to have the same comparison base, and I updated my result page. Output is that Exomizer is today the best.

OK, so let me first finish my current work on the boot loader, then I will give a try to the Exomizer solution:
- add 3 fields to the 'a.out' header : compression type, code & data compressed sizes
- add compression of code & data segments in the build (just after the linking)
- insert your routine in the setup
- insert your routine in the kernel

One question : why do you use the backward option ?

tkchia, 2019-02-25:
Hello @mfld-fr ,

> One question : why do you use the backward option ?

Well, the real immediate reason, was that the reference code for the backward option (`rawdecrs/exodecr.c`) was simpler to understand. 😃

I also later figured that the backward option has another potential advantage.  _If_ we are decompressing in-place, then we can place the start of the compressed data buffer at a small distance<sup>(*)</sup> before the start of the decompressed data buffer.  (It should be possible, with a bit of special handling, to allow both the compressed and decompressed data to start at the same address, but I have not implemented this yet.)

<sup>(*) usually 2 bytes, i.e. at an offset of -2</sup>

Thank you!

tkchia, 2019-02-25:
Hello @mfld-fr,

A few things about `exomizer mem` which I found, and which bear some explanation:
  * The (uncompressed) input file is expected to start with 2 bytes giving the expected _load address_ of the uncompressed data.
  * The compressed output file will also start with a 2-byte load address, which is the original address adjusted with the "safety offset" (the distance I mentioned above).  The last 2 bytes of the output give the size of the original data.  in The raw compressed data stream lies between these 4 bytes.

Thank you!

ghaerr, 2021-04-06:
Compressed executables are now supported in ELKS, the design discussed in #908 and implemented in #912. 

The "exomizer" compressor/decompressor discussed above is used to implement fast, backwards in-place kernel-based decompression of text, fartext and data sections of a.out files using the supplemental header developed during medium model application support. 

While the decompression process is very fast and requires less than 1500 bytes in the kernel and only 16 bytes overhead per application, the compression process is run on the build machine and may take a second or so per binary while compressing the binary to the maximum extent, which seems to be typically in the 31-36% range.

To create a distribution floppy of compressed binaries, select "Support compressed executables" in the kernel -> filesystem menu of `make menuconfig`, and recompile the kernel. Then, use the following steps to separately compress all executables in /bin and make a distribution floppy. Presently, using this method frees up a 1.44M floppy from 24k to 335k free space available:
```
$ make
$ cd image
$ make compress   (runs elks-compress on target/bin/*)
$ make copyminix   (recreates fd1440.bin from target/* files)
```

After more testing and feedback, the build may automatically compress application binaries to save space.

The compression mechanism also works on FAT as well as ROM filesystems.

jbruchon, 2021-04-06:
Yay!

mfld-fr, 2021-04-06:
Very nice !

closed 2021-04-08 by ghaerr