ghaerr/elks #230: Add a license

czipperz, 2019-01-20:


mfld-fr, 2019-01-21:
Asking for the licensing status is actually a good question... Today ELKS project includes not only a specific 16 bits kernel, but also a bunch of applets, each of them having its own license (ash, dash, etc). In some way, it could be considered as a kind of "source distro".

As the kernel forked from an early version of Linux, it is under the same GPL license (as stated in https://github.com/jbruchon/elks/blob/master/elks/COPYING) , and I think the best option would be to upgrade to the latest GPL used by Linux mainstream.

For other parts, one should refer to the included licensing information in each subfolder of `/elkscmd`, as stated in https://github.com/jbruchon/elks/blob/master/elkscmd/COPYING.

czipperz, 2019-01-21:
Thanks. Was working on an air package and missed COPYING

closed 2019-01-21 by czipperz

jbruchon, 2019-01-24:
I fixed up that COPYING problem. The master directory and elks/elkscmd subdirectories now have LICENSE files and COPYING was renamed to LICENSE as needed.