ghaerr/elks #232: Number.c Build Fail

Szum123321, 2019-02-16:
When building every configuration of elks, I get this message:

flex -I -8 scan.l
mv lex.yy.c scan.c
expect 1 shift/reduce conflict
yacc -d bc.y
bc.y: warning: 1 conflict offset / reduction [-Wconflicts-sr]
mv y.tab.c bc.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/include -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o scan.o scan.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/include -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o util.o util.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/include -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o main.o main.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/include -I/home/szymon/Dokumenty/The_8086_8088_pc/elks/elks-0.2.1/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o number.o number.c
                   
02669                                           if bc_divide.off=0
***** relocation impossible.......................................^
02683                                           if bc_divide.off=0
***** relocation impossible.......................................^

00002 errors
00000 warnings
../Make.defs:109: recipe for target 'number.o' failed

mfld-fr, 2019-02-16:
Hello @Szum123321, are you building ELKS with the `dev86` package of your distribution, or the one that is built by `tools\build.sh` ?

Szum123321, 2019-02-16:
I'm building with dev86 built by tools/build.sh

mfld-fr, 2019-02-16:
Here is the successful build of `bc` in v0.2.1, to compare with your failed one:
```
flex -I -8 scan.l
mv lex.yy.c scan.c
expect 1 shift/reduce conflict
yacc -d bc.y
bc.y: avertissement: 1 conflit par décalage/réduction [-Wconflicts-sr]
mv y.tab.c bc.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o scan.o scan.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o util.o util.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o main.o main.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o number.o number.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o storage.o storage.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o load.o load.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o execute.o execute.c
bcc -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" -c -o bc.o bc.c
echo \"\" > math.h
bcc -c -0 -O -ansi -D_POSIX_SOURCE -I/mnt/data/home/mfld/advantech/elks-master/include -I/mnt/data/home/mfld/advantech/elks-master/elks/include -D__ELKS__ -DELKS_VERSION=\"0.2.0-prei86\" global.c
bcc -o fbc -s -ansi bc.o scan.o util.o main.o number.o storage.o load.o execute.o  global.o
bcc -o bc -s -ansi bc.o scan.o util.o main.o number.o storage.o load.o execute.o  global.o
```
Looks like there is a side effect from your build environment.

Szum123321, 2019-02-16:
Ok. thanks for help, but I just installed dev86 into system and everything works.

closed 2019-02-16 by Szum123321