ghaerr/elks #233: Build for 286 cpu fails

Szum123321, 2019-02-20:
When I set cpu to anything diffrent then 8086 I get:

gcc -E -traditional -I/home/szymon/Dokumenty/The_8086_8088_pc/elks-master/include -I/home/szymon/Dokumenty/The_8086_8088_pc/elks-master/elks/include -DELKS_VERSION_CODE=0x00020000 -DUTS_RELEASE=\"0.2.0\" -D__KERNEL__ -DUSE_IA16 -o boot/bootsect.s boot/bootsect.S
ia16-elf-as -mtune=i80286  -o boot/bootsect.o boot/bootsect.s
Assembler messages:
Fatal error: invalid -mtune= option: `i80286'
../../Makefile-rules:232: recipe for target 'boot/bootsect.o' failed

I've tested it many times and it seems like ia-elf-as doesn't like -mtune to be set to anything diffrent then 8086.
So I changed line in elks/Makefile-rules (CPU_AS = nothing)

ifeq ($(MK_CPU), 80286)
    CPU_AS	= -mtune=i80286
    CPU_CC	= -mtune=i80286
    CPU_LD	=
endif

And it compiles fine until I enable 286 protected mode. Then this appears:

make -C 286pmode pmode286
make[3]: Wejście do katalogu '/home/szymon/Dokumenty/The_8086_8088_pc/elks-master/elks/arch/i86/286pmode'
ia16-elf-as -3  -o crt0.o crt0.s
crt0.s: Assembler messages:
crt0.s:1: Error: no such instruction: `assembler bootstrap hooks. This is called by setup'
crt0.s:6: Error: no such instruction: `the next 3 int 3 instructions are part of the kernel_restart fix'
crt0.s:7: Error: no such instruction: `for protected mode.'
crt0.s:8: Error: operand size mismatch for `int'
crt0.s:9: Error: operand size mismatch for `int'
crt0.s:10: Error: operand size mismatch for `int'
crt0.s:12: Error: no such instruction: `setup passes these in registers (it used to use the stack,but that'
crt0.s:13: Error: no such instruction: `would clobber the ELKS kernel)'
crt0.s:16: Error: too many memory references for `mov'
crt0.s:17: Error: too many memory references for `mov'
crt0.s:18: Error: too many memory references for `add'
crt0.s:19: Error: too many memory references for `mov'
crt0.s:20: Error: too many memory references for `mov'
crt0.s:24: Error: too many memory references for `mov'
crt0.s:25: Error: expecting operand after ','; got nothing
crt0.s:34: Error: no such instruction: `break point if it returns'
crt0.s:36: Error: operand size mismatch for `int'
crt0.s:39: Error: too many memory references for `mov'
crt0.s:40: Error: too many memory references for `mov'
crt0.s:42: Error: too many memory references for `sub'
crt0.s:43: Error: too many memory references for `xor'
crt0.s:44: Error: expecting operand after ','; got nothing
crt0.s:49: Error: no such instruction: `this routine moves the ELKS kernel to where it won116be clobbered by our bss'
crt0.s:50: Error: junk at end of line, first unrecognized character is `('
crt0.s:51: Error: no such instruction: `that the kernel is sitting in now)'
crt0.s:53: Error: junk at end of line, first unrecognized character is `('
crt0.s:54: Error: no such instruction: `kernel and initializes it. no sense in keeping pmode support code around'
crt0.s:55: Error: no such instruction: `on a system that won116handle it.)'
crt0.s:59: Error: invalid character '!' in mnemonic
crt0.s:60: Error: no such instruction: `linear, they didn116fix the comment at the beginning of setup.S!'
crt0.s:61: Error: no such instruction: `this took me 3 tries to figure out. after figuring it out I remembered:'
crt0.s:62: Error: no such instruction: `ignore the comments, they can be terribly misleading. debug only code."'
crt0.s:63: Error: no such instruction: `if you change DEF_INITSEG,change the next line too.'
crt0.s:64: Error: expecting operand after ','; got nothing
crt0.s:65: Error: too many memory references for `mov'
crt0.s:66: Error: no such instruction: `seg es'
crt0.s:67: Error: invalid char '[' beginning operand 2 `[0x20]'
crt0.s:68: Error: too many memory references for `mov'
crt0.s:69: Error: expecting operand after ','; got nothing
crt0.s:71: Error: no such instruction: `i really hate this assembler. it doesn116have the sense to code contitional'
crt0.s:72: Error: no such instruction: `branch beyond 127 bytes. plus,it doesn116respond to the jmp opcode the'
crt0.s:73: Error: no such instruction: `same way as every other 80x86 assembler in existance.'
crt0.s:76: Error: no such instruction: `br _delete_self'
crt0.s:80: Error: too many memory references for `mov'
crt0.s:81: Error: expecting operand after ','; got nothing
crt0.s:82: Error: expecting operand after ','; got nothing
crt0.s:83: Error: too many memory references for `shr'
[... 60 lines cut ...]
crt0.s:182: Error: too many memory references for `add'
crt0.s:183: Error: too many memory references for `mov'
crt0.s:184: Error: too many memory references for `mov'
crt0.s:185: Error: too many memory references for `mov'
crt0.s:186: Error: expecting operand after ','; got nothing
crt0.s:187: Error: expecting operand after ','; got nothing
crt0.s:188: Error: expecting operand after ','; got nothing
crt0.s:198: Error: no such instruction: `whoops. forgot to clear the direction flag. _arch_init carefully clobbered'
crt0.s:199: Error: no such instruction: `my stack. the result was a return to the beginning. this took three days and'
crt0.s:200: Error: no such instruction: `a debugger constructed expressly for the purpose to fix. how annoying.:-'
crt0.s:205: Error: no such instruction: `what is the correct response when we can116find the kernel image?'
crt0.s:208: Error: no such instruction: `dw 0xfeeb'
crt0.s:212: Error: no such instruction: `no sense in hanging around here...'
crt0.s:214: Error: junk at end of line, first unrecognized character is `('
crt0.s:215: Error: no such instruction: `of code to somewhere else and use said code to relocate the kernel to'
crt0.s:216: Error: no such instruction: `def_initseg and transfer to it as if it were called directly from setup)'
crt0.s:218: Error: no such instruction: `db 0xeb,0xfe'
crt0.s:220: Error: no such instruction: `segment beginnings'
crt0.s:228: Error: unknown pseudo-op: `.zerow'
crt0.s:233: Error: junk at end of line, first unrecognized character is `7'
crt0.s:234: Error: junk at end of line, first unrecognized character is `5'
crt0.s:235: Error: no such instruction: `the BSS'
../../../Makefile-rules:232: recipe for target 'crt0.o' failed

I've tried may diffrent versions of both ai16-elf and bcc I always get the same result.

mfld-fr, 2019-02-20:
@Szum123321 Hello, thanks for the report. This is an expected yet unwanted regression after moving the kernel code from `dev86` to `gcc-ia16`, because the specific sources for the i286 protected mode have not been reworked yet (especially the assembly code that stills for `as86`).

Actually, we already had a discussion about the protected mode support in issue #207 where we proposed to drop it for simplification, but we are still waiting for @jbruchon agreement, because such decision is somehow 'strategic' for ELKS.

Could you please elaborate why you need the i286 protected mode in your use case ? That would help us to make the best choice.

mfld-fr, 2019-02-20:
About the first problem, there is a defect in `elks/Makefiles-rules`, as the `i80286` is not a valid value for the `-mtune` option:
```
mfld@home ~/advantech/elks-master $ ia16-elf-as --help
Usage: ia16-elf-as [option...] [asmfile...]
Options:
(...)
  -mtune=CPU              optimize for CPU, CPU is one of:
                           generic32, generic64, i8086, i186, i286, i386, i486,
                           i586, i686, pentium, pentiumpro, pentiumii,
(...)
```

mfld-fr, 2019-02-20:
First problem fixed in commit 8a95ba9.

Szum123321, 2019-02-20:
Firstly, thanks for quick response and reparing this problem
And secondly I want to build 80286 based computer to learn about assembly
and x86 architecture in general. I want this computer to run ELKS and to have more than 1MB of memory so because of that I'd like to have protected mode.

mfld-fr, 2019-02-21:
The first problem (defect in the build script) was solved. The second problem (protected mode not maintained) is now part of the more general question tracked by #234. Therefore closing this one.

closed 2019-02-21 by mfld-fr

tkchia, 2019-02-21:
Hello @Szum123321, hello @mfld-fr ,

By the way, I see that `ia16-elf-as` (which is mostly just mainline GNU `as`) uses option names `-mtune=i186`, `-mtune=i286`, etc. --- which differ from those of `ia16-elf-gcc` (whose options were, I think, crafted separately), which uses `-mtune=i80186` etc.

I think I will also add some option aliases to `ia16-elf-gcc` so that it can understand the `-mtune=i186` form of names.

Thank you!

mfld-fr, 2019-02-21:
@tkchia Yeah, it would be great if the options could be consistent, so aligning with the mainline looks like good to have.