ghaerr/elks #241: Configurable process heap + stack size

mfld-fr, 2019-03-13:
Postponed from #225:
- restore the heap + stack size parameter (previous ld86 -H option)
- rework the heap + stack space allocation in kernel `execve`

Current default heap + stack fixed to 16K.

tkchia, 2019-04-22:
Hello @mfld-fr,

I believe this can be done in the linker script: instead of saying `LONG (0)` for the chmem field, check whether a particular symbol is `DEFINE`'d and use that in an expression.  Then on the command line to GCC we can use `Wl,--defsym,`... to tell the linker script what to do.

Thank you!

mfld-fr, 2019-05-06:
@tkchia Good idea ! It would be more simple than a new command to hack the header after the linking.

tkchia, 2019-07-03:
Hello @mfld-fr,

I recently added a `-maout-total=`_t_ option to `gcc-ia16` to allow setting the heap + stack top address in the a.out header (with some co-operation from the linker scripts).

The option should both work when linking against Newlib (e.g. `-melks -mcmodel=small -maout-total=0xc000`) and against ELKS's `libc` (e.g. `-melks-libc -mcmodel=small -maout-total=0xc000`).

The `gcc-ia16` version currently downloaded by the ELKS build (https://github.com/jbruchon/elks/commit/0fb8d0a6b9807db2157a1e17d98c59a02c912bd6) should already include this option.

Thank you!

mfld-fr, 2019-07-04:
Hello @tkchia, thanks for your pro-active attitude ! Mmm... I am wondering about the right naming + semantic for this additional parameter when building a program... don't you think that in place of `aout-total`, it should be better to have a more intuitive name like "minimum size for heap + stack + env" for the user ?

tkchia, 2019-07-05:
Hello @mfld-fr,

> thanks for your pro-active attitude !

😃

> don't you think that in place of `aout-total`, it should be better to have a more intuitive name for the user to mean "minimum size for heap + stack + env" ?

Perhaps -- but at least for now, that `total` is the least confusing name I can find, for this particular a.out field.

From what I have found about the [Minix a.out format](https://github.com/gdevic/minix1/blob/master/commands/chmem.c), which ELKS uses, the header field is a total size for everything -- including static storage.  So in a way, it _does_ stand for the total size of the data segment.

I also considered naming it something like `-mchmem=`_t_, but decided it was too confusing -- the argument to Minix `chmem` excludes the static storage size, i.e. it is not the value that will go into the a.out header.  The name `__heap_top` [used by `ld86`](https://github.com/jbruchon/dev86/blob/master/ld/writex86.c) to label the top address also seems a bit too confusing to me.

Thank you!

mfld-fr, 2020-02-13:
Mmm... IMHO I think the developer should not have to deal with the size of the static data (.data + .bss), as this is handled by the compiler, but only to tell what minimum size he needs for the dynamic data (heap + stack), because it means for him. Up to the build tool to reject if the computed total data size is >64K. So I would prefer to change a bit the meaning of the `a.out` field : not an absolute value, but a relative one.

tkchia, 2020-02-15:
Hello @mfld-fr,

Thanks.  It seems though that this will break binary compatibility with Minix `a.out`, and perhaps with older ELKS binary programs.

Another way might be to expose to the user an option for adjusting the dynamic heap size --- something like `-mheap-len=`_x_ or `-mchmem=`_x_ --- _but_ still internally record the total data size (`.data` + `.bss` + heap) in the output `a.out`, using linker script magic.

I will try to come up with something.

Thank you!

mfld-fr, 2020-02-15:
Mmm... 'heap' for actually 'heap + stack + env' does not look a good naming... 'dyn-data' ?

ghaerr, 2020-02-15:
Can someone clarify what exactly is the data segment layout with the new compiler? In Documentation/memory.txt, the order shown is init-data, bss, heap, then stack. In the just-deleted bin-formats.txt, Riddoch mentions a new format (using the old bcc compiler) which has the stack below data+bss, and the heap above it.

With @mfld-fr's latest documentation update, it appears the a.out 'chmem' field is the total data segment size, which will be availble with a new compiler option for tuning (much needed). Correct? Would this be the field that the (missing) `chmem` program writes directly from its argument?

I looked at @Mellvik's recent report in #123 that `ls | sort | uniq | grep root` fails. I checked this, and turns out that everything works great, except for `sort` itself, which won't ever run. Remove `sort` and ELKS isn't running out of memory, it's entirely a `sort` problem:

```
# sort
sys_brk failed: len 27972 > endseg 19792
Not enough memory. Use chmem to allocate more: Out of memory
#
```

Where is `chmem`? I seemed to remember ELKS built it as an ELKS binary, which would be very useful for tuning programs while running ELKS. I looked in dev86 but couldn't find it there either. It would be nice to have this for fixing `sort`. 

As an aside, `bc` also fails to run, trying successively smaller brk requests, all failing.

tkchia, 2020-02-15:
Hello @mfld-fr,

> Mmm... 'heap' for actually 'heap + stack + env' does not look a good naming... 'dyn-data' ?

I am now thinking of `-maout-chmem=`_x_ or maybe `-maout-dynamic-data-size=`_x_ or thereabouts.  Whatever the name, it seems I will need to make some changes on the ELKS linker scripts first, to support such a thing.

Hello @ghaerr,

> In `Documentation/memory.txt`, the order shown is init-data, bss, heap, then stack. In the just-deleted `bin-formats.txt`, Riddoch mentions a new format (using the old bcc compiler) which has the stack below data+bss, and the heap above it.

As far as I can tell, ELKS currently uses the former order.  I do not think ELKS supports having the (default) stack below `.data` and `.bss` yet.  (This will mean that the static data storage does not start at offset 0, and there seems no clean way to specify that in the current `a.out` format.  But I may be wrong.)

>  it appears the a.out 'chmem' field is the total data segment size, which will be availble with a new compiler option for tuning (much needed). Correct? Would this be the field that the (missing) `chmem` program writes directly from its argument?

In more recent versions of `gcc-ia16` I have already added a `-maout-total=`_t_ option to control the maximum size of the data segment (including `.data`, `.bss`, heap, etc.).  The _t_ will be the value plugged directly into the `a.out` header.

As I wrote about, I am also thinking of adding a `-maout-chmem=`_x_ option to `gcc-ia16`, to control the maximum size of the _non-static_ part of the data segment.

> Where is `chmem`? I seemed to remember ELKS built it as an ELKS binary, which would be very useful for tuning programs while running ELKS. I looked in `dev86` but couldn't find it there either. It would be nice to have this for fixing `sort`.

Minix have had a [`chmem`](https://github.com/gdevic/minix1/blob/master/commands/chmem.c) for, I suppose, quite a long time.  Since ELKS uses practically the same executable format as Minix, perhaps we just need to ... port Minix `chmem` to ELKS?  It will probably be useful for tweaking the maximum data size for an existing binary program.

Thank you!

ghaerr, 2020-02-15:
> In more recent versions of `gcc-ia16` I have already added a `-maout-total=`_t_ option to control the maximum size of the data segment (including `.data`, `.bss`, heap, etc.). The _t_ will be the value plugged directly into the `a.out` header.
> 
> As I wrote above, I am also thinking of adding a `-maout-chmem=`_x_ option to `gcc-ia16`, to control the maximum size of the _non-static_ part of the data segment.

So both the -maout-total and -maout-chmem options both write to the same 'total data size' a.out field, but just allow the user to specify it either as a total or just heap+stack, with -maout-chmem reading the .data+.bss size and adding it to the command line value?

 I am trying to understand exactly which a.out fields are written by the compiler options, so I can write `chmem`. Sounds like we're only talking about the total data segment size field.

> Minix have had a [`chmem`](https://github.com/gdevic/minix1/blob/master/commands/chmem.c) for, I suppose, quite a long time. Since ELKS uses practically the same executable format as Minix, perhaps [...]

Thank you, I'll look at that.

tkchia, 2020-02-15:
Hello @ghaerr,

> So both the -maout-total and -maout-chmem options both write to the same 'total data size' a.out field, but just allow the user to specify it either as a total or just heap+stack, with -maout-chmem reading the .data+.bss size and adding it to the command line value?

Yes, that is so.  The "maximum total data size" field is here:
```
00000000  01 03 30 04 20 00 00 00  20 00 00 00 0d 00 00 00  |..0. ... .......|
00000010  00 00 00 00 00 00 00 00  00 50 00 00 00 00 00 00  |.........P......|
                                   ^^ ^^ ^^ ^^
00000020  ba 0d 00 b9 00 00 bb 01  00 b8 04 00 cd 80 bb 00  |................|
00000030  00 b8 01 00 cd 80 66 90  66 90 66 90 66 90 66 90  |......f.f.f.f.f.|
00000040  48 65 6c 6c 6f 2c 20 77  6f 72 6c 64 0a           |Hello, world.|
```

Thank you!

ghaerr, 2020-02-15:
@tkchia, thank you for your explanations.

What is the default heap+stack given to programs with `gcc-ia16`?  It is possible we should consider raising it, I am not sure whether the problem with `sort` and `bc` are a result of switching from bcc, which I think allocated 16k, but I'm not sure.

tkchia, 2020-02-15:
Hello @ghaerr,

`gcc-ia16` on default writes 0 in this field, which will cause the kernel to allow the data segment to grow by 16 KiB (see `elks/fs/exec.c`).  Thank you!

ghaerr, 2020-02-15:
Hi @tkchia,

I increased the kernel INIT_STACK (which covers both stack and heap) from 0x4000 (16k) to 0x8000 (32k) and both `sort` and `bc` run. Testing with 0x4800 (24k) isn't enough, sort still fails. 

Thanks for the reference to MINIX's `chmem`. I'll port that to ELKS if you add the -maout-chmem option (however named) to support specifying the 'chmem' field in the minix_exec_hdr.

Playing around a bit, it turns out that `sort` is badly broken, regardless of its data segment size. For instance, even after allocating 32k for all processes in the kernel, `sort` never actually writes anything to stdout. For instance, running `cd /; ls -l | sort` produces nothing.

@mfld-fr: The ELKS commands obviously need more testing, or perhaps seriously consider replacing some of them with the busyelks versions rather than trying to fix them. I have a growing list of ELKS application problems. Another thought would be changing INIT_STACK and allocating 32k to applications so they'd run by default rather than not, until either hearing about not enough physical memory problems from users, or looking at each command individually to set its chmem value.

mfld-fr, 2020-02-15:
@ghaerr : 16K or 32K default has big impact on a 640K PC system, so the need for the `chmem` tool to adjust dynamic data sizes. Some commands do not ever need heap space. Some others need large heaps.

tkchia, 2020-02-16:
Hello @mfld-fr, hello @ghaerr,

I have added a [patch](https://github.com/tkchia/gcc-ia16/commit/d17c71d30bd3b6911f484d030f2ab16101700755) for `gcc-ia16` to introduce a `-maout-chmem=`_x_ option to the GCC front-end.  I will update this ELKS tree to use the new GCC version, after running GCC through its regression test suite.

Thank you!

jbruchon, 2020-02-23:
Where does this issue currently stand?

tkchia, 2020-02-23:
Hello @jbruchon,

My recently merged pull request https://github.com/jbruchon/elks/pull/365 should have solved the "restore the heap + stack size parameter (previous `ld86 -H` option)" sub-issue.  I think the other sub-issues still need work.

Thank you!

ghaerr, 2020-06-06:
Completed by sequence of PR's ending with #649.

closed 2020-06-06 by ghaerr