ghaerr/elks #242: Nano-X running?

ghaerr, 2019-04-01:
Hello everyone!

I just happened by ELKS and am glad to see there's a lot going on lately!

I noticed that dev86/bcc has been recently replaced by gcc-ia16 and that the Nano-X make still uses bcc. Has anyone tried compiling it up? Is there any interest in making it work on the current ELKS?

Also, elks/nano-X/LICENSE states that there wasn't a license file provided, but actually there was, buried in elks/nano-X/src/misc/LICENSE, which should be moved to replace the former.

Let me know if you need any help; I'm reading that the gcc-ia16 compiler generates smaller code, and Microwindows v0.86 was the last that was made for ELKS (20 years ago!) because it was getting too close to going over 64K when more features were being planned.

Greg

mfld-fr, 2019-04-01:
Hello @ghaerr,

Thanks for the clue about the license file, it will be fixed as you instructed.

About Nano-X in the context of ELKS: it has not been moved to GCC-IA16 yet, not because lack of interest, but because priority is today to stabilize the kernel and the core utilities on that new build chain.

So any help to complete the move for remaining parts like Nano-X is welcome !

ghaerr, 2019-04-01:
Thanks @mfld-fr. I'll pull the project down and build GCC-IA16 on my OS X system, and test it under QEMU. Any known issues on OSX cross-compilation?

I think there may be a .s file or two for the Nano-X drivers. Is the calling convention identical from bcc to gcc, along with the register use (save si/di, scratch ax/bx/cx/dx)? Does gcc support #asm (probably not).

Is there a particular video card/setup which should be supported (EGA/VGA 4 planes, 16 colors)?

mfld-fr, 2019-04-02:
The GCC-IA16 calling convention is quite the same as BCC, except that you must save ES and don't use it for other things than segmentation.

GCC has a better support of the 'asm' statement in a 'C' context than BCC, see:
https://gcc.gnu.org/onlinedocs/gcc-6.5.0/gcc/Using-Assembly-Language-with-C.html#Using-Assembly-Language-with-C.

No special requirement about the video HW, testing can be performed in QEMU in first intent with emulated VGA. If somebody needs to run it on an antic CGA or EGA, then he would have to request it through an issue :smile:.

mfld-fr, 2019-04-02:
LICENSE issue fixed in 7d0920b

ghaerr, 2019-04-06:
I have rewritten the first of the Nano-X driver .c files with extensive #asm to a .s file in the new gcc-ia16 assembly format, and will test compilation and issue a patch. I will unlikely be able to test for reasons stated elsewhere. This should hopefully allow nano-X to be brought back into being built normally as part of the other ELKS applications.

ghaerr, 2019-04-08:
@mfld-fr: As mentioned above, I rewrote the .s driver files for Nano-X, which assemble correctly. The nano-X link is now failing very similarly to the ELKS kernel link discussed in another thread. Many present symbols are reported to be missing. I have attached the output. I have two questions: 1) is there a gcc/ld option that I need to specify some a linker .ld file? and 2) Can you look at the compiler options and let me know if these are what are required for user program links (I copied these from the kernel build for gcc-ia16)? I am a bit concerned about the relocation errors mentioning R_386_16 also.
[nanolinkerror.txt](https://github.com/jbruchon/elks/files/3056122/nanolinkerror.txt)

mfld-fr, 2019-04-09:
I suggest you use the same way to link the executable as within the `elkscmd` folder.

From `elkscmd/Make.defs`, with `$(OBJS)` as your object files:
```
LD=ia16-elf-ld
LDFLAGS=-T $(TOPDIR)/elks/elks-small.ld $(TOPDIR)/libc/crt0.o
LDLIBS='-(' $(TOPDIR)/libc/libc.a '-)'
$(LD) $(LDFLAGS) -o myprog $(OBJS) $(LDLIBS)
```
Especially the `$(LDLIBS)` with the 'reparse' options, because I remember to had some trouble when tuning the linking.

ghaerr, 2019-04-10:
Thanks, that worked, almost. There's still an undefined reference to a compiler helper function __ashlsi3. I'm guessing its arithmetic shift left something. Any idea where this is? They are usually in the gcc lib and added automatically by gcc, but we're linking directly using ld now.

mfld-fr, 2019-04-10:
According to GCC documentation, this is an integer arithmetic support routine: https://gcc.gnu.org/onlinedocs/gccint/Integer-library-routines.html

But I am wondering why we need such routine here ? 8086 provides the "SHL <reg>,cx" instruction, so it looks like something is missing in the GCC-IA16 backend... so let us ask the current maintainer first.

Hello @tkchia, what is your opinion about this issue ?

tkchia, 2019-04-10:
Hello @mfld-fr, hello @ghaerr,

`__ashlsi3` will be the routine for left-shifting a _4-byte integer_ (`SImode`) operand by a number of bits.  (That particular section of the GCC documentation [assumes](https://gcc.gnu.org/onlinedocs/gccint/Libgcc.html) that `SImode` = `int`.  But this is not true for the IA-16 port, where `int` is really the same as `short`, and the same as `HImode`.)

The 8086 of course has no direct way to left-shift a 32-bit operand by an arbitrary bit count, so yes, GCC may decide it needs to output a call to such a helper routine.

I suspect it might be possible to link in the `libgcc` routine simply by specifying `-lgcc` on the command line or in `LDLIBS`.  I have not really tried this myself, though.

Thank you!

ghaerr, 2019-04-10:
Hi @tkchia, -lgcc isn't working because we're having to use ld directly to link userland executables as discussed above, rather than gcc, which could possibly find the proper libgcc.a.

As a side note, it would be nice to link ELKS userland executables together using gcc and pass options to the linker, rather than using ld. I realize we need to use a special linker script, but all of this just adds to the incredible complexity required to do much of anything in ELKS.

In looking at the gcc build, I see 7 different libgcc.a's in elks/cross/build/gcc-build: there's gcc/regparmcall/any_i86, gcc/rtd/any_i86, gcc/rtd/size, gcc/rtd/any_i86/size, gcc/size, ia16-elf/any_i86/libgcc, all of which have a libgcc.a. Any idea which one I should link with? There appear to be tons of different versions, depending on whether the program is compiled with frame-pointer, regparmcall,... not sure what size/ is, etc.

mfld-fr, 2019-04-10:
@tkchia Thank you, I did not catch that `__ashlsi3` actually operates actually on a 32-bit integer, because of the 'int' in the GCC documentation (so a 16-bit integer for IA16). It is more clear for me now.

@ghaerr We don't use `libgcc.a` because 1) some routines are very long and consume the code segment space and 2) it helps us to spot the operations that are not directly supported by the IA16 architecture, so to be optimized in the code. In this case, one could wonder why you need to shift a 32-bit integer and if it could be coded another way ?

If not, I can add the missing routine to the ELKS 'libc', as other language support routines (like the 'idivmod' ones).

ghaerr, 2019-04-10:
@mfld-fr: thanks for the explanation.

I'm not sure I agree with the decision to not make libgcc.a available to userland programs, all of this just adds complexity and user programs may not be running next to the code segment space limit. To be straightforward, I don't see much point in requiring ia16-gcc to be used for userland executables unless medium model is supported from the start (think of all the .S/.s routines that might require yet another rewrite), instead of allowing bcc cross-compilation as an option like ELKS used to have. Requiring user programs to be rewritten to avoid compiler helper routines, or manually adding routines into an ELKS libc both seem like going the wrong direction with bringing ELKS back to life.

The __ashlsi3 routine is called by at least 9 separate functions in separate files (the loader stops reporting after that), so I can't recommend rewriting each of these functions. So given your libgcc.a decision I would prefer having the routine added to the ELKS libc.

mfld-fr, 2019-04-10:
@ghaerr That's your opinion :smile:

OK, let me add `__ashlsi3` to ELKS 'libc' to solve your current linking problem...

tkchia, 2019-04-10:
Hello @ghaerr, hello @mfld-fr,

> In looking at the gcc build, I see 7 different libgcc.a's in elks/cross/build/gcc-build: there's gcc/regparmcall/any_i86, gcc/rtd/any_i86, gcc/rtd/size, gcc/rtd/any_i86/size, gcc/size, ia16-elf/any_i86/libgcc, all of which have a libgcc.a. Any idea which one I should link with?

If you really need/want to link against `libgcc`, try putting something like `$(ia16-elf-gcc -print-libgcc-file-name` ...`)` ("..." = compiler options) into the command line, to get the correct `libgcc` path.

(On second thought, it might be simpler to just invoke `ia16-elf-gcc -nostdlib` ... --- and wrap linker options with `-Wl` as appropriate --- instead of running `ia16-elf-ld` directly.  Then writing `-lgcc` on the command line will automatically select the right `libgcc` multilib.)

> Requiring user programs to be rewritten to avoid compiler helper routines, or manually adding routines into an ELKS libc both seem like going the wrong direction with bringing ELKS back to life.

I for one hope in the near future to improve the ELKS support so that one just needs to write `ia16-elf-gcc -melks` ...

We are nowhere near there yet though, so please bear with us for the time being. 🐻

Thank you!

ghaerr, 2019-04-11:
@tkchia: Thanks, I didn't know about the -print-libgcc-file-name option. I have added the libgcc.a library to my nano-X demos link line, and the __ashlsi3 routine is pulled from it for the time being. So there's not an immediate need to add this function into elks/libc (and why add just one routine at a time anyways?)

@mfld-fr: I have finished porting Nano-X to the new compiler. Would you like me to submit a patch?
I have a patch for just elks/nano-X, which also includes fixes for the other non-committed items required for elks compilation on OS X.

There are two nano-X demos that won't link because of missing routines in the elks libc library, these are rand/srand and gmtime at the moment. I was suggesting earlier that the libc be used from dev86, as both these routines and all of elks/libc is already in there. I was going to add these three routines to elks/libc, but have decided to wait for you to either add dev86/libc to elks, or add each routine separately as needed to the partial version that was copied previously. The nano-X demo links can then be uncommented.

Let me know what you'd like to see in regards to a patch or two, and this thread can be closed.

mfld-fr, 2019-04-11:
@ghaerr Good news ! Yes, please submit a PR with your changes.

mfld-fr, 2019-04-11:
Let me explain why using the current `__ashlsi3` routine from `libgcc` is not the good option for optimized code. First, the relevant object file for the `-Os` option:
```
mfld@home ~/advantech/elks-master $ ia16-elf-objdump -a cross/bin/../lib/gcc/ia16-elf/6.3.0/size/libgcc.a
In archive cross/bin/../lib/gcc/ia16-elf/6.3.0/size/libgcc.a:
_ashlsi3.o:     file format elf32-i386
rw-r--r-- 1000/1000    628 Mar  9 22:20 2019 _ashlsi3.o
```
Then the object file dump:
```
mfld@home ~/advantech/elks-master $ ia16-elf-objdump -x _ashlsi3.o 
_ashlsi3.o:     file format elf32-i386
_ashlsi3.o
architecture: i386, flags 0x00000010:
HAS_SYMS
start address 0x00000000
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         0000004e  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .data         00000000  00000000  00000000  00000082  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  00000000  00000000  00000082  2**0
                  ALLOC
  3 .comment      00000012  00000000  00000000  00000082  2**0
                  CONTENTS, READONLY
SYMBOL TABLE:
00000000 l    d  .text	00000000 .text
00000000 l    d  .data	00000000 .data
00000000 l    d  .bss	00000000 .bss
00000000 l    d  .comment	00000000 .comment
00000000 g     F .text	0000004e .hidden __ashlsi3
```
Routine size: 78 bytes. Now, do you understand why we prefer to implement ourselves the routine (or to avoid using that 32-bit operation on a 16-bit architecture) ? Using the (AX,DX,CX) scratch registers, the carry flag and a few instructions make the things obviously quicker and smaller.

ELKS targets a real 8086 running at a few MHz with less than 1 MB of memory. A modern processor running 1000x faster with 1000x more memory is not our playground.

ghaerr, 2019-04-11:
@mfld-fr: Thanks for your explanation, and I hadn't realized that you had already written a replacement routine. I decided to link nano-X with the libgcc.a only to determine that __ashlsi3 was in fact the last undefined symbol preventing the port from being completed.

I've actually learned quite a bit about gcc from this exercise. And looked deeply into the libgcc source. After doing this, I have to say I agree wholeheartedly with you, I had expected libgcc to be fairly lightweight and simple. As it turns out, it is unbelievably complex, in almost every routine. For instance, the __ashlsi3 implementation has 32 entry points for unrolled optimized shifting, and there are many versions of it, hopelessly complicated unless one is very familiar with the compiler.

So - I'm coming around to your view on how various elks issues. I still think that you should pull the entire dev86/libc over at once now, which would allow more programs to run on elks (2 out of the 5 demos failed because of unimplemented libc routines present in dev86 and not in elks). This does not affect code bloat and makes elks more usable.

mfld-fr, 2019-04-12:
@ghaerr Yes, sure, porting BCC 'libc' as ELKS 'libc' is still in progress and will be progressively completed, so all the demos could run.

mfld-fr, 2019-04-12:
@ghaerr I merged your PR #251, reverted the MacOS related commit (because it breaks the compilation of the `build` tool in the kernel part), built all Nano-X successfully, added `demo2` to the FD1440 image, booted ELKS under QEMU, logged in as 'root' and ran `demo2` program. That program hangs without any change in the display.

ghaerr, 2019-04-12:
@mfld-fr: I figured the "dubious" portion of the MacOS commit might not work. The #include <sys/macros.h> needs to be protected with either a #if LINUX or #if !MACOSX, I didn't know how you'd like that handled. The sleep 1 in libc/Makefile will still need to be addressed for the OS X build.

I wasn't able to test the nano-X demos because I can't get QEMU to run on OS X, the version I installed doesn't support SDL and there is no X11 support. So I will debug this on Linux when I have time to delete some space from the gcc-ia16 build from which my VM ran out of space. I will be running QEMU
on Linux on Virtualbox on OS X, I hope those many translations aren't another problem for graphics display!

The nano-X initialization works by writing the VGA registers directly which change the display out of text mode. I'll write a much simpler demo that should allow for straightforward debugging of the startup. The only real code change in the port was rewriting some elks routines in the new asm format, so all of the nano-X commit should stay in, and the problem is likely minor, unless the C function calling sequence changed.

mfld-fr, 2019-04-12:
It looks like the program hangs at this point:
```
static int
VGA_open(PSD psd)
{
(...)
#if ELKS
	/* disallow console switching while in graphics mode*/
	ioctl(0, DCGET_GRAPH); /*continue if failed*/
#endif
```

ghaerr, 2019-04-12:
Both that ioctl and the one in VGA_close can be safely commented out, perhaps they are incompatible with the current kernel.

Looking at the code, I spoke incorrectly: if HWINT is not 1 then we use an int10 bios call to switch to graphics mode.

Out of curiosity, what debugger are you using to find the routine demo2 hangs in?

mfld-fr, 2019-04-12:
I am using the `printf` universal debugger :laughing:

The `ioctl` function is variadic, and I remember I had trouble when porting such ones to GCC-IA16. So I commented out the line, and now the screen switched to a blank white.

mfld-fr, 2019-04-12:
@ghaerr If VirtualBox is running on your Mac, why don't you boot ELKS directly in VirtualBox in another VM ?

ghaerr, 2019-04-12:
I had tried that in the past… set up VB as Linux 2.2, but couldn’t get a floppy attached to the full3
image, VB said it was in an unsupported format. I’m not that familiar with VB.
> @ghaerr <https://github.com/ghaerr> If VirtualBox is running on your Mac, why don't you boot ELKS directly in VirtualBox in another VM ?
>

ghaerr, 2019-04-12:
@mfld-fr: great idea on VB, got ELKS 0.2.0 running. Turns out VB needs image/full3 renamed to image/full3.dsk or it won't work.

However, now I'm back to what @Mullvik was saying... no tools on OS X to write demo2 to the full3 ext2 image. Also, I haven't been able to actually build an ELKS kernel on OS X yet, kernel link fails.

My plan for debugging demo2 is either to run it on the ELKS 0.2.0 kernel, or recompile it using bcc from last year's branch before gcc-ia16 and run that on the ELKS 0.3.0 image, as there are just too many variables otherwise. (BTW, where are the ELKS 0.3.0 images, seems like only the source and no images were released?)

ghaerr, 2019-04-23:
Hello @mfld-fr,

I am trying to debug Nano-X on the ia16-elf-gcc tree, and have found something very strange. As previously commented, I have nano-X running on the bcc tree. The only two changes are using the new compiler, and a single file that I translated into the new gcc as format. I am trying to trace through this code, so I instead disassembled the .o file using ia16-elf-objdump -d.

Notice how in the very first function, GETBYTE_FP, when I coded "mov 4(%bp),%bx" to get the functions first argument, the disassembler is reporting the code generated as "mov 4(%esi),%ebx"!!!
Instead of using %bp offset, %si is used!?!?

I have attached both files. Any help would be appreciated. Is this an objdump bug, or am I doing something terribly wrong in my .s file conversion?

Thanks, Greg
[elksutilasm.s.txt](https://github.com/jbruchon/elks/files/3105713/elksutilasm.s.txt)
[elksutilasm.disasm.txt](https://github.com/jbruchon/elks/files/3105714/elksutilasm.disasm.txt)

tkchia, 2019-04-23:
Hello @ghaerr,

It is (in a way) an issue with `ia16-elf-objdump` --- it defaults to assuming that the disassembled code is to be run in 32-bit mode.  To work around this, add the option `-Mi8086` to make it disassemble in 16-bit mode.

This should probably be fixed some day.

Thank you!

mfld-fr, 2019-04-26:
Just tried to run the `demo` program after #259 in QEMU, and got the same as in your screenshot [here](https://github.com/jbruchon/elks/pull/260#issuecomment-485586402). Looks like to run now on GCC-IA16. Nice work @ghaerr !

tkchia, 2019-05-09:
Hello @mfld-fr, hello @ghaerr,

In case you are interested: I just added a different assembly implementation of `__ashlsi3` to the GCC IA-16 toolchain's `libgcc` (https://github.com/tkchia/gcc-ia16/commit/21b95e80adb7a080b850a9a047505ebc653ca580).

It is a bit longer than your above looping implementation (https://github.com/jbruchon/elks/issues/242#issuecomment-482295822), but I believe it takes fewer clock cycles in the general case.

Thank you!

closed 2019-05-23 by ghaerr