ghaerr/elks #243: pre-built images?

toncho11, 2019-04-01:
Hi,

Sorry, but where can I download pre-built images of the latest release? It is only the source code I see in the downloads. 

Cheers,

mfld-fr, 2019-04-01:
Hello @toncho11,

There is no more pre-built image delivered with the release, because it would be hard to generate as many images as possible configurations. Just download the sources, run `tools/build.sh` to get the cross tools, and run `./build.sh` to get the image with the options you want.

If you cannot build the sources, we could upload an image for you, but you would have to specify your actual target where you intend to run ELKS.

toncho11, 2019-04-01:
Thanks for the quick reply! :)

What are the options for Intel 8086 on Amstrad 1640? 
http://www.old-computers.com/museum/computer.asp?st=1&c=19

Can I connect it to the Internet with serial null modem or serial to ethernet adapter? There is a TCP/IP stack? But then drivers might be needed or terminal program to dial up.

mfld-fr, 2019-04-01:
According to the technical information given in your link, the default options (suited for a standard IBM PC  / XT) should be fine, except the format of the image that should be FD720 (default is FD1440), and consequently, you would have to drop a few user programs for the whole to fit in a 720 K image.

toncho11, 2019-04-01:
Thanks.
You mean to put it on a diskette?

Can I build it and then flash it to a Compact flash card formatted with FAT32? Can I dd the resulting image to the card? Then I will boot from it using XT-IDE for 8086.

mfld-fr, 2019-04-02:
Yes, a 720K 5.25'' diskette. FAT support was introduced in 2015, but boot from FAT32 HDD is not yet supported (TODO tracked by #199).

toncho11, 2019-04-02:
OK, but I can do it with FAT16, right? Build image -> dd the image to FAT16 -> boot from FAT16 ?

mfld-fr, 2019-04-02:
Booting from FAT32 or FAT16 is actually the same : you should use a boot loader for the FATxx filesystem. Currently ELKS has only boot loaders for the Minix and the ROM filesystems. FATxx boot loader is planned for a next phase, unless a gentle contributor takes the action before :smile:...

toncho11, 2019-04-02:
OK, then what is the file system of ELKS by default? Ext2? This way I will format the compact flash card (the hard drive) with this type of filesystem, dd the ELKS image and boot from it directly. 

I do not have floppy disks currently, that is the problem.
The ELKS image is a bootable Linux image or an installer ELKS image?

mfld-fr, 2019-04-02:
The default file system is the legacy Minix version 1. I understood from the link you gave earlier that you had a system with floppy drives, but now I understand you 'only' have HD implemented on a compact flash card. The ELKS image can be bootable if you select the 'bootable' option ('on' by default).

One remark: no need to format the card if you 'dd' the image on it... as the image is already 'formatted' with the filesystem you select in the configuration options.

toncho11, 2019-04-02:
Yes, I have floppy drive, but not yet disks ...
Great! So just compile and dd.

closed 2019-04-05 by mfld-fr