ghaerr/elks #245: Mac OS X cross compilation build

ghaerr, 2019-04-02:
Hi,

Relating to my comments to getting nano-X running again in the "Nano-X running?" thread, I pulled down both gcc-ia16 and elks, and started the process of building on my Mac OS X system. After a bit, I got tools/build.sh to work, which required installating wget, and the gnu mpc, mpfr, and gmp libraries. The cross-compiler built and runs on OS X :)

Getting ELKS built is a problem, because OS X is running a case-insensitive file system.
This means that, say, setjmp.S and setjmp.s are the same file. One is overwritten by the next in the git clone. More of a problem is that elks/elks/Makefile-rules runs the C preprocessor "gcc -E" in order to handle #if's in .S files, which then writes the output to the same-named .s file, which overwrites the source on OS X. In addition, the make clean runs "rm -f" on the .s intermediate file, which also removes the original .S file.

I could fix all this, but wanted to ask what you feel a better source extension for c-preprocessing-required source files should be? Instead of .S, how about .ss, or perhaps .scpp or .Scpp? 

The fix would require that all .S files get renamed to .scpp, and the Makefiles modified with the renamed source files. Another option would be to keep .S and have the intermediate .s file be named something different, but this would be even stranger since bin/as accepts .s as default.

Finally, elks/libc/system/syscall.awk line 31 has an awk printf moving register %ax,
this needs to be changed to %%ax or awk fails with a bad printf.

I realize that ELKS was built for Linux, but given that the system is run and tested under QEMU, I thought that building for OS X would be useful. I regularly build multiple targets for Microwindows/Nano-X on OS X and would like to take a pass at this on ELKS.

Mellvik, 2019-04-02:
Gregory, 

The first thing to do is to change the setup of your Mac file system, to become case sensitive - as any ***x file system should be.

Then some practical advice: I gave up on ELKS and OSX almost immediately. You don’t need that pain when a Raspberry Pi will set you back only a few $ - and will make life a lot easier for you.

My 2c...

Helge

[ iPhone ]

mfld-fr, 2019-04-02:
@ghaerr : I think you are right about the file extensions. As '.s' is the *de facto* standard for ASM files in the GNU world, I would change '.S' into something else like '.sc' to mean "ASM to be CPP", and be able to build on case-insensitive systems.

ghaerr, 2019-04-03:
I found a better way to solve the problem, after seeing the the gcc-ia16 build itself uses .S files:

In elks/elks/Makefile-rules, the .S.s target that runs cpp -E and outputs a .s file should be eliminated and the .S.o target updated to run cpp -E and output a temp, say .sa file, then run the assembler on the .sa and delete the temp .sa file. In this way, the temp .s file doesn't overwrite the source .S. I could submit a patch for this, but it's only tested on OS X. Nonetheless, I recommend this change as there is no reason to keep the intermediate .s files around anyways, and is very confusing.

ghaerr, 2019-04-03:
A bigger problem with an OS X build is that there are several places where sed is used in the Makefile to create .h files from .c -> .s output, such as asm-offsets.h. The sed -e string operates differently on OS X than Linux, sadly, and the resulting .h file is empty, which breaks the build elsewhere. I tried debugging but given the nature of sh and sed \t, \\, and $ processing, could not get it to work on OS X without changing the string drastically. Not sure if this kind of stuff should be put in a tools/scripts area and a different interpreter used, or not, as I'm not a multiplatform sed expert.

I finally got the build to at least compile and create all the libraries, but the kernel link failed with undefined symbols, like kernel_ds, run_init_process, tcpdev_inetwrite and a host of others. Probably due to the sed and other incompatibility issues I hadn't found yet.

In desperation, I booted my Linux VM, rebuilt and downloaded the tools from scratch (tools/build.sh works fine), then used ./build.sh to build the kernel, which failed in syscall.awk on the printf(...%ax) statement mentioned above. Given that the kernel currently doesn't even compile on Linux, I think now may not be the time to try to get an OS X build working.

Mellvik, 2019-04-03:
Gregory, you're repeating the experiences I had a couple of years back, and I repeat my recommendation: It's not worth it. You are going to spend a lot of time fighting OSX instead of doing interesting (and useful) things with ELKS. Shortly you'll want to mount the minix filesystem on OSX. It takes you one minute or less on Linux, a week or more on OSX (or money to license some commercial product). Etc.

Hook up a usb floppy to a Raspberry Pi and you're home (almost)  free! It takes a little loooonger to (re)compile the whole shebang, but nothing that cannot be cured with a great (dbl) espresso.

Med vennlig hilsen/best regards,
Helge Skrivervik
myMAYDAY.com - 9201 7146

mfld-fr, 2019-04-04:
@ghaerr I am quite surprised about the Awk script. You reported there is a mistake in the format string, but I am wondering why it succeeds on my side ? Let me dig that point...

mfld-fr, 2019-04-04:
@ghaerr I would agree the temporary file, but there are both '.S' and '.s' in the sources. So I would prefer to replace '.S' by a new file extension, at the cost of modifying many files.

But first of all, how this case is manager by the ELKS big brother ?

ghaerr, 2019-04-04:
@mfld-fr Since both .S and .s are standard and well-known, and both have default rules in GNU Make, there is no good reason to change them. The only problem is when the same-base-named file is used in the same directory. This is only the case because of an ELKS-only rule in Makefile-rules which builds a .s file from a .S file, and a temporary one at that. The obvious solution is to change the Makefile-rules rule (3 line change) and the whole problem goes away. The IA16-GCC build works just fine and does not create a .s file from .S, instead uses temp name.

mfld-fr, 2019-04-04:
@ghaerr OK, that's fine then, so let us keep the standard, and let us patch as you propose.

mfld-fr, 2019-04-04:
Amazing: despite the theoretically missing '%' in the `syscall.awk` script, the file `syscall1.s` is correctly generated with my `awk`:
```
	printf ("%s:\n", funcname);
	printf ("\tmov $%d,%ax\n", callno);
	printf ("\tjmp _syscall_%d\n\n", $3);
```
produces (for the first function '_exit'):
```
_exit:
	mov $1,%ax
	jmp _syscall_1
```
Looks like the interpreter does not translate the `%ax` (because no corresponding second argument to `printf` ? because `%a` has no meaning and is ignored ?).

Anyway, fixed the mistake in commit 82b700d41059132c8d8491345e1c9cc9e88aa6ca.

mfld-fr, 2019-04-04:
Patched the kernel make rules to avoid intermediate '.s' files in commit 93ddb9a.

mfld-fr, 2019-04-04:
@Mellvik I agree your more efficient way, but look, we fixed two problems thanks to the try on Mac :laughing:

Mellvik, 2019-04-05:
I appreciate the effort - and result. Increasing the span of supported development platforms is good for the project in general. 👍

/helge

mfld-fr, 2019-04-05:
@ghaerr Thanks to your tries, we improved 2 points. Do you intend to continue on MacOS and try together to solve the `sed` issue, or do you switch to Linux for Nano-X build ?

ghaerr, 2019-04-05:
@mfld-fr Your patch looks great for building .S files, I haven't actually tried it yet on OS X but I'm sure that will work. Of course, I would like to get ELKS running on OS X. The sed issue is just one of a number of concerns... The configurator almost works but fails, along with other script issues. I did switch to a Linux VM that used tons of disk just building gcc-ia16, and now I'm out of disk space almost lol, and updating that is lots of work. I did start the Nano-X build on Linux with what little VM space I have left and it quickly compiled all .c files. Unfortunately, all #asm in .c files, as well as all .s files, are entirely incompatible, as first, the asm input to the compiler is completely different, and then, the new asm style is ATT format, with src & dst reversed and register notations different. So there's tons of work just getting user programs changed over, and I'm not sure anyone has any real interest in nano-X on ELKS. (yes, there needs to be less assembly in userland programs, but we'll talk about that later).

That said, having written a lot of nano-X, I would like too see it continue to run on ELKS.

I have since been studying ELKS and the old dev86 toolchain, including its libc, very extensively.

My current thoughts are that ELKS is far too complicated for it's own good. There's a libc in ELKS, there's two libc's in dev86, compiler language support in gcc-ia16 as well as dev86 libc. I think it needs a major slimming down.

As an example, After switching from OS X to Linux to get ELKS to build, I modified elks/nano-X/src/Makefile and reset the CC= and INCLUDE= to the updated compiler. There are no macros for this, each user program contributor/builder is completely on his own with tons of new required compiler switches. Nano-X v0.86 is a small-footprint window system designed to use a small set of system resources, and fits in less than a single 64k segment and runs in userland.

The make failed on the first file, src/devdraw.c, which #includes <stdio.h>.
The userland stdio.h, which lives in elks/libc/include, #includes <sys/types.h>.
Elks' libc is quite small, and includes just a bare minimum of routines, but still has a nice malloc
implementation based on sbrk, string functions, sys calls, and a few other things. That libc <sys/types.h> #includes __SYSINC__(types.h) which is included from the kernel include directory elks/include/linuxmt/types.h. That file immediately #includes <linuxmt/config.h> which, along with tons of other #includes, #includes <autoconf.h>. What the heck is autoconf.h doing in a <stdio.h> include??! Far too complex, and really no need for in a small system, unless we're trying to literally emulate all the bad things about modern day Linux and its zillions of options on an ancient 8086 based system.

I have since got this working with a simple second (or third) -I<dir>, but wanted to show how hard it is to quickly learn what could/should be something quite simple - the way stdio works on ELKS.

The beauty of ELKS, and also of dev86's libc and a number of other programs included in both, is that many of the programs are quite simple, few #includes, few #ifdefs, small footprint, compared to their modern-day bloated equivalents. People who come to ELKS need to be able to understand it in order to improve it. And of course there's no room left in 64k segments to do much more anyways. (I presume that one of the major reasons for the huge project of moving to gcc-ia16 is to get large-code segment support in user applications?)

That said, this whole topic has lots of other issues, and I've realized there are no easy answers. I would like to ask a few questions:

1) Did the old bcc-based system allow running bcc from ELKS, eliminating the need for cross-compilation of all programs?

2) Is dev86 to be completely thrown out? I could make a case for leaving libc in dev86, having separate directories for the .s files required by the new gcc-ia16 compiler, and removing libc complexities from ELKS and leaving ELKS to be just a kernel, rather than duplicating tons of stuff. We could also use some of the other non-compiler applications in dev86.

3) The bcc compiler option has been thrown out, is there a way to allow that back in, only for userland programs? It is a lot harder to port nano-X on a new ELKS system *and* use a new compiler requiring major .s file conversions that to port it first using the old compiler, test it, then port it to the new compiler. I didn't check deeply to see whether the entire executable format changed, or whether there were just incremental updates to a similar a.out.h.

Another thought would be to allow bcc for a bit longer backwards compatibility for userland programs ONLY. Thus, the ELKS tree could be split into kernel stuff ONLY, and userland programs built separately, possibly adding those user land programs into a forked dev86 tree so that other utilities would be available for running on ELKS itself.

4) I realize that ELKS is (or was) supposed to be Linux... but having tons of subdirectories, which are certainly required for a modern Linux kernel, just makes finding files and modifying many Makefiles all the more harder. The entire make process, the configuration management, hidden .config file, etc are vastly too complex for what could be a simpler system. For comparison, see the very cool swieros OS, which is a reimplementation of v6 UNIX on an emulated machine architecture: it's from-scratch C compiler doesn't support #defines nor #ifdefs, and the operating system itself, (full v6 UNIX), is in a single .c file. I am working on a 64-bit port to it along with a full Nano-X in a single .c file, forked on my homepage.

I better stop now, its hard to type out all the thoughts I have on this.

Don't take any of this the wrong way, I love these kind of projects! Just trying to see how I might be able to help more quickly :)

mfld-fr, 2019-04-05:
Wow, many points to think about 😄 ! Thank you for this long feedback. I
globally agree for the need of simplification. I just want to remind a few
facts about this project:

* Please forget BCC / DEV86. That companion project is in a very bad shape,
and we cannot be any more dependent on a "sinking boat". We now have a
modern / mainstream compiler expected for a long time, very good in
optimizing size, and it is worth to spend time in assembly rewrite.

* ELKS C library is a fork of the DEV86 one, ported to GCC-IA16. Many parts
were previously duplicated in ELKS and DEV86, the first step was to gather
the C library, the kernel and the user land in the same place. Now we just
started the next step to make the whole consistent and more friendly.

* ELKS contains "Linux". One goal is to be a roadmap to help understanding
its big brother. So it tries to find the good balance between simplicity
and fidelity. It implies preserving Linux core structure and configuration
/ build process.

* And last but not least, one ELKS target is real hardware. Emulation is
nice and antic machines are fun, but not as modern technology like a 8086
soft core in FPGA.

ghaerr, 2019-04-05:
@mfld-fr: Thank you very much for your clarifications!

> * Please forget BCC / DEV86. That companion project is in a very bad shape,
> and we cannot be any more dependent on a "sinking boat". We now have a
> modern / mainstream compiler expected for a long time, very good in
> optimizing size, and it is worth to spend time in assembly rewrite.

You didn’t answer my question: did discarding BCC mean that ELKS lost the
ability to compile programs from within it? Not having an option for an OS-hosted
compiler severely limits usability, and places ELKS more firmly into ultra-embedded
applications, rather than something one can tinker or develop with while running.

I can understand why BCC/AS86/LD86 should be replaced. But you’re saying that
DEV86 is a sinking boat, while then copying most of libc from it to ELKS. The copied ELKS libc
is apparently especially complicated by the decision to keep it with the kernel (something most other Linux kernels do not), as evidenced from the huge numbers of kernel files ultimately unnecessarily 
#included.

I read through the entire mailing list archives, but for some reason didn’t catch
the reason that DEV86 is a sinking ship, other than that the compiler was outdated;
for that I can agree. But there are other interesting programs in dev86. Why discard
it and complicate ELKS by adding in duplicated userland source that ends up being
over-complicated for both the userland libraries and the kernel? Why does the C library
and userland have to live with the kernel source?

What exactly is wrong with dev86 other than the compiler/assembler/linker?

jbruchon, 2019-04-06:
Dev86 does not work within ELKS and never has. The kernel has to statically link in code modules in the C library that it requires; Linux is no different, thus why it has built-ins that seem redundant, like printk instead of printf.

I think you will quickly realize why Dev86 is being abandoned as the chosen compiler for ELKS if you try to work on the code yourself, particularly the code for bcc. ELKS has had to work around its quirks for decades. It's not even an ANSI C compiler! Run with -ansi and it internally converts ANSI C prototypes to the very old-style K&R C prototypes so most ANSI C code can be ported without changing every single function prototype too. None of us is entirely sure where to even begin fixing some of the major missed optimization opportunities in bcc's code generation. Useful comments in the bcc code base are, to put it politely, sparse.

In short: get your hands dirty on the bcc code and you will understand in short order why it's now disfavored.

ghaerr, 2019-04-06:
Jody, thank you for your comments.

I misrepresented my question, I know Dev86/bcc doesn’t work “within” ELKS kernel, I was
asking whether bcc could be run from an ELKS shell after booting to a command prompt. That is,
did bcc allow users to compile userland programs while running on ELKS?

With regards to the libc directory, I’m not talking about printk(), an internal kernel routine,
I’m talking about elks/libc/*, which is a userland C library not linked to the kernel. Rlght?

I wasn’t trying to dig up the apparently old discussion of whether BCC is useful for ELKS kernel
compilations or not, I fully understand why gcc-ia16 is used for the kernel compilations.
My comments were really about the ability to compile programs “on ELKS” (not cross-compile), and was
also interested to see which non-bcc portions of dev86 were a sinking ship. I was also taking issue
to whether userland programs should be forced to use a different compiler at this time, thus causing major
porting issues, while the kernel stuff is being moved forward with a new compiler. I don’t
yet understand whether the syscall ABI or a.out format changed.

Thanks again,

Greg

jbruchon, 2019-04-06:
ELKS cannot host a remotely decent C compiler and never has. (I'm aware that a C compiler can be written for ELKS, but it would necessarily be primitive and crummy.) Dev86 doesn't support shared code or libraries at all, so what you're dealing "the C library" is not technically even a library in the modern understanding of the term. Dev86 statically linked .o files from the "C library" individually into the final object code, meaning that (other than some linkage to the kernel for syscalls and whatnot) all object code is fully standalone static code. With an appropriate syscall emulation layer, you could just as easily run an ELKS executable on top of DOS or OS/2 or Wine or Windows x86, provided 16-bit real-mode code is supported by the underlying system.

Because all code was statically linked by bcc and ld86 anyway, ELKS actually could use code bits from the C standard library directly in the kernel, but divorcing from Dev86 means dropping the C library cheat too, because bcc is no longer the compiler and reliance on its static linking is not possible. To make ELKS more independent of its compiler, that pile of code fragments had to come along with the kernel, unless of course someone unnecessarily rewrote it all from scratch.

jbruchon, 2019-04-06:
I keep hitting that stupid close button. They really ought to have a prompt on mobile for that.

mfld-fr, 2019-04-06:
Could we please go back to the initial topic of this issue, i.e. building ELKS on MacOS ? There is already issue #242 to discuss the Nano-X build under Linux, and mixing the threads makes the things hard to manage.

I understand @ghaerr realized after some tries that it would be a bigger task to build ELKS on MacOS than he initially figured out. @Mellvik also came to the same conclusion quite earlier. So the actual question here is to know if somebody could continue the trial on MacOS and kill the portability problems one after one with our help. If not, we could stop here, close this issue, and save our time for other tasks, like the suggested overall simplification.

ghaerr, 2019-04-06:
I will continue trying to get ELKS to build under OS X.

The current problem is the sed -e which creates asm-offsets.h from asm-offsets.s. 
For reasons I haven't figured out, the first -e option fails searching for a tab, any characters and a dollar sign to delete, from the start of a line and ends up creating an empty .h file. Any ideas?

mfld-fr, 2019-04-06:
Thanks @ghaerr to continue the effort !

I think `sed` to extract offsets is not really the right tool, and it should be simpler and more portable with `awk`. Let me try to replace that...

mfld-fr, 2019-04-06:
Replaced `sed` by `awk` for `asm-offsets` in commit 9e7a48e2.

ghaerr, 2019-04-06:
@mfld-fr Thanks, awk script works, and I think that's a better approach than sed :)

The elks kernel and libc compilations now work, but make fails on the kernel link. It appears that many symbols, like kernel_cs, can't be found. I looked for that symbol, and it's defined in init.c, so I namelisted init.o, and it is defined, and checked the mm.a archive, and init.o is included. I've attached all the details in link fail.txt. Pretty strange.

I did make one other change, which was to elks/arch/i86/tools/build.c, where I commented out the #include <sys/sysmacros.h>, because it is not on OS X. Build.c compiles, even though there is a comment "We NEED these macros". I don't know what build.c is used for.

Any ideas on the link problem?
[linkfail.txt](https://github.com/jbruchon/elks/files/3050933/linkfail.txt)

ghaerr, 2019-04-06:
Additional build problem: running "make clean" from the elks/ causes various .s and .S files to be deleted, because of the OS X case-insensitive filesystem. I haven't dug in to where in the Makefiles this occurs, but found the problem by "git status" showing the files were deleted, which I have attached.

[deleted.txt](https://github.com/jbruchon/elks/files/3050954/deleted.txt)

Mellvik, 2019-04-07:
Gregory,
From my point of view (and as pointed out before), this is a configuration problem of your own making, not an ELKS issue. MacOS is a UNIX system, case sensitivity. In files is its natural habotat so to speak. Thus, choosing case insensitive file systems will ALWAYS break things if you do software development and non-apple tools.

I strongly recommend you change this setting. 

/Mellvik

mfld-fr, 2019-04-08:
For the cleanup, I think it would be easy to fix. Let me try to patch it...

ghaerr, 2019-04-10:
@mfld-fr: another couple build problems:

1) The .S.o: fix you made for elks/Makefile-rules also needs to be made in  libc/Makefile.inc,
which runs gcc -E to a .tmp file then runs the assembler on it. 

2) I found a very strange make problem building libc/libc.a. It's a bit hard to describe, but if you look at each of the libc/*/Makefiles, they have a $(OBJS): $(SRCS) that is empty, then followed by a $(LIBC): $(OBJS), which then is supposed to run $(AR). The problem is, I think, that my OS X make or system is so fast, that the time on the $(LIBC) target is such that the $(AR) command doesn't get run for some of the libc/*/Makefiles. That is, it appears that the time stamp is the same for libc.a between adjoining $(SUBDIRS) makes in libc/Makefile that the AR command isn't being run.

I have fixed this by adding a "sleep 1" command in libc/Makefile under $(SUBDIRS): \ $(MAKE) -C $@ all.

To fix this, each of the libc/*/Makefiles should NOT have a $(LIBC): target which is used to modify (not create) libc/libc.a. I hope you understand what the problem is I am describing. I am thinking that a phony target should be specified, which ALWAYS runs the $(AR) command after the $(OBJS) are built.

mfld-fr, 2019-04-17:
Should I understand that after your latest changes in #255, anyone is now able to build the whole ELKS on OSX ?

ghaerr, 2019-04-17:
> 
> Should I understand that after your latest changes in #255 <https://github.com/jbruchon/elks/pull/255>, anyone is now able to build the whole ELKS on OSX ?
> 
> 
Yep, all of ELKS kernel and most of elkscmd can now be built on OS X.

The libc.a build bug is fixed with the sleep 1 kluge, but @tkchia says he’s got a better solution for it. This is all part of the over-complicated nature of present ELKS, and I have lots to say about that, as well as libc, coming shortly.

Elkscmd builds but prems still has a library perm.a problem linking (it’s not included in the ‘-( … )’, I think, so prems must be configured out or the build will fail.

“make image” fails and I am working on a solution to that. The image build uses
a linux-only mkfs.minix to create the filesystem, but ELKS should build a simple minix
mkfs instead. I have found source for that. That same replacement should eliminate
the need for “dd” for both creating the (say) 1.5Mb file as well as copying the boot
block later. ELKS is really only usable on Linux, and its got Linux’s vast overly-complicated
nature all around it!

The big problem (as previously stated by @Mellvik), is that OS X (or any other OS) can’t mount a minix filesystem. Using osxfuse, one can mount filesystems using user-mode code, but the minix filesystem
support is readonly. What is needed is simple user-mode programs, that both read and
write a minix filesystem, How easy would that be, for a user to read and write a disk image
without having to go through complicated os-specific mount/umount just to add or remove
a file from an ELKS disk image?

I have some possible solutions to that too, but have been sidetracked dealing with
other pressing issues, like for instance easily emulating the ELKS image just created,
so that a program can be tested. More on all this later.

mfld-fr, 2019-04-18:
Problem around building 'libc' solved by @tkchia in #256.

mfld-fr, 2019-04-19:
About the image build: I like the idea, it should not be hard to write a program that build a MINIX v1 disk image without the host kernel filesystem support. I already did that kind of exercise with `mkromfs` to build a ROM image (but the ROM FS is quite simple compared to MINIX FS).

ghaerr, 2019-04-19:
@mfld-fr: Any ideas about where some usable source code might be to start on building a userland minixfs copy, short of grabbing the kernel routines which seems quite complex due to VFS? I found ext2tools, but its still quite complicated.

I was thinking of using the kernel routines and just replacing the buffer system with direct disk i/o, but the multi-layered support for various filesystem types adds lots of complexity.

It almost seems that it could be simpler to run a user-mode ELKS kernel that ran a modified /sbin/init that mounted a filesystem and executed a copy, getting its arguments from the host shell.
What do you think of that idea?!?! I have a single .c file 8086 emulator (8086tiny)nrunning that boots ELKS in 0.1 seconds. It could be modified to pass arguments to /sbin/init through a modified boot, then essentially update it's own image along with the rest of the filesystem! :) This process could be bootstrapped from a single checked-in ELKS image.

tkchia, 2019-04-19:
Hello @ghaerr,

> Any ideas about where some usable source code might be to start on building a userland minixfs copy, short of grabbing the kernel routines which seems quite complex due to VFS?

I just found [`mfstool`](http://mfstool.sourceforge.net/) --- it might be worth a try.

Thank you!

ghaerr, 2019-04-20:
@tkchia: mfstool is exactly what I was looking for, thanks a lot!!! It has lots of little bugs and is a bit rusty, but I have it almost all working and tested now. This userland tool now allows creating, reading, writing and showing files on minix v1 and v2 filesystem images, along with mkfs options. I'm going to add bootblock handling and genfs/filesystem duplication capabilities, and this will allow replacement of mount, umount, dd, tar and the MAKEDEV script if desired, as well as being an absolute must-have on OS X.

mfld-fr, 2019-05-06:
Looks like that we are done for the OSX portability after the latest commits & merges. Thanks @ghaerr @tkchia @Mellvik @jbruchon for your participation to this issue. Let us now maintain that portability as it is really a good thing (at least because that OS is about 8-10 % of the desktop market). Closing this issue now.

closed 2019-05-06 by mfld-fr

ghaerr, 2019-05-06:
> Looks like that we are done for the OSX portability after the latest commits & merges. Thanks @ghaerr <https://github.com/ghaerr> @tkchia <https://github.com/tkchia> @Mellvik [...]
@mfld-fr: OSX will only build when the image branch is merged into master, as OSX does
not have MINIX mount capability. That was another reason mfs was written.

mfld-fr, 2019-05-06:
@ghaerr True, and that branch will be merged to master, but after the completion of #261.