ghaerr/elks #246: Swap file

toncho11, 2019-04-02:
ELKS does not support swap file, right?

mfld-fr, 2019-04-02:
Yes and no. It does not support memory swap, but still supports process swap with the `CONFIG_MEM_TABLE` allocator.

toncho11, 2019-04-02:
How do you enable this?

mfld-fr, 2019-04-02:
Please run `make menuconfig` and check the CONFIG_SWAP option. More details on the options here: https://github.com/jbruchon/elks/blob/master/elks/config.in

toncho11, 2019-04-02:
And this will make a 8086 computer use in practice more memory. If there is not enough memory the kernel will swap process/program automatically to give the memory for other processes that I am currently using? So it should be enabled by default?

jbruchon, 2019-04-02:
The swap feature has not had any attention or testing in over a decade. I'm not sure that you will see any benefits whatsoever from enabling it. If you do get it up and working correctly, there has to be a backing store to swap to, and if that backing store is going to be a disk, it's going to slow things down to a dead crawl every time a process is swapped in and out anyway. It should not be enabled by default. You are absolutely welcome and encouraged to play with it, but please don't expect it to be at all useful at this stage of development. There are more serious architectural issues that are currently being worked on which will open the door to a much more practical implementation of swap such as using EMS/XMS memory when available, especially on 286 and higher processors that can address more than 1MB.

Mellvik, 2019-04-02:
Thank you toncho11 for pointing  out the existence of swapping in the code.

I had all but forgotten, and I believe this may become really useful as the system becomes more complete and useable. After all this was the only way to make space for more processes than memory could hold in the old days, before virtual memory and interruptible instructions. The 8088/86 -to and including the 80286, belongs to this group of old-timers. No paging.

Like Jody points out, this is slow, but the alternative may be worse. For those of us who used the PDP-11s and DG Novas in the 70s, swapping was just fine (admittedly they had more I/O bandwidth than the early x86es). And even for old clunkers like my collection of 286 Compaq portables, the original hard disk (most of them, not all) died long time ago, now replaced by CF/SD-cards with IDE interfaces. These cards are fast enough to make for a useable system.

That said, having a XMS/EMS driver and RAM disk for ELKS would be great. I look forward to play with this one off these days.

—helge

toncho11, 2019-04-02:
Yes, but if the hard drive is a compact flash card, which might be as fast as the memory then it might actually be OK.

jbruchon, 2019-04-02:
There will be more CPU usage due to I/O not being as straightforward as a memory copy, especially in old x86 machines with PIO-only disk controllers that have to "babysit" the transfers to and from the disk interface. Not having DMA disk transfers really hurts. Also remember that CompactFlash is still flash memory. It is nowhere near as fast as RAM and writes will be worse than reads. Swapping active processes will wear on flash memory pretty hard too.

toncho11, 2019-04-02:
I see.
One can try: XT-IDE from Glitch works -> IDE to SATA converter -> SATA SSD.
Still the bandwith problem and the extra CPU usage.

closed 2019-04-05 by mfld-fr