ghaerr/elks #247: gcc-ia16 instructions

toncho11, 2019-04-05:
Hi,

It is not clear where to start with gcc-ia16. Can you add this to the description of the project?

1) Which branch to use? 
- tkchia/gcc-ia16 which has the advantage of pre-built images
or
- jbruchon/elks which has some fixes for ELKS

2) After that one should focus on the ./configure script? and find all the dependencies for it?

3) Is the gcc-ia16 the only dependency for ELKS to build?

mfld-fr, 2019-04-05:
Please read... the README :smile: !

toncho11, 2019-04-05:
I managed with the pre-built images of tkchia.

But when starting ./qemu.sh I encountered two problems:
- "Image format not specified ... probing ..."
- missing "vgabios.bin"

toncho11, 2019-04-05:
> Please read... the README  !

The README does not specify the version of gcc-ia16 to use.

mfld-fr, 2019-04-05:
It states:
```
How to build ?
--------------

To build ELKS, you need a cross build tool chain, mainly based on the latest
GCC-IA16 (DEV86 including BCC was used for previous versions, but has been
dropped because it was obsolete and no more actively maintained).

A script is provided to automatically download and build that tool chain:

  'tools/build.sh'
```

toncho11, 2019-04-05:
Did you read what I said?

mfld-fr, 2019-04-05:
Yes. Actually the `tools/build.sh` downloads the right version of the tools for you, including GCC-IA16. Just run it, and you will see...

toncho11, 2019-04-05:
I am confused. When I first started it, it said:

ERROR: GCC-IA16 not found. You must build the cross tools

so that is why I installed it my self from the pre-compiled images for ubuntu. Please tell me the correct version.

mfld-fr, 2019-04-05:
There are two `build.sh` scripts : the first (`tools/build.sh`) downloads the right versions of GCC-IA16, BINUTILS-IA16, etc., and build them locally in the `cross/` subfolder. Then the second (`build.sh`) uses them to build ELKS.

jbruchon, 2019-04-05:
The root build script should test for a working gcc-ia16 reachable from $PATH and attempt to locally build gcc-ia16 if there is no working system version (or if gcc-ia16 reports a version too old to build ELKS). The root script is nothing more than a one-stop easy build tool and it sounds like gcc-ia16 hasn't been incorporated yet.

mfld-fr, 2019-04-05:
@jbruchon Sure for the future, but not for now. We are still working closely with @tkchia to solve some issues raised during the current stabilization, and we must ensure we are using the proper version of GCC-IA16 that contains the changes he did for us (not the N-1, and not the N+1). Until a stable situation again, I won't spend any time to support any version I do not trust in because I did not test it.

jbruchon, 2019-04-05:
That doesn't negate what I'm saying. Users should be able to run the root build script and have everything done correctly for them. If that means adding a git clone of the correct gcc-ia16 repo and a git branch and build then so be it. ELKS failed in the past because it was a broken inaccessible mess. Priority one is making it easy to build and run. I created a build script in the root because the process of building wasn't terribly complex but also not trivial enough to be friendly to new users. It serves as both a very gentle introduction point and as documentation of the process required to bootstrap a working system. The vgabios problem is definitely a missing package which should be easy to install, but the arcane dependency on a very specific version of gcc-ia16 is not. The build script needs to hide this (hopefully temporary) oddity from users so they can get into the project and hopefully contribute to development in a meaningful way.

mfld-fr, 2019-04-05:
As the `tools/build.sh` script downloads the right versions of the tools, @toncho11 has to erase any previously installed `gcc-ia16` or `binutils-ia16` Ubuntu package, and use that script, without wondering about the needed versions. So closing this issue.

closed 2019-04-05 by mfld-fr