ghaerr/elks #248: Qemu not starting my image

toncho11, 2019-04-05:
I managed with the pre-compiled ubuntu packages of gcc-ia16 of tkchia. But I am still not sure it is the good branch to compile ELKS.

But when starting ./qemu.sh I encountered two problems:

"Image format not specified ... probing ..."
missing file "vgabios.bin"

mfld-fr, 2019-04-05:
`Image format` is a known warning in the test bench. You can ignore it.

QEMU needs `vgabios.bin` to emulate the VGA board. Looks like your local QEMU installation is missing a package. You could also tweak the QEMU options in `qemu.sh` to select another VGA emulation (see the "QEMU user manual" for more details).

toncho11, 2019-04-05:
It does not want to change. The option "romfile="does not work ...

mfld-fr, 2019-04-05:
I see no option `romfile` in the manual linked above.

QEMU has the following link for `vgabios.bin`:
`/usr/share/qemu/vgabios.bin -> ../seavgabios/vgabios-isavga.bin`

So it depends on the `seabios` package. Please check your installation. This problem is not related to ELKS but to your local environment, so closing this issue.

closed 2019-04-05 by mfld-fr