ghaerr/elks #249: Compilation

toncho11, 2019-04-06:
1) One needs to install these packages in advance:

sudo apt-get install gcc g++ git unzip texinfo make flex bison

I am not going to compile 20 times, just to find strange messages and then install packages.

2) The following script must be added to be executed in the tools/build.sh script:

/elks/cross/build/gcc-src/contrib/download_prerequisites

otherwise the gcc-ia16 can not be built automatically.

toncho11, 2019-04-06:
I have an error in the end (see below) and I have an empty fd1440.bin.

> ia16-elf-ld -T /home/toncho/elks/elks/elks-small.ld /home/toncho/elks/libc/crt0.o -o poweroff poweroff.o '-(' /home/toncho/elks/libc/libc.a '-)'
> make[2]: Leaving directory '/home/toncho/elks/elkscmd/sys_utils'
> make[1]: Leaving directory '/home/toncho/elks/elkscmd'
> make -C image all
> make[1]: Entering directory '/home/toncho/elks/image'
> sudo umount -d /home/toncho/elks/image/fd1440.bin
> umount: /home/toncho/elks/image/fd1440.bin: mountpoint not found
> Makefile:14: recipe for target 'all' failed
> make[1]: [all] Error 32 (ignored)
> sudo rm -rf /home/toncho/elks/image/fd1440.bin
> dd if=/dev/zero of=/home/toncho/elks/image/fd1440.bin bs=1024 count=1440
> 1440+0 records in
> 1440+0 records out
> 1474560 bytes (1.5 MB, 1.4 MiB) copied, 0.023106 s, 63.8 MB/s
> sudo mkfs.minix -1 -n14 -i360 /home/toncho/elks/image/fd1440.bin
> 384 inodes
> 1440 blocks
> Firstdatazone=16 (16)
> Zonesize=1024
> Maxsize=268966912
> 
> sudo rm -rf /home/toncho/elks/target
> mkdir -p /home/toncho/elks/target
> sudo mount  /home/toncho/elks/image/fd1440.bin /home/toncho/elks/target
> mount: /home/toncho/elks/target: mount failed: Unknown error -1
> Makefile:14: recipe for target 'all' failed
> make[1]: *** [all] Error 32
> make[1]: Leaving directory '/home/toncho/elks/image'
> Makefile:34: recipe for target 'all' failed
> make: *** [all] Error 2
> Build script has terminated with error 5

mfld-fr, 2019-04-06:
Did you activate the Minix filesystem support in your Linux kernel ?

mfld-fr, 2019-04-06:
I think you are not taking the easy way to setup your Ubuntu for development... instead of installing packages one by one, install the `build-essential` meta-package: https://help.ubuntu.com/community/InstallingCompilers

toncho11, 2019-04-06:
I am not sure build build-essential installs: bison, flex and texinfo.
Second problem still remains.

jbruchon, 2019-04-06:
Issue 2 will need to be fixed by us, so thank you for reporting it. Issue 1 is not an ELKS issue; installing basic development packages is distribution-specific and we can't realistically support that here. If you install everything required to build your own Linux kernel on your distribution, you should have everything needed to build ELKS.

Your host OS must support the MINIX filesystem to create disk images. You may need to modprobe minix to load that support. The disk images are MINIX-formatted and the build system loop-mounts and copies the files directly on the host. MINIX support is mandatory in your ELKS kernel as well, otherwise you can't boot from the image you just made.

toncho11, 2019-04-06:
Thank you. I see the problem and I know why I can not load the module and it is not compiled in the kernel either. It is because I use the Windows Subsystem for Linux. I can not recompile the kernel either.

Please copy/ paste the last lines of the build script that handles the images from your console. 
I will copy the folders to another Linux (with minix support) and just execute these to create the final image.

mfld-fr, 2019-04-06:
> It is because I use the Windows Subsystem for Linux.

WTF !!!

toncho11, 2019-04-06:
I do not have a computer with Linux at home and my computer is slow, so I decided to give it a try. And it worked pretty well - it did compile gcc-ia16 and elks. But it did fail on the mounting part. 
So if you can be so kind to just copy/paste the last part, I will be grateful. For example what happens after the mount?

mfld-fr, 2019-04-06:
Knowing that @toncho11 is actually working on WSL, I am not sure the issue 2 is "to be fixed by us".

The `download_prerequisites` references the following packages:
MPFR=mpfr-2.4.2
GMP=gmp-4.3.2
MPC=mpc-0.8.1
ISL=isl-0.15

The latest one is not needed because we configure GCC-IA16 with the `--without-isl` option.

For the first ones, they are normally installed by the distribution as dependencies of the host GCC package. Just verified it on my Gentoo. And looks like it's the same for Ubuntu: https://packages.ubuntu.com/xenial/gcc-5

mfld-fr, 2019-04-06:
The next steps after the mount are in the `image/Makefile`:
```
	sudo mount $(LOOP) $(TARGET_FILE) $(TARGET_MNT)
	(cd $(ELKSCMD_DIR)/rootfs_template; \
		tar cf - --exclude .keep *) | \
		(cd $(TARGET_MNT); sudo tar xpf - --no-same-owner)
	sudo find $(TARGET_MNT) \
		-type d ! -name root -execdir chmod go+rx '{}' \; -o \
		! -type d ! -name .profile -execdir chmod go+r '{}' \;
	(cd $(TARGET_MNT)/dev; sudo ./MAKEDEV; sudo rm MAKEDEV MAKESET)
	$(MAKE) -C $(ELKSCMD_DIR) install
	sudo bash -c "$(ELKSCMD_DIR)/tools/ver.pl $(ELKS_DIR)/Makefile-rules > $(TARGET_MNT)/etc/issue"
	sudo install $(ELKS_DIR)/arch/i86/boot/Image $(TARGET_MNT)/linux
	sudo umount -d $(TARGET_FILE)
	dd if=$(FD_BSECT) of=$(TARGET_FILE) bs=512 count=2 conv=notrunc
```

toncho11, 2019-04-06:
The console log from the build, not the makefile pls.

mfld-fr, 2019-04-06:
Here it is, with the default configuration:
```
sudo rm -rf /mnt/data/home/mfld/advantech/elks-master/image/fd1440.bin
dd if=/dev/zero of=/mnt/data/home/mfld/advantech/elks-master/image/fd1440.bin bs=1024 count=1440
1440+0 enregistrements lus
1440+0 enregistrements écrits
1474560 octets (1,5 MB, 1,4 MiB) copiés, 0,00520817 s, 283 MB/s
sudo mkfs.minix -1 -n14 -i360 /mnt/data/home/mfld/advantech/elks-master/image/fd1440.bin
384 inœuds
1440 blocs
Firstdatazone=16 (16)
Taille_zone=1024
Taille maximale=268966912

sudo rm -rf /mnt/data/home/mfld/advantech/elks-master/target
mkdir -p /mnt/data/home/mfld/advantech/elks-master/target
sudo mount  /mnt/data/home/mfld/advantech/elks-master/image/fd1440.bin /mnt/data/home/mfld/advantech/elks-master/target
(cd /mnt/data/home/mfld/advantech/elks-master/elkscmd/rootfs_template; \
	tar cf - --exclude .keep *) | \
	(cd /mnt/data/home/mfld/advantech/elks-master/target; sudo tar xpf - --no-same-owner)
sudo find /mnt/data/home/mfld/advantech/elks-master/target \
	-type d ! -name root -execdir chmod go+rx '{}' \; -o \
	! -type d ! -name .profile -execdir chmod go+r '{}' \;
(cd /mnt/data/home/mfld/advantech/elks-master/target/dev; sudo ./MAKEDEV; sudo rm MAKEDEV MAKESET)
make -C /mnt/data/home/mfld/advantech/elks-master/elkscmd install
make[2] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd »
for i in bootblocks inet  ash bc disk_utils file_utils levee ktcp minix1 minix2 minix3 mtools misc_utils prems sh_utils sys_utils; do make -C $i install || exit 1; done
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/bootblocks »
make[3]: rien à faire pour « install ».
make[3] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/bootblocks »
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet »
for i in  httpd nettools telnet telnetd tinyirc urlget; do make -C $i install || exit 1; done
make[4] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/httpd »
sudo install httpd /mnt/data/home/mfld/advantech/elks-master/target/bin
sudo mkdir -p /mnt/data/home/mfld/advantech/elks-master/target/var/www
sudo install sample_index.html /mnt/data/home/mfld/advantech/elks-master/target/var/www/index.html
make[4] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/httpd »
make[4] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/nettools »
sudo install netstat nslookup /mnt/data/home/mfld/advantech/elks-master/target/bin
make[4] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/nettools »
make[4] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/telnet »
sudo install telnet /mnt/data/home/mfld/advantech/elks-master/target/bin
make[4] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/telnet »
make[4] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/telnetd »
sudo install telnetd /mnt/data/home/mfld/advantech/elks-master/target/bin
make[4] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/telnetd »
make[4] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/tinyirc »
sudo install tinyirc /mnt/data/home/mfld/advantech/elks-master/target/bin
make[4] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/tinyirc »
make[4] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/urlget »
sudo install urlget /mnt/data/home/mfld/advantech/elks-master/target/bin
sudo ln -s urlget /mnt/data/home/mfld/advantech/elks-master/target/bin/ftpget
make[4] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet/urlget »
make[3] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/inet »
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/ash »
sudo install ash /mnt/data/home/mfld/advantech/elks-master/target/bin
sudo ln -sf ash /mnt/data/home/mfld/advantech/elks-master/target/bin/sh
make[3] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/ash »
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/bc »
sudo install bc /mnt/data/home/mfld/advantech/elks-master/target/bin
make[3] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/bc »
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/disk_utils »
sudo install fdisk ramdisk mkfs fsck partype /mnt/data/home/mfld/advantech/elks-master/target/sbin
make[3] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/disk_utils »
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/file_utils »
sudo install cat chgrp chmod chown cmp cp dd grep l ln ls mkdir mkfifo mknod more mv rm rmdir sync touch  /mnt/data/home/mfld/advantech/elks-master/target/bin
make[3] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/file_utils »
make[3] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/elkscmd/levee »
sudo install lev /mnt/data/home/mfld/advantech/elks-master/target/bin/vi
[... cut ...]
sudo install /mnt/data/home/mfld/advantech/elks-master/elks/arch/i86/boot/Image /mnt/data/home/mfld/advantech/elks-master/target/linux
sudo umount -d /mnt/data/home/mfld/advantech/elks-master/image/fd1440.bin
dd if=/mnt/data/home/mfld/advantech/elks-master/elkscmd/bootblocks/minix.bin of=/mnt/data/home/mfld/advantech/elks-master/image/fd1440.bin bs=512 count=2 conv=notrunc
1+1 enregistrements lus
1+1 enregistrements écrits
965 octets copiés, 0,000210258 s, 4,6 MB/s
make[1] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/image »
```

toncho11, 2019-04-06:
Thanks ... lucky me that I speak French :)
For building gcc you need the 3:

MPFR=mpfr-2.4.2
GMP=gmp-4.3.2
MPC=mpc-0.8.1

as source I think in the gcc build folder. And if you have newer versions then building gcc might fail.

mfld-fr, 2019-04-06:
Using newer versions without any problem...
```
[IP-] [  ] dev-libs/mpfr-3.1.6:0/4
[IP-] [  ] dev-libs/gmp-6.1.2:0/10.4
[IP-] [  ] dev-libs/mpc-1.0.3:0
```

closed 2019-04-08 by mfld-fr

mfld-fr, 2019-05-06:
See also #245 about building on MacOS. Maybe the latest changes could help on WSL too, as the image build does not require the MINIX support from the host OS anymore.

mfld-fr, 2020-02-10:
Confirmed through the CI setup that even with the `build-essential` package, one needs to install 'flex', 'bison', 'texfile' and 'libcurses-dev' on Ubuntu 18.04. See https://github.com/elks-org/elks/blob/master/.github/workflows/main.yml. README updated in commit bfb0496f369fdcc022c62556a29e692191a5e80c.

For the other GCC dependencies, automatic download as suggested has been implemented in commit 21ca7a69f692cd2722d36386af94355146538989.