ghaerr/elks #252: Integrating elks-libc installation more closely with gcc-ia16 toolchain?

tkchia, 2019-04-14:
Hello all,

@ghaerr has [complained](https://github.com/jbruchon/elks/issues/242#issuecomment-481756600) --- and I quite agree --- that compiling a program for ELKS using the `gcc-ia16` toolchain is more complicated than it needs to be.

Yesterday I had made some tweaks to the [Newlib](https://github.com/tkchia/newlib-ia16/commit/14af178d4403d018db6a2bba96226568890ee3c4) and [GCC](https://github.com/tkchia/gcc-ia16/commit/dbbadd1b237963a87c58d2dab97f4d627cfac4a4) components of the toolchain, so that there is now a semi-working setup for compiling ELKS programs by just saying `ia16-elf-gcc -melks` ...&nbsp;.  However, this setup is based on Newlib, which [is quite bloated](https://github.com/jbruchon/elks/issues/166#issuecomment-341820826) and may not be the best choice of C runtime.

Therefore I hope to find a way to somehow integrate ELKS's own `libc` into the `gcc-ia16` toolchain installation.  Ideally, one can easily ask to link a program against `elks-libc` as an alternative to Newlib, or even as a default replacement for Newlib.

For secondary goals, I hope
  * to add multilib support to `elks-libc` (automatically build and use different versions of `elks-libc` for different function call conventions, etc.), which Newlib already has; and
  * to incorporate the improved compiler helper routines (`__ashlsi3`, `__divsi3`, etc.) that are currently in `elks-libc`, into the GCC source tree itself.

Thank you!

mfld-fr, 2019-04-19:
Hello @tkchia !

I understand that adding 'multilib' support to ELKS libc is nice exercise, but I still wonder what you could do more after ? For kernel side, the default calling convention is assumed in low level routines, and supporting more would make the code bigger. For user side, I agree building a program could be simpler with only GCC as the front-end and no more direct LD invocations, but is it worth to make so many changes just because of that ?

Except that point I see no complexity in the current build, and I do not see any problem to implement the needed C helper routines into the ELKS 'libc', as it is fundamentally the "C runtime library" ? Am I missing something ?

tkchia, 2019-04-19:
Hello @mfld-fr,

> I understand that adding 'multilib' support to ELKS 'libc' is a nice exercise, but I still wonder what you could do more after ? For kernel side, the default calling convention is assumed in low level routines, and supporting more would make the code bigger.

As far as I can tell, there should be no need to modify the _kernel_ in order to support other calling conventions in userland.

OK, some changes are needed to `elks/arch/i86/kernel/syscall.dat` to output the proper code for the `open`, `fcntl`, and `ioctl`, which should be variadic functions in userland (and may need to accept and pop their arguments differently), but this does not really affect the kernel itself.

Outside of that: system calls already get their parameters through registers anyway.  And I see that signals go through the `_syscall_signal` wrapper, which can be made to transform the kernel's parameter convention to some other convention.

> I do not see any problem to implement the needed C helper routines into the ELKS 'libc', as it is fundamentally the "C runtime library".

Thanks.  Well, I am actually hoping to add the improvements in the other direction.  In particular I think the `__ashlsi3` routine you added to `elks-libc` can be fruitfully used in MS-DOS too, so I hope to add it from `elks-libc` to `libgcc`.

Thank you!

ghaerr, 2019-04-19:
The new changes to building libc.a break the OSX build on "cp -u ..." somewhere, -u is an unknown option.

I really don't understand why a libc is included in ELKS, and would like to strongly suggest another approach: putting the C library and other development tools, like perhaps the mkfs.minix and others I was going to write, into a separate, OUTSIDE, say gcc86 source tree (like dev86, but for gcc-ia16 and other improvements). I can think of the following reasons for this:

1) The kernel doesn't and will never link with libc, what is libc doing in the ELKS build?
(Yes, I know elkscmd/).

2) If it is desired to add/share some kernel low-level routines into a libc, that could easily be done by essentially updating from elks/arch/i18/lib as often as desired, or the opposite direction.

3) Adding libc and its latest build scripts is definitely going in the OPPOSITE direction of making ELKS easier to understand and use - this is a serious point to get others, including myself, back into this project, or at least minimizing the time to get anything done.

4) The gcc-ia16 toolchain, an updated libc, and other of @tkchia's great contributions, should not be limited to ELKS, and require an ELKS build in order to be used. For instance, I am contemplating other interesting i86 projects that won't have an ELKS kernel but will want to use various tools developed by this team. Where will these things live?

5) There needs to be an outside repository, not ELKS an not gcc-i186, that can hold various related development tools (like object disassemblers, possible small, very-easy-to-use emulators (I am working on one) etc.

6) Building an internal ELKS libc runs the ongoing risks of tying the C library header files to ELKS, they touch TOO MANY already (see my previous comments about autoconf.h being included in a stdio.h #include chain). This needs to be simplified - not added to.

The biggest issues for me are making ELKS simpler, and having tools access. For instance, the change to gcc-ia16 was made in-between releases, not at a release point. During that change, we lost some of the tools (objdump86, disasm, etc) that I use to compare objects, debug etc. I ended up forking elks and creating a 0.2.2-bcc branch that runs on OSX just to be able to compare "before" and "after" easily, which was required to get nano-X running again. There is talk that gcc-ia16 is nicer, but there is no easy way to actually look at the data, with the same versions of programs being compiled. It may be seen as no big deal, but as a user-mode maintainer, I'm stuck with not being able to compile either new or old versions, without forking and lots of extra work. </end of diatribe>

With an outside libc, I hope to contribute some tools that would allow nano-X graphical windowing systems, immediate ability to emulate ELKS and userland programs without installing a cross-8086-emulator, and being able to build these applications WITHOUT having to put them in the ELKS tree, as is currently required. The outside gcc86 could have an ELKS option that would map the system calls. Dev86 accomplished this and its been thrown away.

I have many more thoughts that I'll leave to a different thread.

tkchia, 2019-04-19:
Hello @ghaerr,

Incidentally --- as I mentioned above --- the recent commits for my `gcc-ia16` toolchain now have fully integrated support for building ELKS programs, if you say `ia16-elf-gcc -melks -mcmodel=small` etc.

The setup uses the Newlib library rather than `elks-libc`.  Binary-wise, this means it probably produces more bloated code than it needs to.  But source-wise, `newlib-ia16` lives in its own source tree separate from ELKS's, and I suppose that has its own advantages, as you said.

Thank you!

closed 2019-06-23 by tkchia

mfld-fr, 2019-06-24:
Thank you @tkchia for this big & nice contribution.

tkchia, 2019-06-24:
Hello @mfld-fr,

Thank you very much! :-)