ghaerr/elks #261: Tools for build system

ghaerr, 2019-04-26:
I have a working version of a new tool, mfs, a user-mode minix file system utility that allows mkfs, cp, ls, mkdir, mknod, rm, rmdir, set boot block, and other operations to create and manipulate a Minix V1 system image.

This tool is required on OS X to build bootable ELKS images, but for a variety of reasons which I can detail later, may or may not want to be used as the main mechanism for creating ELKS images. It certainly makes the ELKS build process more portable and eliminates the need for dd, minix.mkfs, mount/umount, sudo, and others.

I have a couple questions:

1) Where should this tool live, and when be built? As part of the cross-compiler toolchain (that is, built before building the rest of ELKS but before elkscmd's "make image")? Since dev86 was thrown out and not replaced, there isn't an external place to store tools, and the gcc-ia16 build already occupies elks/cross and is tremendously complicated. Libc was removed from an external devkit and placed in elks/libc, so should we use elks/tools for this? Mfs is quite convenient and ultimately should be part of an external dev86 environment, along with other tools, a cross-compiler make facility (supporting either bcc for the old 0.2.2 branch testing and gcc-ia16) and libc), IMHO.

2) We need a mechanism in elks/image/Makefile for testing the OS environment and using mfs to create the ELKS image.

mfld-fr, 2019-04-27:
I fully agree that using such tool should **entirely** replace the current image generation. Now, 2 cases:
- either the tool can be sourced and built "as is", or needs only a few patched lines, so it can be added as another third-party in 'tools/',
- or it has to be ported with many changes, so one should place the modified sources (including the license) in 'elks/tools/'.

tkchia, 2019-04-27:
Hello @ghaerr, hello @mfld-fr,

A compromise might be to include it as a Git "submodule" --- such that it has its own Git tree, but this main ELKS tree can refer to it, and arrange to check it out when needed.

The GCC IA-16 toolchain is itself already spread out across several Git trees --- `build-ia16`, `binutils-ia16`, `gcc-ia16`, and `newlib-ia16` --- even in Mr. Jenner's upstream version.  (The setup does not use Git submodules, though it probably can.)  So I think there should be no problem if you adopt a similar arrangement for the Minix filesystem tools and other ELKS tools.

Thank you!

jbruchon, 2019-04-27:
Submodules are bad. https://codingkilledthecat.wordpress.com/2012/04/28/why-your-company-shouldnt-use-git-submodules/  Having the compiler strongly tied yet separate is already bad enough. 

This tool would eliminate the need to be root to build images. As long as it's under an appropriate free software license, stuff it in a directory in the root of the project and have the build script build it, then have the image building process check for it and use it if it was built successfully.

ghaerr, 2019-04-30:
The mfs tool is entirely functional for what is needed to replace the current sudo/mount method of producing MINIX filesystems, as well as adding boot blocks.

At this point, it depends on how you want to proceed to create disk images in a new way. As mentioned, the current method uses a single "make install" to copy all elkscmd binaries to a mounted image. Spending lots of time creating an "mfs genfs" capability doesn't change the need to rethink how and which applications should be installed, and whether a new approach to creating a few different ELKS images is desired. 

The MAKEDEV script can be pretty easily converted to run "mfs mknod" commands instead.

If it is desired to just replace sudo/mount with mfs and create the exact same bootable ELKS image, then just creating a list of applications to be installed in /bin and /usr/bin is all that's needed.

ghaerr, 2019-04-30:
In looking a bit further into the ELKS application installation process, elkscmd/Makefile::install calls the Makefile::install entry point for each of the applications. Each of these applications are installed using "sudo install ...", if the config-specified CONFIG_APP_xx option is set.

A reasonable option to keep the installation method within each application's Makefile would be to create a replacement "install" shell script which effectively runs "mfs $(TARGET_FILE) cp ...". This approach would keep the .config file to specify which applications are installed, and allow a single "make install" from image/Makefile to install apps. While this approach keeps things the same, it doesn't allow for easily creating different ELKS distributions disks, which would be nice considering most of the current ELKS images take up lots of space with programs very infrequently, if ever, used.

mfld-fr, 2019-04-30:
Today ELKS supports not only the MINIX filesystem, but also the ROM one. Full FAT support is also requested for interoperability. Application installation script should not be coupled to the target filesystem, otherwise one would have to write & modify that script for each supported filesystem.

Common practice in Makefiles is to specify the root installation folder with the `PREFIX` variable (or with the `--prefix` option in GNU Autotools), then up to the application installation script to create the subfolders and copy the files to that root folder. One should not change this well known workflow.

Some projects (like Debian and BuildRoot) use the [fakeroot](https://wiki.debian.org/FakeRoot) wrapper to deal with the `root` user in a generic & elegant manner, without patching files one by one in the target filesystem (so without the coupling mentioned above).

Therefore I think the nice `mfs` tool should be only used in the image build script to copy the staging root folder (`target/`) from the host filesystem to the target one (for the MINIX case).

About the distributable disk images, I already responded to that question [here](https://github.com/jbruchon/elks/issues/243#issuecomment-478542219). We could consider to configure a CI robot for configuring / building / testing / deploying such images for different configurations, but that's again another story.

For now, we try to build by default the biggest kernel and the maximum number of applications, just to detect more easily any broken dependency. Any user has to customize that default configuration to suit his actual needs.

ghaerr, 2019-04-30:
> Today ELKS supports not only the MINIX filesystem, but also the ROM one. Full FAT support is also requested for interoperability. Application installation script should not be coupled to the target filesystem, otherwise one would have to write & modify that script for each supported filesystem.
> 
> 

Agreed that application installation should not be tied to target filesystem. However, the
current method uses “sudo install … $(TARGET_MNT)” which assumes a host-mounted
filesystem, whether that be FAT or MINIX.

Are you suggesting this should change? I assume the sudo is required because the install
script always copies to a mounted filesystem, which occurs in image/Makefile.
> Common practice in Makefiles is to specify the root installation folder with the PREFIX variable (or with the --prefix option in GNU Autotools), then up to the application installation script to [...]

Both of these methods require a mounted filesystem (which ELKS uses) or a copy to a staging folder (target/,
which is not implemented).

Currently, the ELKS build is not using target/, are you suggesting that be a change, with
files installed into elkscmd/rootfs_template/…”?  I don’t see any of the code to do this in image/Makefile.

> Therefore I think the nice mfs tool should be only used in the image build script to copy the staging root folder (target/) from the host filesystem to the target one (for the MINIX case). 
> 
> 

Ok. Except ELKS only supports building MINIX images now.

This idea could work from the standpoint of having the application Makefiles
install into a specified $(TARGET_MNT), but then the mfs utility would need a genfs option to 
read a complete filesystem and copy it. Is this what you are suggesting?

I am a bit confused, because there is no current mechanism to build a FAT image, but it seems mkromfs
does compile a filesystem, but I don’t see where that image template is.

Ultimately, I’m just asking for what needs to be done in the image/Makefile or other places to get this finalized,
as I want to create ELKS images under OSX, which cannot use mount. It appears that each application’s
Makefile “sudo install” will need to be changed, and new scripts added to build a /target, then generate
mfs commands from that. This is why I just punted and added a “min” target to images/Makefile for my
OSX system so I can continue programming and testing enhancements.

> For now, we try to build by default the biggest kernel and the maximum number of applications, just to detect more easily any broken dependency. Any user has to customize that default configuration to suit his actual needs.
> 
> 

Using the current .config mechanism, the build tries to both compile and install each application specified.
Since the default 1440 disk is too small to hold all the applications, the default build won’t allow seperating
the app compilation from installation, which forces applications which don’t fit to not be compiled, encouraging
code rot. This is exactly what happened to nano-X. IMHO, the ELKS build should always compile all
applications, since everything is self-contained, including libc and all headers. This will better detect broken dependencies
than having them turned off when they don’t fit on a disk.

mfld-fr, 2019-05-01:
Yes, I am suggesting to remove the 'sudo' from all the application installation scripts. This is currently a "quick & dirty" shortcut to avoid using 'root' for the whole build as in earlier versions of ' build.sh', but stills not a standard workflow.

One is used to first invoke "make all" to build, then "make install" to copy out of the build tree. Acting as 'root' should be required only to install on the local host into the system folders. Wrapping with 'fakeroot' allows to build a disk image as current user, but mimicking the 'root' user only in the context of that build.

But before converging to this better design, the actual question is : does 'fakeroot' works on OSX ?

ghaerr, 2019-05-01:
To my knowledge, fakeroot is not available on OSX.

Mfs removes the need for sudo mounts of a MINIX filesystem. Why is sudo required for the application installs in the first place? Is it to set the installed program uid/gid and mode?

The ELKS build and install process doesn't move any files/applications outside the tree, except for the mounted filesystem, correct?

mfld-fr, 2019-05-01:
> To my knowledge, fakeroot is not available on OSX.
That's a blocking point for my design proposal to be portable :disappointed: 

> Mfs removes the need for sudo mounts of a MINIX filesystem. Why is sudo required for the application installs in the first place? Is it to set the installed program uid/gid and mode?
Yes, `sudo install` makes the target folder / file to be owned by `root`. But all the `sudo` could be factorized in the top level script when iterating on the applications. That could be a first improvement.

> The ELKS build and install process doesn't move any files/applications outside the tree, except for the mounted filesystem, correct?
Not correct: the applications are built "in the source tree", then copied "out of the source tree" to the staging folder `target`, that is actually a mount point for the MINIX case.

jbruchon, 2019-05-01:
I would like to see this `mfs` utility become the preferred way to build a standard image. It solves a lot of problems. The commands that `mkfs` and copy files to the image can be exported variables from higher-level scripts, i.e. `$IMAGECP`, with a suitable stupid default that assumes you're already running as root (sudo by default just needs to go) declared at the top of each Makefile, i.e. `test -z '$IMAGECP' && IMAGECP=cp && export IMAGECP`. A higher-level build script or Makefile will declare i.e. `IMAGECP='mfscp'; export IMAGECP` to invoke `mfs`How's that sound?

ghaerr, 2019-05-01:
> Mfs removes the need for sudo mounts of a MINIX filesystem. Why is sudo required for the application installs in the first place? Is it to set the installed program uid/gid and mode?
>
> Yes, sudo [...]
I think things can be made much simpler than what you or Jody are describing. I see
no reason for requiring or testing for root/sudo access at all. Here’s a step-by-step
of a new image build process (I can make all the changes if you want, super simple):

1. Builds for kernel, elks/tools/mfs, elkscmd/ etc remain unmodified.

2. Remove sudo from all elkscmd/*/Makefiles and image/Makefile.

3. Make the following changes in elks/image, conditionally ifdef CONFIG_IMG_MINIX:

	a) set the target image location to be the rootfs_template using TARGET_MNT=$(ELKSCMD)/rootfs_template” in image/Make.defs.

	b) Replace the all: entry point in image/Makefile with the following commands:
		“$(MAKE) -C $(ELKSCMD_DIR) install - this will install all applications into rootfs_template, using the current users’s uid/gid.
		“mfs $(TARGET_FILE) genfs $(MKFS_OPTS) $(TARGET_MNT) - this will create a MINIX filesystem at image/fd1440.bin and copy all applications, directories, and special files from rootfs_template to it, replacing uid=0/gid=0 within the image.

	c) “mfs $(TARGET_FILE) cp $(ELKS_DIR)/arch/i86/boot/Image $(TARGET_FILE)/linux - this will copy the elks Image to /linux on the image.

	d) “mfs $(TARGET_FILE) boot $(BOOTSECT)” is run, adding boot blocks to the MINIX image.

The only other issue is getting the character and block special files created in rootfs_template in the first place. This is easily solved quickly by moving MAKEDEV and MAKESET sh scripts out of rootfs_template/dev and into a single shell script called “specials.sh” which can be run only once after cloning the source tree, creating the special files in the template. If this is deemed not a good idea, then a simpler version would be to put all the mknod commands into a shell script and run them after the mfs genfs above, using something like “mfs $(TARGET_FILE) mknod dev/XXX c 4 0” for each special file. Mfs will create the special files without the need for root privileges each time.

> The ELKS build and install process doesn't move any files/applications outside the tree, except for the mounted filesystem, correct?
> Not correct: the applications are built "in the source tree", [...]
In this proposed mechanism, the applications are copied within the elkscmd tree to rootfs_template, and can be removed during make clean in images/Makefile using “rm -f rootfs_template/bin, …usr/bin, etc.

What other issues do we have here?

mfld-fr, 2019-05-05:
Completely removed the `sudo` modifier from the application installation scripts in commit 4334632.

mfld-fr, 2019-05-05:
The variable `TARGET_MNT` became misnamed. Replaced by `DESTDIR` that is more standard:
https://www.gnu.org/prep/standards/html_node/DESTDIR.html

mfld-fr, 2019-05-05:
Implemented @ghaerr's suggestion to build all the applications whatever the configuration to help to detect any broken source dependency in commit 4829877.

mfld-fr, 2019-05-05:
@ghaerr One question: what is the purpose of the files you added in `elkscmd/lib/vga/` ?

ghaerr, 2019-05-05:
@mfld-fr: When I was first re-porting Nano-X, I found most of those files in lib/vga, someone put them there. They are a very small implementation of line drawing tests for VGA and Hercules, from code that is already in Nano-X. Since Nano-X wasn't running, I used them to test the low level drivers rather than debug Nano-X, and the patches were included in my nano-X contribution.

They really aren't of any use, and can be deleted, since the Nano-X engine can substitute as a library and performs clipping.

I would also suggest that all of elkscmd/lib, and possibly elkscmd/lib/bsd, be moved over to libc. If you're going to keep libc in ELKS, there is no reason to keep elkscmd/lib. (delete lib/vga anyways, since it's not really a library and not being built). I have it saved in the dev86 branch should anyone want them.

mfld-fr, 2019-05-06:
OK, thanks for the hint, so I removed the useless folder `vga/` in commit daf99d9.

I keep the `lib/` folder, because there could be some common code between the applications that is not part of a standard `libc`.

mfld-fr, 2019-05-08:
Implemented @jbruchon's suggestion in commit b456497: replace direct invocation of the installation commands by Make variables, to allow easily replacing the standard `install` & `ln` commands by the image tools.

mfld-fr, 2019-05-08:
Restored the output folder `target/` to separate it from the source one of the root filesystem template in commit c6245e0. This simplifies the clean up and makes the git status output correct. This also concludes my review of @ghaerr's changes.

ghaerr, 2019-05-09:
@mfld-fr: Now that all applications in elkscmd/ are forced built, the fact that elkscmd/prems doesn't link is breaking the OS X build again. I've attached a two-line patch that fixes this.

The couple issues to keep in mind are: OS X and Linux need a different $(AR) as well as different $(AR) arguments. In this case, the 's' argument is required to build an index, or the .a file won't work.

The other issue, which overrides the previous, is that for some reason, prems.a is having the same linker issue as I had previously with the kernel where the linker running on OS X isn't reading the .a symbols correctly. Since there now isn't any way to turn off applications from being built, I punted and changed the 'pres' build link to use ../prem/*.o instead of ../prem/prem.a and now OS X builds again.

ghaerr, 2019-05-09:
Oops, here's the patch for above.
[prems.patch.txt](https://github.com/jbruchon/elks/files/3164284/prems.patch.txt)

mfld-fr, 2019-05-10:
Patch applied in commit c4c451f. Thanks for the fix !

ghaerr, 2019-05-20:
What remains to be done to finalize our new method of image creation, is it just mkromfs?
Mkromfs does not create the /dev special files, is everything else is done, or not?

I have come up with a less-tricky way of a possible mkromfs enhancement, if that is what is remaining.

If generating a romfs image can be done exclusively from the make generating a minix image, then image/Makefile can use an ifdef to set 'MKDEV=mkromfs ...' separately from mfs to generate /dev entries in the rom image after the rom image's initial creation. In looking hard at mkromfs, it isn't easy to add /dev directory entries and their associated inodes afterwards, unless space for them was previously allocated. Having a command line argument in the initial mkromfs rom file generation that reserves some set number of directory entires in /dev and inodes would work, with a later invocation of mkromfs filling them in, in just the same way that mfs does.

mfld-fr, 2019-05-21:
Yes, restoring ROMFS build is the only thing that is missing. I am currently working on 'mkromfs' to replace the parsing of the target folder by a parsing of a filesystem description from a text file. That text file is filled during the installation phase. For example, $(INSTALL) is substituted by "echo {description} >> list.txt". I will push that new version as soon as it works. Thank you for your following !

ghaerr, 2019-05-21:
The idea of mkromfs building from a text file will solve the existing problem of not having device files in $DESTDIR, which is needed. However, building the a text description of a filesystem could be useful outside of mkromfs.

May I suggest an alternative solution? Build a separate tool that creates a text description of a filesystem, and then have mkromfs use that as input. With this approach, no need to change $INSTALL, let it copy files to $DESTDIR as it does now. Note that you still run a find command to execute chmod's etc even on the MINIX build prior to mfs. No need to reengineer all that within or just for mkromfs. However, this separate tool becomes very useful for creating, say, an MSDOS filesystem using the same image/Makefile with very little additional effort. That is, it could generate a filesystem list using macros that would map to "mwrite" (DOS filesystem copy) commands that could write an unmounted MSDOS filesystem at $TARGET_FILE. The text description would use Makefile-replaced macros just the same way that Make.devices does in image/, and have the make system change those according to the desired target.

I hope the above make sense. I don't see any need to change the ELKS application installation method just for a romfs build, and with an additional tool, could add MSDOS disk making with no further work or sudo requirement.

mfld-fr, 2020-01-26:
I just synchronized the working branch 'image' with 'master' before resuming our work. Please feel free to modify 'mkromfs' as you suggested, before I have time to do it, because I am still quite busy in stabilizing the master branch.

ghaerr, 2020-01-27:
@mfld-fr: I have completed the modifications to support mkromfs creating /dev entries from a passed specification file in rom filesystems, and it fits in nicely with how the device files are specified for MINIX filesystems in the image branch. Since this was the last remaining item prohibiting the image branch to be merged into master, the community should now be able to use the benefits of building ELKS on OSX or non-Linux systems, as well as not requiring root to build MINIX images on Linux. In the future, creating MSDOS images will also be easy using a similar mechanism and not require root nor a kernel FS driver.

The revised mechanism creates a "romfs.devices" file from Make.devices and is used with the new "mkromfs -d romfs.devices $(DESTDIR)" option. The rom filesystem /dev directory is extended with the character and block devices from this file, which is generated from the master device file, Make.devices, in the image/Makefile.

I am attaching a patch file for the two files that needed modification for this work to be complete. I am not good at generating PR's for branches and tend to screw those up, and this patch is small anyways. I would like to ask that you apply the patch please, thank you.

I have some other ideas on how to more quickly generate MSDOS bootable disks if you want help, I will be able to more easily work on ELKS soon now that this work is finally done!
[romfs.txt](https://github.com/jbruchon/elks/files/4115598/romfs.txt)

mfld-fr, 2020-01-27:
@ghaerr : Nice patch to solve the remaining 'mkromfs' problem ! Will apply it in the 'image' branch and perform some non-regression testing in 'emu86' before pushing it to 'master'.

mfld-fr, 2020-01-28:
Default configuration `defconfig` is OK under QEmu, but I got some build errors when setting the configuration `.config` to `config.romfs` (test configuration to be run under `emu86`).

ghaerr, 2020-01-29:
> Default configuration `defconfig` is OK under QEmu, but I got some build errors when setting the configuration `.config` to `config-emu86` (test configuration to be run under `emu86`).

Is this something I need to fix? I don't use emu-86, but instead a single-file 8086 emulator.

mfld-fr, 2020-01-29:
ROMFS build error is now fixed by my latest commit in the 'image' branch, but there is a regression when running 'emu86.sh' : the /dev/tty1 is not found at boot.

ghaerr, 2020-01-30:
Sorry for not testing, I didn't realize there was emu86.sh for testing rom configuration.

I have inspected mkromfs.c and have found my error, the /dev inodes were allocated but not the directory entries - I'm preparing a fix now. I don't have the ability to run emu86 on OSX, trying to think of how to test without having to setup a virtual linux box. Have you tested ROMFS with QEMU, or is emu86 required for the romfs mapping?

mfld-fr, 2020-01-30:
Yeah, I came to the same conclusion after reading more attentively the patch, just some lines missing to add entries.

About 'emu86', it looks bad that tou can't run it under OSX, as we agreed on the policy of project compatibility with that OS... could you please elaborate ?

mfld-fr, 2020-01-30:
Just fixed the missing entries in `/dev`, now able to boot under `emu86` and to enter some commands in `ash` to navigate into the filesystem.

mfld-fr, 2020-01-30:
Submitted for merge to `master` in PR #291

closed 2020-01-30 by mfld-fr

ghaerr, 2020-02-03:
> About 'emu86', it looks bad that tou can't run it under OSX, as we agreed on the policy of project compatibility with that OS... could you please elaborate ?

@mfld-fr: I haven't inspected the source to emu86 yet, but can get more involved if you think this emulator might be easily ported to OSX. 

On a newer OSX system running Mohave 10.14.6, emu86 build immediately fails in elks.c with "asm/ldt.h" file not found.

On an older desktop running Sierra 10.12.6, it produces warnings compiling emu-serial.c with a warning about getpt() not defined, and then fails in the link with _getpt not found.

I use a single .c file emulator "8086tiny" (https://github.com/adriancable/8086tiny) which is EXTREMELY cool and portable - it boots ELKS, emulates a full 8086, and has source and binary for a complete BIOS, which it uses to do the hard stuff, like emulating graphics cards. The cool thing about it is that it's BIOS redirects the BIOS console calls to your terminal console, which means that non-graphical programs, including ELKS boot and running, run right within your console window, rather than in a separate display. Cursor and attribute handling also work. Of course, being a single .c file means that it's not as complete as other emulators out there, but it does boot and run in 10 ms.

I've attached a screen shot of my terminal window running ELKS 0.3.0 :)

[image]