ghaerr/elks #275: Find cannot get path from environment

mfld-fr, 2019-10-02:
Find command in ELKS not working as expected. When I do "find ." from
ELKS command prompt it says
'find: can't get path from environment". But when I do env command I
see my PATH defined. Also echo $PATH returns correct PATH

Reported by Derek Johansen on the mailing list:
https://www.spinics.net/lists/linux-8086/msg00893.html

tkchia, 2019-11-04:
Hello @mfld-fr,

I happen to have encountered a bug in `libc/crt0.S` which might be related (see my pull request https://github.com/jbruchon/elks/pull/279).

Thank you!

mfld-fr, 2019-11-21:
Reproduced on latest master. Going to process @tkchia fix and test again after it.

mfld-fr, 2019-11-21:
Fixed by 878f6640 and merged in #279.

closed 2019-11-21 by mfld-fr