ghaerr/elks #276: New release?

toncho11, 2019-10-17:
Can you please make a new release?

And this time please put some binaries.

You can not expect everyone to compile everything before trying on his computer.

Please.

mfld-fr, 2020-01-05:
Sorry, but the rational is still the same as in issue #243:

"There is no more pre-built image delivered with the release, because it would be hard to generate as many images as possible configurations. Just download the sources, run tools/build.sh to get the cross tools, and run ./build.sh to get the image with the options you want."

Closing this one as duplicate.

closed 2020-01-05 by mfld-fr

toncho11, 2020-01-05:
Yo are missing the point. You need to install dependencies for the built ... it does not always work flawlessly.  

You could have a single default image for 360 KB Intel 8086/8088 IBM PC/XT.

mfld-fr, 2020-01-05:
Even with the FD 360 KB as the medium and the IBM PC / XT as the target, there are still many choices for the userland. This project has no bandwith to manage a catalog of several images, even generated through a CI. That's the point.

About the dependencies, any contribution to implement a "platform / host" checker before the build is welcome...

ghaerr, 2020-01-26:
> "Just download the sources, run tools/build.sh to get the cross tools, and run ./build.sh to get the image with the options you want."

@mfld-fr: I feel for @toncho11 on this issue. While I understand there are no resources for building evaluations of ELKS apparently at any time, I would like to bring back up the bigger problem that it is NOT just a matter of downloading sources and running build.sh, etc. to build images, as currently thought.

On my system, OSX, and any other operating system that does not have MINIX filesystem support built into the kernel, or on any Linux system where one does not have root privileges, one cannot build ELKS images at all. Despite having done a lot of work creating the "mfs" and "mfschk" tools to allow the seamless creation of ELKS images on any operating system as well and not requiring root, these tools aren't even in the master branch, but relegated to a forgotten branch for almost a year now and inaccessible for use with updates since then to the master.

I would like to kindly ask that the branch containing all my improvements, elks/tools etc, be merged into the master branch, so that at least I also can create images, and others can create images without requiring root access. The last item debated was that since mkromfs didn't support /dev entries from an external spec file, that my entire submission couldn't be added. IIRC @mfld-fr volunteered to update mkromfs, but I will happily update it if required, it is no big deal to add that feature. The images/Makefile changes of mine could be added as a separately-named target until the mkromfs changes are made, by myself or mfld-fr.

Please?

I have a number of other enhancements but stopped thinking about contributing when I saw that the master branch got ahead of my branch and it became just too much work to maintain the differences on OSX.

toncho11, 2020-01-26:
There is a difference between developers (tech people) and users. Users do not compile stuff, they just use it. No binary releases is cutting all "users". Don't you want more people to download, test, use and admire your work? I do not understand. You can always built 4 images so that at least some people can download and test.

mfld-fr, 2020-01-26:
@toncho11 : let me clarify something there: ELKS definitely does not target users. Only developers.

This project has obviously no resources to support end users, and I have personally no bandwidth to allocate to that kind of support . I am not the owner of ELKS, just a collaborator with its own priorities.

Getting any recognition on a 'market of the past' with only a few people and even less ancient systems is not one of them, even if I understand the feelings of some nostalgic hobbyists. In my mind, ELKS targets education & training first.

Therefore, contributors that are able to enter the code and to propose solutions are highly appreciated. So any PR that provides a way to automatically build 'ready to use' images for end users is welcome.

mfld-fr, 2020-01-26:
@ghaerr : please stop thinking your highly valuable work is lost.

Indeed,  it is planned as a must-have for the next release. The thing is that this project is moving on slowly. Transitioning often takes some months, but the roadmap is clear (at least in my mind).

Priority today is to stabilize the master branch after the GCC-IA16 transition, then to resume the 'image' branch.

mfld-fr, 2020-02-10:
@toncho11 : you can say 'thanks' to @ghaerr, he worked hard to make 'ready to use' binaries for the next release. See #290 and #307 for more information.