ghaerr/elks #278: "Directive Argument is null"

DaemonChickenofDOOM, 2019-11-01:
Running a ./build.sh from the root directory, the program exited with an error 1 with the following messages:
`../../binutils-src/binutils/wrstabs.c: In function ‘stab_start_class_type’:
../../binutils-src/binutils/wrstabs.c:1476:25: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
    sprintf (vtable, "~%%%s", vstring);
                         ^~
cc1: all warnings being treated as errors
make[5]: *** [Makefile:1061: wrstabs.o] Error 1
make[5]: Leaving directory '/build/binutils-build/binutils'
make[4]: *** [Makefile:1103: all-recursive] Error 1
make[4]: Leaving directory '/build/binutils-build/binutils'
make[3]: *** [Makefile:748: all] Error 2
make[3]: Leaving directory '/build/binutils-build/binutils'
make[2]: *** [Makefile:3574: all-binutils] Error 2
make[2]: Leaving directory '/build/binutils-build'
make[1]: *** [Makefile:849: all] Error 2
make[1]: Leaving directory '/build/binutils-build'
make: *** [/media/raymond/45F51A291BA14383/Documents/code/elks/tools/Makefile:34: .binutils.build] Error 2
Build script has terminated with error 1`

This was run on ubuntu 19.04 under bash. 

Here are the STDERR and STDOUT files:
[error.txt](https://github.com/jbruchon/elks/files/3799514/error.txt)
[stdout.txt](https://github.com/jbruchon/elks/files/3799515/stdout.txt)

DaemonChickenofDOOM, 2019-11-03:
On further examination, this was an issue with the first program the makefile tried downloading and compiling. A bit of tweaking to replace it with the latest version seems to have fixed that, though it did lead to a ton of warnings from the compiler.

mfld-fr, 2019-11-21:
Okay, but what is the actual version that fixed the issue ?

mfld-fr, 2020-01-05:
Looks like to be the same problem as in tkchia/binutils-ia16#5. Fixed by commit 935b8649286c640b2d329a81afb528a8b2cea756.

closed 2020-01-05 by mfld-fr