ghaerr/elks #288: panic: No init or sh found

alejandrocobo, 2020-01-25:
Hi. I've tryed ELKS 0.2.0 on my 8088 with 640kB of RAM laptot and I am unable to run it. I downloaded 0.2 version because it is precompiled. I tried to install gcc-ia16 to compile 0.3.0 on Debian but something is missing in intkchia repositories.

The PC has no hard drive; just two dd floppy drives. I tried both "comb" and "comb_net" with the same result. The errors I saw where these:

Console: Direct 80x25 emulating vt52 (3 virtual consoles)
PC/XT class machine, NEC C30 CPU
640KB base RAM
[...]
Partition check:
bda:hd: error: AX=0x0
BIOSHD:I/O error
dev 300, sector 0
unable to real particion table
bdb:hd: error: AX=0x0
BIOSHD:I/O error
dev 340, sector 0
unable to real particion table
ELKS version 0.2.0
[...]
Unable to open /dev/tty (error 2)
dup failed
No init - running /bin/sh
hd: error: AX=0x0
BIOSHD:I/O error
dev 380, sector 0
sys_execve("/bin/sash", args, 18) => 2.

panic: No init or sh found
apparent call stack
[...]
SYSTEM LOCKED

Thanks!

Thanks.

mfld-fr, 2020-01-25:
Hello @alejandrocobo, the release 0.2.0 is quite old and numerous bug fixes & improvements have been made since. Could you please retry with the latest master ? What is missing in @tkchia repositories ?

alejandrocobo, 2020-01-26:
apt-get sais something like (this is a free translation from Spanish) "the repository "http://ppa.launchpad.net/tkchia/build-ia16/ubuntu" doesn't have a release file". 

And then "The repository data cannot be authentified so it use is highly risky".

Finally "Fail to obtain  http://ppa.launchpad.net/tkchia/build-ia16/ubuntu/dists/focal/main/binary-amd64/Packages 404 Not found".

I'm on Debian 10 and following these steps: https://launchpad.net/~tkchia/+archive/ubuntu/build-ia16

BTW, why there is no precompiled disk images for 0.3.0 like in 0.2.0? That would easy the use of ELKS for those who don't have Linux or don't have the development chain.

Thank you very much!

mfld-fr, 2020-01-26:
@alejandrocobo : Okay, thanks for the additional info.

@tkchia : would you be so kind to have a look ? It seems related to your binaries packages you pushed some times ago.

tkchia, 2020-01-26:
Hello @mfld-fr, hello @alejandrocobo,

> Finally "Fail to obtain http://ppa.launchpad.net/tkchia/build-ia16/ubuntu/dists/focal/main/binary-amd64/Packages 404 Not found".

I have not yet set up my Launchpad repository to build packages for Ubuntu Focal (upcoming 20.04).  There are currently packages for Ubuntu Xenial (16.04) and Ubuntu Bionic (18.04).

I can try to build `gcc-ia16` packages for Focal at short notice, but they will not be well-tested --- I had made some changes since the last "stable" release, and have not yet got around to running the regression tests on them.

Thank you!

mfld-fr, 2020-01-26:
@tkchia : Thank you for the quick reply. By the way, this demonstrates perfectly that maintaining a binary distribution channel has a cost, as stated in #276 😄 !

mfld-fr, 2020-01-26:
@alejandrocobo : even without the binary package provided by @tkchia, the 'tools/build.sh' script dowloads the needed sources and build the non-standard tools (including GCC-IA16) for you. Did you try that way ?

alejandrocobo, 2020-01-26:
@mfld-fr I was able to compile ELKS. Now, when bluild.sh is creating the disk image by leaving everything by default, it appears this:
install: error writing '/home/alex/elks/target//usr/bin/mcopy': No enough spain in the device
install: error writing '/home/alex/elks/target//usr/bin/mkdfs': No enough spain in the device

I guess a 720kB image is not enough for the default settings. Nonetheless, a image is created. When I try to boot that disk, the computer do nothing: there is a cursor blinking while the floppy drive reads the disk but I don't hear the head moving. It remains like that for several minutes until I switch it off..

I'm not sure if that's because not everything required is in the disk or is that I have to change the hardware configuration. I tried by disabling network, VGA and hard drive support.

Thanks to both of you for the fast replies and the support.

PD: BTW, now I understand why there is no precompiled disk images.

mfld-fr, 2020-01-26:
Yeah, the default selection of applets in the user land does not fit into a 720K image, so you have to deselect some in the configuration (make menuconfig), and keep only what you want (at least `init`, `ash` and the basic utilities like 'ls', 'cat' and so on...). You don't need to change the default configuration of the kernel that fits well on an IBM-PC compatible.

alejandrocobo, 2020-01-26:
Ok, I deselected everything except ash, fileutils, minix1, minix2 and minix3 and at boot time is says:
"Minix boot
Press key"

At pressing any key, appears the same message once again.

I've tried including "sysutils" instead of "fileutils" with the same result. If I include both packages, there is not enough space. I left ash in all cases but i didn't find any "init" option. Furthermore, I guess ls, cat, and so are in "fileutils".

So since the 720kB are so tight, what's the minimum system possible?

Thanks for your time!

mfld-fr, 2020-01-26:
Do you have any error code between "MINIX boot" and "Press key" ?

alejandrocobo, 2020-01-26:
Nothing else. Just those two lines of text.

I've tried with the default setting and by disabling de hard drive.

mfld-fr, 2020-01-26:
Before going on real HW, could you please test your image with the 'qemu.sh' script (maybe you would have to replace fd1440 with fd720 in that script), to check the build is OK ?

tkchia, 2020-01-26:
Hello @mfld-fr, hello @alejandrocobo,

I think I managed to reproduce the first problem in QEMU after building a 720 KiB disk image:
```
fd: /dev/fd0 BIOS: 18 sectors, 80 cylinders
VFS: mounted root (minix filesystem).
hd: error: AX=0x20
BIOSHD: I/O error
dev 380, sector 32
```
The sectors-per-track count (18) returned by the BIOS is wrong here, since 720 KiB disks are formatted with 9 sectors per track.

Turning off the option

> Kernel &amp; Hardware → Hardware → Use Int 13h for disk parameters

in `make menuconfig`, and then doing a
```
make clean
make
```
solves this problem for now.

(But in the longer term, perhaps there can be a way for the kernel to extract the correct drive geometry parameters from the boot record --- like what MS-DOS does with its BPB.  The boot record, after all, does know the correct settings to be able to load the kernel in the first place.)

Thank you!

tkchia, 2020-01-26:
Hello @mfld-fr, hello @alejandrocobo,

I find that I get a
```
MINIX boot
4!
Press key
```
error, if I change the build settings to build a 1.4 MiB disk image, after building a 720 KiB image, without properly doing a `make clean` first.

The reason is that the `elkscmd/bootblocks/*.o` files are not recompiled, so they retain the previous geometry settings (for a 720 KiB disk).

Thank you!

alejandrocobo, 2020-01-26:
@mfld-fr I got the exact same message under Qemu.
@tkchia I have not created any image different from the 720kB one. I did run `make clean`, created a new fd720 image and got the same result. If I create a 1.44MB image, it works fine under Qemu.

Sadly, my laptop's floppy controller doesn't have support for high density disks.

Thanks!

tkchia, 2020-01-26:
Hello @alejandrocobo ,

Perhaps try redoing the build with the

> Kernel &amp; Hardware → Hardware → Use Int 13h for disk parameters

option disabled?

Thank you!

alejandrocobo, 2020-01-26:
@tkchia It comes disabled in the default settings, so all my tests were with the 13h int disabled.
Doing it with the int 13h enabled gives the same result. I'm doing it just under Qemu. If you think I can get something different under my laptop, tell me.
BTW, i don't get that `4!` code you got. My message is just
```
MINIX boot
Press key
```

tkchia, 2020-01-26:
Hello @alejandrocobo ,

Now I am baffled.  I am able to get a 720 KiB image working under QEMU, so I am not sure what went wrong on your end.  Perhaps you can upload a copy of your resulting disk image here, so that we can take a look?

Thank you!

mfld-fr, 2020-01-26:
I think there is a little misunderstanding here : the option referenced by @tkchia is that one, that is enabled in the default configuration:
```
[*] Floppy drive(s)
  [ ]   Use INT 13h for floppy count
[*] Hard drive(s)
  [ ]   Use INT 13h for disk parameters <--- THIS ONE
```
Keep the HD option on.

mfld-fr, 2020-01-26:
@tkchia : I also have a working 'fd720' under QEmu with 720K image option, only 'ash', 'sh_utils' and 'sys_utils' as applets, and of course the option above disabled. Nothing else changed in the default.

@alejandrocobo : maybe you could post your .config for us to compare with ours ?

alejandrocobo, 2020-01-26:
@mfld-fr That options comes disabled by default for me too. I mean, both of them floppy and hard disk come disabled by default.  Please, find attached my .config file. [config.zip](https://github.com/jbruchon/elks/files/4114002/config.zip)

I must be doing something wrong.

Thanks!

mfld-fr, 2020-01-26:
Your `.config` looks correct, the only difference with mine is one option not turned off on your side (but I can hardly see the link with the problem):
```
mfld@home /mnt/data/home/mfld/advantech/elks-master $ diff --color .config.work .config.fail
158d157
< # CONFIG_APP_NANOX is not set
```
I am really starting to wonder what could be wrong and could explain the difference in QEmu where it works on our both sides and not on your ???

What is also strange is the fact that you get no error code between lines 'MINIX boot' and 'Press key' in one case... if you look in `elkscmd/bootblocks/minix_first.S`, just after error function `_except`, you could see that at least one character should be issued followed by '!' ?

@tkchia is right : please post us the fd720 image so that we could compare and test it.

alejandrocobo, 2020-01-26:
Please, find attached the fd720 image: [fd720.zip](https://github.com/jbruchon/elks/files/4114077/fd720.zip)

mfld-fr, 2020-01-26:
By the way, here is my `fd720` for a cross test.
[fd720.zip](https://github.com/jbruchon/elks/files/4114085/fd720.zip)

alejandrocobo, 2020-01-26:
Your image works fine in Qemu. But in the laptop I get this:
```
MINIX boot
2!
Press key
```

mfld-fr, 2020-01-26:
OK, thanks for the image file. Well, I got the same result as you under QEmu, and it appears the boot sector is wrong. Are you sure you are on the same commit as us ? Latest on the 'master' branch is: 
```
87b8467b (HEAD -> master, upstream/master) Merge pull request #287 from tkchia/tkchia/devel)
```

mfld-fr, 2020-01-26:
Let us focus now on the failure on real HW. Could you please try this patched image : 
[fd720.zip](https://github.com/jbruchon/elks/files/4114163/fd720.zip)

alejandrocobo, 2020-01-26:
I'm sorry guys, I was using the release channel (cf77a8a). Now I'm rebuilding everything.

@mfld-fr Your last image still gives the `2!` message.

mfld-fr, 2020-01-26:
OK, one point solved :smile: 

Now we have to understand why the `disk_read` function in the boot sector fails on real HW. Could you please uncomment the following line in `minix_second.c` and test again ?
```
		if (!strcmp ((char *)(d_dir + 2 + d), "linux")) {
			//puts ("Linux found\r\n");   <--- THIS ONE
			i_now = (*(int *)(d_dir + d)) - 1;
```

alejandrocobo, 2020-01-26:
The result of uncommenting that line is:
```
MINIX boot
Linux found
2!
Press key
```

mfld-fr, 2020-01-26:
OK, so it means the boot loader succeeded to load the root directory and to find the `linux` file, but failed to load that larger file... mmm... let me try to insert a 'drive_reset' in the 'disk_read' function to recover any error reported by the BIOS.

mfld-fr, 2020-01-26:
Here is the patched image with the reset floppy / retry in the boot sector: [fd720.zip](https://github.com/jbruchon/elks/files/4114291/fd720.zip)

alejandrocobo, 2020-01-26:
Now this is shown:
```
MINIX boot
Linux found
```
Then the floppy drives sounds like the head is going up and down. The sound is similar to the one the a floppy drive makes when it is unable to read the disk because it's damaged. I wrote the image in other disk and the result is the same. I left it during like a minute and nothing else happened.

mfld-fr, 2020-01-26:
Wow... so many read failures :open_mouth: ? Let me patch again the boot sector to print traces to figure out what is the pattern...

mfld-fr, 2020-01-26:
Here is the patched image with traces on read sector failure : [fd720.zip](https://github.com/jbruchon/elks/files/4114327/fd720.zip)

Could you try it several times to see if the pattern is always the same or if the failure randomly happens ?

alejandrocobo, 2020-01-27:
This is what happens every time I try to boot your last image:

[image]

mfld-fr, 2020-01-28:
Thank you very much @alejandrocobo for your testing time, it really helps. Let us now try to understand why the loading of the 'linux' file fails every time at the second call to 'disk_read'...

alejandrocobo, 2020-01-28:
In case this gives you a clue, the original double density floppy drive was replaced by a high density floppy drive. The floppy controller is compatible just with double density disks so if I try to use a formated high density disk, it doen´t work. 

High density floppy disks work just when I put a sticker on the hd hole, otherwise I cannot read nor format them. The computer works fine with MS-DOS 3.30. I´ve installed WordPerfect 5.1 from the install discs to other discs (a: to b: installation) and works just fine. Furthermore I played Prince of Persia plus runned several diagnostics tools to check out other things I was testing. Everything works fine; no issues related with floppy drive/disk to report over here.

ELKS 0.2.0 goes further away at boot time but a read you made some changes in this regard in #253 so maybe this hardware replace is related...

Thanks!

mfld-fr, 2020-01-31:
@alejandrocobo : I added some code to the `emu86` emulator to exercice the boot block and I found 2 defects while executing the code step by step. Could you please test this new image where I fixed that defects ? Here it is : [fd720.zip](https://github.com/elks-org/elks/files/4141400/fd720.zip)

alejandrocobo, 2020-02-01:
It works! 

Thank you very much for your time.

mfld-fr, 2020-02-01:
Yeah :smile: ! Thank you also for your patience. Let me now commit the fixes to the master branch so that you can build yourselves...

mfld-fr, 2020-02-01:
Issue #253 fixed by commit ea167e1, that also solves the last problem on real HW.

closed 2020-02-01 by mfld-fr

tkchia, 2020-02-01:
Thank you @mfld-fr and @alejandrocobo!

Two things on my mind for now:

  * @mfld-fr: I may have missed something, but in your experience, did the "don't move ES, only BX" patch really make the boot code more robust than before?
  * It seems that we can always use some better debugging facilities in the boot block code.  I will see how I can enable more debug dumps in the code, while keeping it within the 2-sector limit.

Thank you!

mfld-fr, 2020-02-01:
@tkchia : It looks like some BIOS (or something beyond) are quite intolerant. I got some reports of read failure with INT 13h whenever ES is not aligned to very big powers of 2. This is the main reason why the new boot sector relocates to the highest & aligned 64K :
```
	// Move to the latest rounded 64K
	// To avoid unaligned buffer for BIOS INT 13h

	sub $0x1000,%ax
	and $0xF000,%ax
	mov %ax,%es
```
I am even wondering if the default load address for the kernel (0x1000:0) is not the safest one for the same reason...

About the "2-sector limit" : I would like to unify all the ELKS boot sectors in one, and the MINIX one is just a draft for now. The specific payload for MINIX is in the next sector, so that the whole loader fits into the first 1K block as MINIX constrains it (= the "boot block").

That payload could be more sectors for another filesystem (like "reserved sectors" in FAT, or the whole system for RAW)... but the non FS-specific code should be within the 1-sector limit to achieve that unification goal, including some debugging facilities.

One could even think to a special payload that won't load the system, but would probe the HW and help the ELKS user to configure that system...

tkchia, 2020-02-02:
Hello @mfld-fr,

> @tkchia : It looks like some BIOS (or something beyond) are quite intolerant. I got some reports of read failure with INT 13h whenever ES is not aligned to very big powers of 2.

Argh! --- I understand now.  In that case perhaps a way to work around this, while allowing reads beyond 64 KiB, is to advance `%bx` until it overflows, then start advancing `%es` in increments of `0x1000`.

> That payload could be more sectors for another filesystem (like "reserved sectors" in FAT, or the whole system for RAW)... but the non FS-specific code should be within the 1-sector limit to achieve that unification goal, including some debugging facilities.

I think this will work for filesystems such as `ext2` and FAT32, which already leave at least 2 sectors free at the filesystem start.

But it may be a bit problematic for FAT12 and FAT16 --- I _think_ there is a general expectation for these filesystems that the file allocation table proper will start on the second sector.  (Certainly `mkdosfs` creates FAT12/16 filesystems with only 1 reserved sector on default, though it can be configured to reserve more sectors.)

Thank you!

tkchia, 2020-02-02:
Hello @mfld-fr,

I can think of one way to allow an OS-independent boot sector to be used with FAT12 and FAT16: namely, make the location of the OS-specific payload sector configurable, instead of being fixed at the second sector.

We can probably add one more parameter to the list of parameters at `boot_sect.S`'s end, like this:
```
sect_max:
	... .word ...
head_max:
	... .word ...
track_max:
	... .word ...
payload_logical_sect:
	.word 1
```
In addition, to allow programs to locate this parameter block, perhaps add some sort of magic marker before the parameters, perhaps like this:
```
	.ascii "<eLkS>"
sect_max:
	... .word ...
head_max:
	... .word ...
track_max:
	... .word ...
payload_logical_sect:
	.word 1
```
Then a program that processes FAT12/16 disks or disk images can use the magic marker to locate the parameter block, and adjust `payload_logical_sect` accordingly.

Thank you!

mfld-fr, 2020-02-02:
@tkchia : Sure ! As the new boot sector knows the geometry, I fully agree that a next step would be to add the logical (first sector, sector count) of the payload to be able to place it anywhere on the disk, and cover all the use cases.