ghaerr/elks #289: Move ELKS ownership to an organisation

mfld-fr, 2020-01-29:
Today I spend much time to rebuild and test ELKS on two main configurations (one for the standard PC and another for my SBC), after processing almost every PR or making some critical changes, and even with that testing discipline, I often miss some regressions.

So I think it is time now to implement a CI to automatically build and perform some sanity checks before the PR, or after any commit, to save our already limited time out of these repetitive tasks (and open a way to setup an automatic deployment of the releases in the form of 'ready-to-use' binary packages for end users, but personally I won't invest time into that direction, as explained in #276).

To begin the CI workflow, one need to be able to change the project settings in GitHub (especially the web hooks / actions / apps to dialog with a CI server). Currently I am only a collaborator in this project, and have no access to that settings. As far as I know, only the owner of the repository could do that.

I don't want to challenge the current ownership, but need to have such access. One solution is to create an 'ELKS' organization, transfer the ownership of this repository to that organization, and setup permissions in a more fine-grained way.

@jbruchon : if you agree that proposal, could you please create such organization and set me as a member ?

jbruchon, 2020-01-29:
Done. Have fun.

closed 2020-01-29 by jbruchon

ghaerr, 2020-01-30:
> (and open a way to setup an automatic deployment of the releases in the form of 'ready-to-use' binary packages for end users
 
I have thought through a design for a lightweight, text-based ELKS package management mechanism using make and command line tools that would allow easy maintenance and addition of ELKS applications to be specified to be copied to MINIX or FAT images during or after each build, or automatically, is there interest in this contribution?

I am thinking of a Packages file in the elks/images directory that would specify which applications would be copied by the "mfs" (and upcoming "msfs") program from the template dir to various images, based on image names like fd360-minix, fd720-msdos, hd-minix, network-minix, etc that both ELKS maintainers, the CI system, and users could use to manage a variety of ELKS distribution or test images. All ELKS applications would be built in every build (unless explicitly turned off) which should help regression testing/rot and the package manager used to produce desired sets of distribution binaries, instead of the current mechanism of specifying each application wanted in the .config file.

Mellvik, 2020-01-30:
I strongly support this proposal. The recent infusion of energy into the project is inspiring. It is also a good time to reflect on the goals of the project, in order to focus whatever our collective eneriges may accomplish towards something both useful and achievable. 

I may have opinions in that regard, but since my contributions (so far) are testing only, I'll rest my case for now.

Thanks for all the good work. 

Mellvik

mfld-fr, 2020-01-30:
@Mellvik : Testers are as important as coders, and your testing helped much in tuning the networking and the FAT features. Thanks for your support !

mfld-fr, 2020-01-30:
@jbruchon : thank you very much for the move and your trust ! Would you have any concern if we short the 'id' of the organization to 'elks' (but keep the full 'name' as it is) ?

mfld-fr, 2020-01-30:
@ghaerr : moved your proposal to #290 to separate the threads and ease the discussion. This issue addresses only the organization setup.

jbruchon, 2020-01-30:
@mfld-fr I have not done much of value for ELKS in some time; I'm fine handing ELKS and its future over to you. I don't have time to devote to it for the foreseeable future. You have "owner" status on the organization, so you may push through whatever changes you wish. I'll be shutting off project notifications to myself, though I'll probably leave myself as an "owner" on the new organization in case I change my mind and want to dive back in one day. Good luck and godspeed, friends.

mfld-fr, 2020-01-30:
@jbruchon : no, really nothing : you just rescued the project in its dark times and saved us from the @lkundrak's syndrome 😆 

More seriously, I think that new organization would save again the project when my turn will come... that's how the life goes on...

jbruchon, 2020-01-30:
I'm happy to have done it. It was a miracle that I got my hands on it in the first place, and I feel like cleaning out all the garbage "wishful thinking" config options and writing `build.sh` were the biggest contributions I made to the project outside of simply keeping it from death. Keeping the build process newbie-accessible is top priority. In the new world of relying on a specialized version of GCC, please keep that in mind. Build troubles will be a constant source of pain, so anything you can do to reduce that will pay off ten times what you put into them.

Mellvik, 2020-01-30:
First of all let me second the kudos to Jody for being the 'rescuer'. Whatever the other contributions are, the rescue itself and the energy delivered was THE life saver for ELKS.

Secondly, and quite possibly repeating myself, the key benefit and potential I see in ELKS is becoming the ultimate tool for 'low resource hardware' - typically vintage PCs and SoCs. Most of us probably have pet projects - systems or hardware we'd like to either bring back to life or make more useful. On many of these systems we can boot some version of MSDOS, maybe even FreeDOS, but getting closer to the hardware, which ELKS makes possible, is the ultimate debugging tool. 

I'm rambling. What I'm really saying is I'm happy ELKS is here, and I look forward to follow the project from the sideline, testing on real hardware whenever I have time and/or someone asks. I have a stack of Compaq portables (III/286, 386, and the like, serial (ISA) cards, Ethernet cards etc.). Odd as they are, they present some peculiar challenges @ the BIOS level. 

I also have a (formerly mentioned) ambition to get ELKS running on the Lantronix Xport, now quite old, but a fun size factor containing a 80186 core and some other (meager) resources. Look it up - this is an ethernet connector that happens to have an SoC inside. Running ELKS - eventually.

Helge (aka Mellvik)