ghaerr/elks #290: Binary packages for end users

mfld-fr, 2020-01-30:
> (and open a way to setup an automatic deployment of the releases in the form of 'ready-to-use' binary packages for end users)

I have thought through a design for a lightweight, text-based ELKS package management mechanism using make and command line tools that would allow easy maintenance and addition of ELKS applications to be specified to be copied to MINIX or FAT images during or after each build, or automatically, is there interest in this contribution?

I am thinking of a Packages file in the elks/images directory that would specify which applications would be copied by the "mfs" (and upcoming "msfs") program from the template dir to various images, based on image names like fd360-minix, fd720-msdos, hd-minix, network-minix, etc that both ELKS maintainers, the CI system, and users could use to manage a variety of ELKS distribution or test images. All ELKS applications would be built in every build (unless explicitly turned off) which should help regression testing/rot and the package manager used to produce desired sets of distribution binaries, instead of the current mechanism of specifying each application wanted in the .config file.

mfld-fr, 2020-01-30:
Moved from #289

mfld-fr, 2020-01-30:
The selection of the applets to be included into the image **during** the build is already implemented in the configuration scripts, based on an old version of KConfig (BTW should be upgraded from Linux mainstream for better dependency management).

I would advise not to modify that design, inspired by (BuildRoot)[https://buildroot.org], that fully decouples the selection of the applets and the selection of the image format. Earlier design was coupling the image formats and their contents, and proved to be unmanageable.

I would prefer a set of scripts that iterates on generating one `.config` for every wanted image, building & packaging & deploying that image, as a second layer based on the existing system.

Your suggestion to build all the applications in issue #261 has already been implemented in commit 4829877a8ba04910b8d7d1539408e283225f4b32.

For selection and inclusion **after** the build, I am fully open...

ghaerr, 2020-01-31:
> The selection of the applets to be included into the image **during** the build is already implemented in the configuration scripts, based on an old version of KConfig.
> 
> I would advise not to modify that design, inspired by [BuildRoot](https://buildroot.org), that fully decouples the selection of the applets and the selection of the image format. Earlier design was coupling the image formats and their contents, and proved to be unmanageable.

My package management idea was not intended to change the build process or .config/KConfig mechanism, and will improve the ability for selecting applications and image format *after* the build has completed.

I would like to point out that we need greater clarity in this discussion when we talk of "images": this could include application selection (the set of applications selected for an image), the filesystem image (which has a filesystem format, an elks kernel usually, and a specific filesystem format and size), and the specific elks kernel being built. My understanding is that the current .config/KConfig process is used for all of these items now (apps, kernel, and image format). Selected apps (default all) are copied into an image in the selected filesystem format along with the specified kernel.

My proposal relates to having much more flexibility in generating multiple filesystem image formats and sizes with different applications, but using the same kernel, all AFTER the build. I realize that one may not need much flexibility with multiple image formats and applications using the same kernel, but this idea would allow for creation of a variety of fd360, fd720, hd etc images with varying applications and filesystem formats using the same kernel, for testing and in larger or smaller environments, or for the CI system.

I plan to produce a PR which will allow this, while leaving the entirety of the .config/KConfig mechanism unchanged. The biggest change for users will be the ability to specify different applications sets on various sized images without having to rerun the build changing the applications selection settings. This will work because the default build now always builds all applications.

mfld-fr, 2020-01-31:
Yeah, I think the discussion would be with greatly clarified if you implement your proposal in one of your branch, so that we could review it and get a clear picture of that proposal.

mfld-fr, 2020-02-28:
Package manager design makes the assumption that selecting applications (or group of applications) can be made independently of the kernel configuration, so after the build.

This is a incorrect assumption. For example, bootable MINIX or FAT volume cannot work without the relevant filesystem enabled in the kernel. Also network application cannot works without the network support in the kernel. 

It could have been a valid design if all the features could fit in the 64K code segment of the kernel, but it is far the case today, as we cannot select MINIX, FAT and NE2K at the same time.

Configuring the kernel and selecting applications are two dependent things that are consistently managed in the existing configuration workflow, not in the package manager 'extension', and result in a more optimal (kernel + user land) in term of disk space.

Another package manager weakness is that the user has to edit text files to modify the application selection, rather than using the more friendly `menuconfig` with its dependency checking.

Building one configuration only takes 2 minutes and a half in the CI (see https://github.com/jbruchon/elks/actions).

Conclusion: the package manager should be dropped, current application selection moved as a set of configuration files (like 'config-pc-at-fd1440-minix' or 'config-pc-xt-fd360-fat'), and a simple script should iterate on the configuration files to build all the predefined images.

In the CI, the build could be faster than this iteration, as one parallel job can handle one configuration. Another script can easily iterate on the configuration files and generate the job descriptions in the same repository.

ghaerr, 2020-03-13:
Hello @mfld-fr (and others),

After your statement in #458 (all applications needs to be built regardless of config settings), and proposed solution (only change install according to config settings), it got me thinking as to the nature of the issue here, that of building binary images (for end users or developers), and how similar the problem is to what we currently are dealing with in elkscmd and #457, #458 and #410.

I have come to see that the ‘package manager’ as it now exists does not belong in the image directory, nor should it be directly involved in building images. You see, the reason that it was created wasn’t really to build images, but actually to select which applications should be in an image. And we have that very problem now, the problem of mixing whether to build an application (all should be built, which is not the case now), and whether or how to install a set of applications (controlled by config, as discussed in #458). Let me explain more:

The problem we now have in elkscmd/ and config, is that of organization: the elkscmd/ directories are mostly organized as to ‘where’ the applications came from, as in minix, GNU shutils 1.12, GNU fileutils 3.14, busyelks, etc. 

On the other hand, config only allows selection of all applications within any of these source directories, or none. It isn’t possible to select a few applications from one source directory, and a few from another (which busyelks needs), without writing complicated makefile changes. The config-selectable categories kind of work, but have no regard for changes required when image sizes change and are tied directly to the source directory, which is too rigid.

In addition, several commands have been stopped from building (e.g. Nano-X), because the resulting required ‘install’ (since the applications are in the same source directory) won’t fit on some sized disk. The user is also completely responsible for determining what will fit and what not through trial and error in config.

Currently, the busyelks maintainer has to be involved messing around modifying build Makefiles every time an application source directory adds/deletes an application from the build, or another application requiring a symlink is added to busyelks. Same for any other set of applications. If another bunch of applications are added in a directory, all of those applications or none have to be installed, and only one config setting can be used to select to include or not.

I propose a better design to handle these issues: separate installation of applications from their build makefiles entirely. In all of 107 applications which are built, all but 7 install with the exact same line ‘$(INSTALL) $(OUTS) $(DESTDIR)/bin’. Except that we have to include two sets of ifdef/ifndef CONFIG_APP_xxx around the install lines in every ‘source’ directory for busyelks, for instance.

In my proposal, all ‘install’ lines are removed from all elkscmd Makefiles, and instead dealt with in a single, separate Make.install file. In this file, all the config-selectable application categories are dealt with, such that the all applications are always built, but only the categories specified in config are installed. The application maintainer does not deal with application installation onto ELKS distribution images.

This is where the revised package manager comes in: rather than building a massive Make.install where all sorts of conflicting ifdef/ifndef CONFIG_APP_ defines are wrapped around various ‘install’ commands, a single ‘Package’ or ‘Applications’ file specifies, using ‘:tags’ that match the config-based selections, which applications to copy. Thus, the entire installation is table-driven, based on an OR-based tag mechanism to select an application for installation into $(DESTDIR). The ‘package manager’ does exactly this. It should be removed from images and it’s basic mechanism used in elkscmd.

With a few modifications, this ‘Package’ file could specify an alternate name to be installed as, and also specify whether to generate a symlink or not to another file. Currently, each set of source has to have its own install and its own symlinks in its own Makefile (and in busyelks using non-portable `ln –r` that breaks the build). In this new design, the application maintainer only has to worry about the application source, NOT all the portability/config/imagesize, etc. issues of getting the application installed.

This proposed design has the major benefit of being able to specify ‘named’ sets of commands that come from portions of multiple source directories extremely easily. Basically, the config-selection specifies a ‘tag’ that can select applications from a multitude of source directories, with new tags easily added based on other config-specified options, like image size, or automatic selection of applications base on new config options not tied directly to source directories.

Going with this design allows getting rid of the ‘package manager’ from image/ and using a script to create a variety of disk images, using the same or different tags from the current config-selected application sets for different image sizes. Also, @marcin-laszewski gets out of maintaining build scripts and can work on busyelks itself, rather than its installation.

What do you think? After discussion and some agreement, I can submit sequential PRs, moving the implementation forward in several smaller steps. Also, @marcin-laszewski won't have to figure out new solutions to his PRs #457 and #458, as this will take care of it.

mfld-fr, 2020-03-13:
@ghaerr : agree the big picture

ghaerr, 2020-03-19:
Now that elkscmd/Makefile is managing application selection and installation through config options, the package manager 'Packages' file and associated image making process can be dismantled and replaced with a simpler script that builds extra binary images based only on config options.

I will start working on the new process now. Do we want 'make images' to make the extra images for testing or end users, or should there be a config option 'Build extra binary images Y/N'? (I think the latter).

There is still the problem of selecting all the application categories being too large to fit on smaller (360k and 720k) disks. My proposed solution for this is to introduce, into config, two options for application selection: the current system, which allows selecting applications by essentially source directory (diskutils, shell utils, etc), OR selecting by virtual category, which would be image size. A question would be asked in config: 'Select applications by image size Y/N'. If Y, then the applications would be selected based on the disk image size selected in config. If N, then the applications would be selected manually, as they are now.

With this modification to elkscmd/config.in and elkscmd/Makefile, the generation of the applications in $(DESTDIR) for the image/Makefile 'template:' target would work with passed CONFIG_IMG_FDxxx and CONFIG_IMG_{MINIX,FAT} values, allowing the process to be driven by the same definitions config uses.  The current package manager would then be removed.

Comments?

ghaerr, 2020-04-05:
Hello @mfld-fr,

>  that would specify which applications would be copied ... to various images, based on image names like fd360-minix, fd720-msdos, hd-minix, network-minix, etc that both ELKS maintainers, the CI system, and users could use to manage a variety of ELKS distribution or test images. 

Now that 'extra' images are able to created from the config file process, how are you thinking they might be made available? Do you have a mechanism in mind that would allow them to be created automatically and made available through Github, or should users/CI/maintainers have to continue to build the images separately? I think this would be the last item to be performed for this issue.

toncho11, 2020-10-03:
What about this: [https://github.com/marketplace/actions/automatic-releases](https://github.com/marketplace/actions/automatic-releases)

toncho11, 2020-10-03:
What about this: https://medium.com/@_oleksii_/set-up-automated-builds-using-github-and-docker-hub-12c3e0f18eba

It explores the idea of automatic builds on Docker Hub. Docker Hub has a free plan for public repositories.

ghaerr, 2020-10-06:
Hello @toncho11,

Thanks for the information, I'll take a look at it. 

Thank you!

closed 2022-02-08 by ghaerr