ghaerr/elks #298: Continuous integration setup

mfld-fr, 2020-02-04:
This issue tracks the setup of the continuous integration:
- [ ] Cross tools pre-build : WIP
- [ ] Default configuration build
- [ ] ROM configuration build
- [ ] ROM configuration test

mfld-fr, 2020-02-04:
@tkchia : I have an error while building the IA16 cross tools on the Ubuntu VM provided by Github: https://github.com/elks-org/elks/runs/425307644

Do you have any idea on how to disable the build of the documentation in 'binutils-ia16' and 'gcc-ia16' not only to fix that problem but also to speed up the build ?

tkchia, 2020-02-04:
Hello @mfld-fr,

I guess one possible workaround is to set the `$(MAKEINFO)` variable to `":"` or `true` (either command will just do nothing) in the sub-`make`'s for building `binutils-ia16` and `gcc-ia16`.  I am currently trying this out.

E.g.
```
	$(MAKE) -C binutils-build MAKEINFO=:
	$(MAKE) -C binutils-build MAKEINFO=: install
```

Thank you!

mfld-fr, 2020-02-04:
@tkchia : Thank you for the trick ! It appeared in the meantime that the VM is quite fast to build the cross tools (around 20 minutes), so reducing the build time is not so critical as I initially thought. So I merely added a step to install the missing dependencies (flex, bison and texinfo).

mfld-fr, 2020-02-05:
Build is now complete in the CI, so any new PR has to pass the pipeline to be accepted.

closed 2020-02-05 by mfld-fr