ghaerr/elks #299: Restore RAW image generation

mfld-fr, 2020-02-05:
This issue tracks a regression introduced by PR #261 from @ghaerr, that zapped out the case of the CONFIG_IMG_RAW option. The user cannot generate a RAW image anymore.

ghaerr, 2020-02-05:
I went back to the source tree of Jun 16, 2018 before any of the mods for removing sudo, etc., and can't find any reference to CONFIG_IMG_RAW. Can you point out where in the previous work this was implemented and I'll prepare a PR to fix it? I don't actually know what CONFIG_IMAGE_RAW is.

ghaerr, 2020-03-03:
Hello @mfld-fr:

Is the broken RAW Image generation just the creation of a 'fd.bin' which is just a copy of 'Image' (boot/setup/system) on it (no filesystem)? If so, I can fix image/Makefile to do this.

Or does it also have to have a ram or rom disk image following?

mfld-fr, 2020-03-05:
Hello @ghaerr 

Yes, just dumping `Image` to the first sectors of `fd.bin` without any filesystem for now. Initial disk or filesystem will came later (on my roadmap).

mfld-fr, 2020-03-07:
Fixed by #413.
Thanks @ghaerr !

closed 2020-03-07 by mfld-fr