ghaerr/elks #303: [config] Config system needs updating

ghaerr, 2020-02-06:
The KConfig-based system is broken on OSX, so I can't update or test various CONFIG options. Following are ones needed in order to test new features being added to MSDOS-FS. These need to be added to various config.in scripts, and arch/i86/defconfig, I think.

These options should be accessible when turning on and off CONFIG_MSDOS_FS
CONFIG_FULL_VFS          (currently working)
CONFIG_MSDOS_DEV     (turns on /dev support on FAT volumes)
CONFIG_32BIT_INODES  (missing from some config scripts it seems)

Will the "make config" option work, instead of "make menuconfig", if the above options are fixed?

It also seems that a "make clean" does not reset the system state to start. Does the .config file need to be removed for that to work?

Sorry for the dumb questions, I haven't been able to run the config system and builds are starting to get messed up strangely.

mfld-fr, 2020-02-07:
As I said, I think it is worth to take a little time to solve the 'menuconfig' on OSX before continuing the rush...

ghaerr, 2020-02-07:
How do you think its best to proceed on fixing menuconfig/config, without having to jump in too far into this old and needs-updated code? You were mentioning that perhaps it should be updated to a new version, I don't know anything about that? On one of my OSX systems, the make config fails with ncurses library entry points not found, even though I have ncurses, and on another, make menconfig runs but crashes when running through various screens.

Would you like me to post error messages so you or others can direct me?

ghaerr, 2020-02-07:
We desperately need to get MSDOS_CONFIG_DEV and the CONFIG_32BIT_INODES inserted properly in a config.in an defconfig files. Would you like me to prepare a PR for this, even though I can't run it?

Also, I have been hand-editing include/autoconfig.h, otherwise I couldn't test CONFIG_MSDOS_DEV support, that might be part of my very strange build problems going on. I have to admit, I don't yet understand the build process with .config and include/autoconfig.h being used to generate make steps...

mfld-fr, 2020-02-07:
@fonin highlighted in his commit bfb0496f369fdcc022c62556a29e692191a5e80c that one needs the 'development package' of 'libncurses' to build `menuconfig`.

fonin, 2020-02-07:
@ghaerr if in urge, you might configure on another system with working menuconfig, and place .config on the target system. Or even edit .config manually, it is pretty simple.

ghaerr, 2020-02-07:
@fonin: Thank you for the advice. I have already been hand-editing .config and it has worked very well, except for when I had to add a new CONFIG_MSDOS_DEV option for the recent /dev implementation on FAT disks. Adding to .config didn't do any good, since the same value needs to be defined in include/autoconfig.h (which I then added). After modifying autoconfig.h by hand (which is says not to), my builds starting acting very strangely. I am currently unable to build my /dev implementation, although I was able to test completely before submitting it! LOL :)

mfld-fr, 2020-02-07:
Added the CONFIG_FS_DEV option to the configuration menu in 367f405.

ghaerr, 2020-02-07:
> @fonin highlighted in his commit [bfb0496](https://github.com/elks-org/elks/commit/bfb0496f369fdcc022c62556a29e692191a5e80c) that one needs the 'development package' of 'libncurses' to build `menuconfig`.

@mfld-fr: Thanks, I will try that on my laptop. Here are the error messages I get when running make menuconfig on my desktop. This compiles and will display the initial screen but crashes with the following after going into Hardware:

make menuconfig
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C config all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C lxdialog all
make[2]: Nothing to be done for `all'.
/bin/bash config/Menuconfig config.in
Using defaults found in .config
Preparing scripts: functions, parsingawk: MCmenu11 makes too many open files
 source line number 22
Awk died with error code 2. Giving up.
.../MCmenu1: line 27: syntax error: unexpected end of file
........../MCmenu9: line 8: syntax error: unexpected end of file
done.

Menuconfig has encountered a possible error in one of the kernel's
configuration files and is unable to continue.  Here is the error
report:

 Q> config/Menuconfig: line 739: MCmenu1: command not found

Please report this to the mailing list: linux-8086@vger.kernel.org
Please indicate the kernel version you are trying to configure and
which menu you were trying to enter when this error occurred.
make: *** [menuconfig] Error 1

mfld-fr, 2020-02-08:
Renamed CONFIG_MSDOS_FS to CONFIG_FS_FAT in commit f6a65184.

ghaerr, 2020-02-08:
@mfld-fr and @tkchia:

I am happy to report, that after all of our latest changes to the boot blocks, kernel floppy driver reading ELKS BPB, updating the various new CONFIG options, adding the FAT /dev driver, setboot boot block writer and package manager upgrades, that all the ELKS MINIX images produced by "make images" now all boot successfully, and produce a working ELKS system with an auto-mounted FAT filesystem that includes /dev support!

In addition, I'm able to run "make config" (graphical system still not working) that generates a correctly built .config file and include/autoconf.h that builds the entire system in one pass! (still requires "make images" afterwards).

Nice work guys!!! This is really coming along!!!

mfld-fr, 2020-02-08:
Ahem... :smirk: 
```
make -f Make.package msdos NAME=fd360-fat SIZE=360 APPS=":boot|:small"
make[1] : on entre dans le répertoire « /mnt/data/home/mfld/advantech/elks-master/image »
awk "/:boot|:small/{print}" Packages | cut -f 1 | sed "/^#/d" > Filelist
dd if=/dev/zero of=/mnt/data/home/mfld/advantech/elks-master/image/fd360-fat.bin bs=1024 count=360
360+0 enregistrements lus
360+0 enregistrements écrits
368640 octets (369 kB, 360 KiB) copiés, 0,0011338 s, 325 MB/s
mformat -i /mnt/data/home/mfld/advantech/elks-master/image/fd360-fat.bin -f 360 -M 512 -d 2 -L 9 -r 14 -c 1 -R 1 -k -N 0
mformat : option invalide -- 'R'
Mtools version 4.0.18, dated January 9th, 2013
Usage: mformat [-V] [-t tracks] [-h heads] [-n sectors] [-v label] [-1] [-4] [-8] [-f size] [-N serialnumber] [-k] [-B bootsector] [-r root_dir_len] [-L fat_len] [-F] [-I fsVersion] [-C] [-c cluster_size] [-H hidden_sectors] [-X] [-S hardsectorsize] [-M softsectorsize] [-3] [-2 track0sectors] [-0 rate0] [-A rateany] [-a]device
make[1]: *** [Make.package:82: msdos] Error 1
make[1] : on quitte le répertoire « /mnt/data/home/mfld/advantech/elks-master/image »
make: *** [Makefile:105: fd360-fat] Error 2
```

ghaerr, 2020-02-08:
What version are you running? (mformat --version). I'm running 4.0.23 and it works fine.

mfld-fr, 2020-02-08:
```
mfld@home /mnt/data/home/mfld/advantech/elks-master/elks/tools $ mtools --version
mtools (GNU mtools) 4.0.18
configured with the following options: enable-xdf disable-vold disable-new-vold disable-debug enable-raw-term 
```
It works very well when dropping the -R option.

ghaerr, 2020-02-08:
Please don't remove the -R option from the build, instead, upgrade your mtools to 4.0.23, as recommended when I made the package manager PR in the first place.

I plan on retesting @tkchia's result that MSDOS doesn't recognize disks that are created with more than 1 reserved sector. All code that I've seen, including MSFT's boot blocks seem to recognize the FAT reserved space BPB field. Thus, the issue could be in @tkchia's use of the 'dosftools' package and its associate disk formatting tool. I plan on testing that in detail shortly and using an easily changed -R parameter in the Make.packages script. Since I'm now maintaining FAT, I want to get too the bottom of this reserved sectors issue. (BTW, ELKS FAT driver supports reserved sectors).

ghaerr, 2020-02-08:
Here's the latest mtools package (4.0.23) for those that are interested in using the "cd images; make images" package manager after the main build, which creates the 3 MINIX and 3 FAT disk images, for seeing all the new ELKS features in an emulator or on your own hardware. It easily compiles with "./configure; make; make install".

[mtools-4.0.23.tar.gz](https://github.com/elks-org/elks/files/4174815/mtools-4.0.23.tar.gz)

mfld-fr, 2020-02-09:
By the way, could you please tell us where you sourced 'mfs' and other MINIX tools from ?

ghaerr, 2020-02-10:
@mfld-fr: The original source for mfs came from a sourceforge project I found, mfs-0.5pre.tar.gz, originally written by Alejandro Liu Ly in 2005. There were no other MINIX v1 filesystem manipulation tools out there that I could find, only v2. There was only one version produced, and it was buggy, also only working on MINIX v2, not v1. I suspect it was a student project. I heavily rewrote it and added many features, all documented in the top of main.c.

Since mfs was basically rewritten from basic MINIX file handling source, I wanted to use a "standard" filesystem checker to ensure that the filesystems were in fact created correctly. For that, I chose a relatively update-to-date single-file version of fsck.minix.c from the Linux kernel tree. This version was originally written by Linus in 1991, and looks like it was last modified in 2008-04-06, all indicated in its source.

ghaerr, 2020-02-10:
@mfld-fr: I reran "make config" on another OS X system and it properly produced a .config file. However, I noticed something I'd like you to fix: The option for CONFIG_32BIT_INODES is always configured ON when CONFIG_FS_FAT is selected. I believe we should have that remain user-selected and default to OFF instead of forced ON, when the FAT filesystem is being built.

The reason for this change is that 1) I have rewritten and heavily tested both cases of 32 bit and 16 bit inodes in the last few PR's for FAT, and I am very confident that FAT will work both ways. 2) FAT12 never needs 32 bit inodes, and many people may be concerned about excess memory usage, which isn't needed on small floppy-only systems. 3) I have added a kernel message "FAT: disk too large, turn on CONFIG_32BIT_INODES" when the sector number overflows in the current FAT code, which also stops the current disk operation, to avoid curdling the disk. 4) We are getting close to code and data segment limits in ELKS with the FAT system planned on being included normally; there's no reason to use more data space unless the user really needs it.

Summary: move CONFIG_32BIT_INODES to be a selectable option at all times, and default to OFF.

ghaerr, 2020-02-15:
Mentioned in #324, CONFIG_IMG_BOOT always needs to be set or any package manager build will fail (MINIX and FAT). 

I bring it up here to discuss whether this is a Config problem or not. Do we even need a CONFIG_IMG_BOOT option anymore? 

In any case, it ought to default to ON and also needs to be available default ON when CONFIG_FS_FAT is set. This issue ended up being @xrayer's entire problem in #324, which took a while to solve.

Another idea would be to have CONFIG_IMG_FAT and CONFIG_IMG_MINIX options to take its place. These could then be tested and possibly used to automatically invoke the package manager builds, which would be a nice integration of our disk image capability into .config.

mfld-fr, 2020-02-15:
Added CONFIG_IMG_BOOT option for FAT in commit b4efc61e410b9f7fffc6ffb5a2496b4a94b2d9d2.

mfld-fr, 2020-02-15:
Option CONFIG_32BIT_INODES now visible to user in commit 9350dd05674d2411c7c0d9aca2cadb7deb1ccc54.

mfld-fr, 2020-02-16:
@ghaerr : Do you need other changes from me in the configuration system ? Or could we close that issue ?

ghaerr, 2020-02-16:
Lets close it, thank you for your help on Config @mfld-fr !!

closed 2020-02-16 by ghaerr