ghaerr/elks #304: Help with SLIP link

alejandrocobo, 2020-02-06:
Hi,

I´ve been playing with ELKS in my 8088 laptop for the last week and now I´d like to try the TCP/IP stack. I´ve followed the intructions in `Documentation/textnetworking.txt`, which are pretty clear and simple but I must been messing something up.

First of all, the "gateway" is a modern laptop with Mageia 7 x86-64 (Linux 5.4.17). The physical conection between ELKS and Mageia is done by a USB-RS232 adapter plus a null-modem cable. I´ve sent messages thru /dev/ttyS0 from ELKS to the "gateway" and vice versa. It works.

When I follow the instruction, before doing the `ktcp ` thing in ELKS, I `cat /dev/ttyS0` to see if there's something happening there. When I ping my 8088 laptop from the gateway, I see some weird characters so it seem to works fine.

If I do this up side down (I do the `ktcp` thing in ELKS without doing the `slattach` thing in Linux 5.4), `cat ` the /dev/ttyUSB0 and try to do something like `geturl` or `telnet`, I see nothing at the "gateware" side.

When I follow full instrutions, I cannot ping ELKS. I have never done a bridge in Linux nor use SLIP. The gateway has a Wi-Fi link to Internet. I´ve tried using the same Wifi IP address for the gateway SLIP IP when I do the `ifconfig sl0` thing plus I´ve tried another IP in other network. None of them worked.

Leaving aside the TCP/IP configuration, as far as I understand all this, I should receive some bit streams if a try to do a `urlget `or `telnet `even if the IP stuffs are missconfigured, isn't it?

Thanks you very much.

mfld-fr, 2020-02-07:
Hello @alejandrocobo !

Yes, I think you are right, running an internet applet should trigger some traffic on the serial port, and we have to investigate that issue. It seems to me that SLIP was working fine just before we added the Ethernet support.

Unfortunately, we are busy these times with the CI, the image generation and the FAT support, plus some reported bugs on real HW, so I apologize if we won't process this issue in the next days.

Thank you for your patience !

alejandrocobo, 2020-02-07:
Finally I was able to manage the problem. I just disabled the eth interface at the init script and now I can ping ELKS. About the TCP/IP configuration, actually it doesn't matter if it is in the same subnet or even you pick the same Wifi IP for the gateway IP.

What I was unable to do is to set up a bridge between an ethernet interface and the SLIP link in the gateway. I've tried a LOT of stuffs and I am absolutely incompetent in this regard. I guess I'll leave that for the future.

Thank you anyway and sorry for the inconveniences.

mfld-fr, 2020-02-07:
No problem @alejandrocobo, nice to see you making it work on your own. About bridging with SLIP, I have also absolutely no experience, but I guess you can only make routing, as it operates at the IP L3 level, while your Ethernet bridge operates at ETH L2 level :smile:

mfld-fr, 2020-02-09:
Closing this one as the problem is now to setup IP forwarding on the gateway, not ELKS.

closed 2020-02-09 by mfld-fr