ghaerr/elks #306: [msdosfs] FAT filesystem bugs

ghaerr, 2020-02-06:
I'm opening this issue to detail and track the many current bugs in the ELKS MSDOS FAT filesystem. For those lurkers and testers out there, please report any additional issues found here, and I will work to fix them. Of course, I am always interested in additional comments, coding or ideas on where the problem might be!

Currently, I am *only* testing with the package-manager-built fd1440-fat.bin image, and mounting as /dev/fd1 in qemu, so as to not introduce possible sector/head configuration issues into the mix.

Current configuration of ELKS is for 1440k disk image, with CONFIG_MSDOS_FS, CONFIG_MSDOS_DEV, CONFIG_FULL_VFS set, and CONFIG_32BIT_INODES *not* set.
The FAT filesystem is mounted immediately after boot and root login with "mount -t msdos /dev/fd1 /mnt".

Following are problems I have found immediately after a fresh ELKS boot as above. The idea is to replicate a problem only immediately after a fresh boot and mount, rather than some time later.

o Initial "mount" sometimes but not always reports "Invalid argument". Not sure if this is a user land mount bug or a problem with the kernel syscall. I would like comments as to whether mount ever reports anything for non-fat filesystems.

o rm /mnt /usr/bin/ed, followed by umount reports "Device or resource busy". Unchanged with sync before umount. So there's a problem with the very basic filesystem after being written.

o ls -F /mnt crashes the system.

Thanks!

ghaerr, 2020-02-07:
@tkchia: I am currently cleaning up the MSDOS filesystem. There is lots of code with register declarations (I suspect coded 20 years ago). With your understanding of gcc-ia16, what does the compiler do with 'register', and how should things be coded? That is, is register always ignored, taken as a suggestion, or forced? Wouldn't it be better, given that we're now using the advanced gcc-ia16 compiler, to completely remove register declarations and let the compiler decide on best code generation?

tkchia, 2020-02-07:
Hello @ghaerr,

> With your understanding of gcc-ia16, what does the compiler do with 'register', and how should things be coded? That is, is register always ignored, taken as a suggestion, or forced?

My understanding is that GCC will simply ignore the `register` keyword --- except in a very few corner cases, and these cases do not really apply to the ELKS code.  (The GCC info file [goes into a bit more detail](https://gcc.gnu.org/onlinedocs/gcc-6.3.0/gcc/Hints-implementation.html).)

It should be OK to remove the `register` specifiers, though there is likely no harm either if we leave them there.

Thank you!

ghaerr, 2020-02-07:
Fixed "rm /mnt/usr/bin/ed, rm /mnt/usr/bin/* bugs". 

MSDOS FAT12 filesystem is feeling fairly stable, only seeing bugs on umount after rare (currently not repeatable) sequences. Not seeing any more soft hangs.

Any MSDOS users please try it out, I am anxious to see whether we can close out the first major bug listed in #123. There are still major bugs in "ls -F" (crashes system), and "ls -i" (incorrect results).

mfld-fr, 2020-02-07:
Good job and big progress ! For better understanding, could you please use rather FAT keyword in place of MSDOS in the code for next steps ? First is the operating system, second is the file system.

ghaerr, 2020-02-07:
> Good job and big progress ! For better understanding, could you please use rather FAT keyword in place of MSDOS in the code for next steps ? First is the file system, second is the operating system.

Sure - but where exactly in the source code should this be changed? Or are you talking about commit messages? The source code is filled with "MSDOS" and we're also using CONFIG_MSDOS_* for our FAT configuration stuff... Let me know what you're thinking. I have a couple more commits coming up soon.

ghaerr, 2020-02-07:
@mfld-fr and others: 

I can confirm that the longstanding ELKS freeze or possibly "buffer hang" when using the FAT filesystem is NOT fixed. While good progress has been made fixing FAT problems when modifying directory entries such as rm and rmdir, I had noticed an occasional hang, and particularly lately it has been coming up in increasingly great frequency. I have been testing "cat > /mnt/file" immediately after ELKS boot and its automatic FAT mount. For a small while in yesterday's development, ELKS would always hang on executing cat. When adding a single printk or other statements for debugging, FAT would then work without problems. I couldn't reproduce it anymore so I submitted the PR. Sounds like data is written into the kernel code or data segment, or there is a buffer overrun, in unknown circumstances.

Does anyone have any ideas for how to approach debugging this nasty problem? I'm thinking of wrapping all low level buffer writes with printk but almost any changes cause the problem to disappear for a bit.

Because of the nature of this problem, I am going to prepare another PR with tunable DEBUG printks, possibly defaulted to ON, so that FAT testers can send screenshots of the problem once crashed. I feel the need to do this because on pulling upstream master after everyone's additional today, the crash occurred twice without debug statements. Arrggghh!

Just after my last commit and other changes to the source tree were pulled from upstream master, I managed to get a panic "Unable to get timer" (see attached). Adding printk and problem disappears. Re-running the emulator without any changes and no problem seen. Possibly writing to a null pointer location in low memory where the timer vector is? I haven't researched the timer panic yet. Can the program counter location be known from the crash call stack?

Thoughts on efficient debugging techniques for this?

[image]

ghaerr, 2020-02-08:
Confirmed: "ls -F /mnt" *or* "ls -F /" crashes ELKS, so not a FAT bug. Unclear what could be the problem, as ls.c code shows -F option is very basic. Could be sizeof(struct stat) isn't passed back properly and overwriting buffer? Except "ls -l /" works.

Only remaining FAT bug I'm seeing is the semi-regular hang problem on "cat > /mnt/file", which has never happened with debug printk's compiled in.

@mfld-fr: Do we want to start keeping track of userland command bugs somewhere? 

"ls -i" is also broken for any filesystem.

ghaerr, 2020-02-09:
Figured out one reason why ELKS MSDOS filesystem has been slow: buggy code trying to update the backup FAT table (disks almost always have two copies) was doing a sector-sized memcpy every time a file required an updated FAT entry. An additional 50% of the time it did another sector memcpy (so 1024 bytes) copied. Second bug is that it was updating the first, rather than second FAT table anyways, thus the backup FAT table never got updated with ELKS, ever. 

Commented it all out for now to increase speed. I propose not doing any backup FAT updates during normal file writing, but only after say, file close or filesystem sync, for speed. Since it's never worked, commenting it out will be OK during our test period.

Will submit a PR after existing PR #315 is accepted, I've screwed up my local git repo again and don't want to commit over that PR.

ghaerr, 2020-02-09:
ELKS MSDOS FAT "buffer" hang bug FOUND AND CRUSHED!!!

I've finally found and fixed the intermittent system hang problem that our users have been experiencing for the last two years when using mounted MSDOS FAT disks. I wrote a shell script that continuously creates and removes files and directories and got the system to hang with my full filesystem debugging on.

The bug occurred whenever a new FAT directory entry needed to be created: an inode directory read was requested with iget(), which is supposed to be passed a pointer to the mounted super block for the requested inode. Instead, the code had ALWAYS passed a device kdev_t value. The compiler has warned about the pointer to integer conversion for years, but no one had noticed (I'm cleaning up compiler output also, and didn't notice until I found the bug and inspected more closely). Well, iget checks for a NULL pointer, but otherwise goes in its merry way, comparing in a for() loop for the requested inode's device to be equal to the mounted superblock's device, at an offset off the passed superblock pointer. So, whenever the value in memory at the s_dev offset of the bogus kdev_t "pointer" happened to match, it would return a bogus inode, and crash, otherwise return 0 for not found. The calling code happened to keep looking for another spare directory entry whenever 0 was returned, so the "intermittent" problem was really one of matching device values, that is, 1 out of 64k chance of matching.

I've been testing with both 32 and 16 bit inodes, and no crashes now while running the shell script. Sometimes, ^C doesn't stop the script anymore, don't know the reason for that. The ELKS FAT filesystem code now has no FAT12 bugs that I know of, and is ready for serious user testing. I'm interested in any shell scripts that anyone might have that work over a filesystem, so that I can build some automated test suites.

There's still more cleanup to do as mentioned in my previous post that should be able to reduce the code size, and I will need to build additional FAT16 and FAT32 disk images with the package manager to start testing that portion of the filesystem, along with some test scripts to be included on the images.

mfld-fr, 2020-02-09:
@ghaerr : Congratulations ! This was a long stand bug that poisonned the FAT support... one step toward the next release 😄 !

tkchia, 2020-02-09:
@ghaerr, this is very cool. :-)  Thank you!

mfld-fr, 2020-02-09:
@ghaerr : please feel free to close this issue as soon as you think FAT is all OK from your testing.

georgp24, 2020-02-09:
I do not know if this is the right thread for this. I wrote fd1440.bin to a real floppy and booted my PC from that. I can log in and use ELKS. However, both hard disks installed are NTFS so I cannot mount a FAT disk. I put in a FAT flash disk but that was not detected.
[image]
[image]

ghaerr, 2020-02-10:
@mfld-fr: I'm going to close this issue, as I don't know of any FAT filesystem bugs at this time, although there could be some. I have also read through #123 while debugging and feel that all those issues have also been handled, or certainly are out of date at this point. I recommend that we open a new issue for any FAT problems during testing, should issues be found. I suggest closing #123 as well.

Also, can you move @georgp24's issue above to a new 'Floppy Probing' issue? We probably need a place for developers and users to comment on that, as it is still ongoing. @georgp24's issue has nothing to do with FAT filesystem bugs, but his issue, including probing of ELKS bootable MINIX, ELKS bootable FAT, and MSDOS bootable FAT, all need some discussion and design direction.

Finally, we may want to think about changing the /etc/motd version number to 0.3.1pre or something like that so that we can be sure as to what they're running.

closed 2020-02-10 by ghaerr

mfld-fr, 2020-02-10:
Moved @georgp24's problem report to #320.