ghaerr/elks #307: New release 0.3.1

mfld-fr, 2020-02-07:
Release criteria:
- All reported boot problems on real HW solved (see mailing list)
- FAT support proven to be dramatically improved (TBC by @Mellvik on real HW)
- Option : Boot from FAT

Draft of the release description:

Main changes (thanks @ghaerr) :
- Build under OSX
- Nano-X restored
- Build as normal user
- FAT images support
- ELKSEMU take over and fixing (thanks @tkchia)

Minor changes:
- GCC-IA16 usage (thanks @tkchia)
- LIBC tuning (thanks @tkchia)
- Safer wait event function
- Kernel code cleanup
- Ram disk restored

Bug fixes:
- Configuration & build scripts (thanks @tkchia)
- Kernel FAT filesystem (thanks @fonin and @ghaerr)

Mellvik, 2020-02-07:
Great work everyone,
I'll do some physical hardware testing (80286/12 and 80386/20) over the weekend. Which configuration(s) should I be focusing on?

Mellvik

mfld-fr, 2020-02-07:
It would be nice if we could kill issues #123 and #306.

So I would say booting from a MINIX floppy, mounting the FAT HD and try to reproduce 🤔 ?

ghaerr, 2020-02-07:
@Mellvik:

> So I would say booting from a MINIX floppy, mounting the FAT HD and try to reproduce 🤔 ?

Agreed.  I will temporarily modify /etc/rc.d/rc.sysinit to try mounting FAT from /dev/fd1 rather than /dev/bda1 which will greatly speed up your testing by not having to mount -t msdos /dev/fd1 /mnt after every boot. I am very interested in bug reports that can be confirmed with commands immediately after boot, although I realize that some #123 issues are stubborn and require more time and not easily reproducible.

I have only been testing on on the 1440k floppy (fd1440-fat.bin) on QEMU, so it would be nice to also test 360k and 720k images if possible (although no big deal, since they're all FAT12 at this point). The FAT16 and FAT32 testing require larger disk images than we have on older equipment, so that will likely be tested on an emulator, right?

xrayer, 2020-02-10:
Wow, I can see "Boot from FAT", well what bootsector code did you implemented and how problem with /dev was solved? I will update the sources, recomplile and try out on my 286  :)

mfld-fr, 2020-02-10:
@xrayer : thank you to be so enthusiast ! But this issue is to draft the next release. Please wait a bit more to get that next release published 😄 ...

For the explanations about /dev handling and boot sector, please see the ongoing discussion in #296.

xrayer, 2020-02-11:
Hi, I tried to compile the new version but now I don't have correct neither MINIX nor FAT bootbable image. I select 1,44MB HD as always and tersted it in BOCHS.
MINIX image says at boot:

Booting from Floppy...
Boot sector
....3!
Press key

and repeat when key pressed.
FAT image doesn't warn it's not bootable but just hang. I can see that FAT bootsector is almost empty - the code, only BPB seems to be there "FAT 12" string visible... MINIX bootsectors seems contain its code. I reused directory cross with build tools from my previous ELKS version to save time to download and build everything.

mfld-fr, 2020-02-26:
Not managing the next release anymore. Closing this one.

closed 2020-02-26 by mfld-fr