ghaerr/elks #313: Reading from /dev/fd1 yields contents of disk in A: (/dev/fd0)?

tkchia, 2020-02-08:
I ran ELKS under QEMU, with a 1.44 MiB disk image for ELKS in drive A:, and a 720 KiB FAT12 disk image in drive B:.

Dumping the contents of `/dev/fd0` and `/dev/fd1` in ELKS yields identical results for some reason, even though ELKS correctly figures out that their drive geometries and filesystems are different:

[image]

Thank you!

tkchia, 2020-02-08:
This bug seems to only affect attempts to read directly from the `/dev/fd1` block device file.  Mounting `/dev/fd1` at a mount point and reading files from the mounted volume seem to work normally.

The bug also seems to also crop up when I `tail -1 /dev/bda`.

Thank you!

mfld-fr, 2020-02-08:
Mmm... looks like a bug around the CONFIG_BLK_DEV_CHAR option and the code to present a block device as a flat file...

tkchia, 2020-02-10:
I found the bug (`inode->i_dev` was used instead of `inode->i_rdev`), and `head -1 /dev/fd1` works correctly now:

[image]

Thank you!

closed 2020-02-10 by tkchia

mfld-fr, 2020-02-10:
Thank you @tkchia for the fix !