ghaerr/elks #323: Build broken by latest FAT boot sector

mfld-fr, 2020-02-11:
Some build failure while maturing the latest changes:
* when CONFIG_IMG_HD selected from default configuration
* using 'config-emu86' as tested configuration

Also what I think a design defect: duplicated FAT BPB in boot sector code and as generated by 'mformat'.

ghaerr, 2020-02-11:
> Some build failures while maturing the latest changes:

Are you talking here about just @tkchia's fat.bin boot sector here or the FAT boot modifications?

\\ 
> Also what I think a design defect: duplicated FAT BPB in boot sector code and as generated by 'mformat'.

Can you describe in more detail?

I have noticed that `mdir` won't work on disks originally properly prepared by `mformat` but then updated with unmodified 'fat.bin' boot file (by setboot). I think the reason for this is that the FAT BPB needs to be an extended format, but I haven't dug into it. Another reason could be bad media byte.

tkchia, 2020-02-11:
Hello @mfld-fr, hello @ghaerr,

Hmm... well, my new `elkscmd/bootblocks/boot_sect.S` barfed if ELKS was configured to run from ROM, because there is no "drive geometry" whatsoever.  Also `elkscmd/bootblocks/boot_sect_fat.h` also barfed if ELKS was configured to create a hard disk partition image, since it did not yet hard disk geometries.

I will try to submit a pull request to at least fix these problems.  (And, I hope to find out more about this 1.68 MiB floppy disk format...)

Thank you!

tkchia, 2020-02-11:
Hello @ghaerr,

> I have noticed that `mdir` won't work on disks originally properly prepared by mformat but then updated with unmodified 'fat.bin` boot file (by setboot).

What was the error message?  Thank you!

ghaerr, 2020-02-11:
@mfld-fr and @tkchia:

In thinking a bit more about FAT support for mounting MSDOS HD images, the current FAT filesystem code should work fine when *mounting* large hard drives, as the boot block FAT BPB hidden sector count correctly skips possible multiple partitions and references the desired partition boot sector. I will have to review the technical aspect of MBR records, I do not know whether they are compatible with FAT BPB's or whether entirely new mounting and boot code is/was required to read these MSDOS HD disks.

I am not informed about our /dev/bda1 driver - does it know about MSDOS HD filesystem partitions using an MBR (multiple boot record)?

tkchia, 2020-02-11:
Hello @ghaerr,

In any case, I will meanwhile try to get the boot sector code not to fail the build even for ROMFS or hard disk configurations.

Currently both the Minix and FAT boot blocks are built, no matter what the chosen target configuration is.  I am not sure, but perhaps that can change?

> I do not know whether they are compatible with FAT BPB's or whether entirely new mounting and boot code is/was required to read these MSDOS HD disks.

Master boot records do not have BPBs --- only the individual partitions have them.  An MBR simply have their boot code at its front, plus a partition table at the end.

Thank you!

Mellvik, 2020-02-11:
Also be aware that the failure to mount the FAT HD partition is not a device driver issue but a file system level issue. We can read (and write presumably) the partition just fine.

The mount worked just fine yesterday.

—Mellvik

ghaerr, 2020-02-11:
@tkchia:

> Currently both the Minix and FAT boot blocks are built, no matter what the chosen target configuration is. I am not sure, but perhaps that can change?

No, we should always build both, so that the package manager has a fat.bin to install,
regardless of the .config configuration.

> > I do not know whether they are compatible with FAT BPB's or whether entirely new mounting and boot code is/was required to read these MSDOS HD disks.
> 
> Master boot records do not have BPBs --- only the individual partitions have them. An MBR simply has its boot code at its front, plus a partition table at the end.

Thank you for your explanation of MBR's, I was getting myself confused!

Thus, neither our /dev/bda HD disk driver, nor our FAT filesystem need be concerned with MBRs, as we just don't support them. For @Mellvik, I assume his hard drive he's testing is just a FAT volume, with the standard FAT12/16/32 FAT BPB. He just needs to ensure that the .config settings are correct for his HD disk, then build using CONFIG_IMG_HD, so that ELKS boot normally from floppy then uses the compiled-in sector/head/track values from .config to read the hard drive. Correct?

Perhaps the whole issue is that CONFIG_IMG_HD was broken with your latest boot block PR, and the one you've just prepared (I just reviewed it) fixes the problem.

Mellvik, 2020-02-11:
@ghaerr,

> Thus, neither our /dev/bda HD disk driver, nor our FAT filesystem need be concerned with MBRs, as we just don't support them. For @Mellvik, I assume his hard drive he's testing is just a FAT volume, [...]

Not entirely. The hard drive on this particular system has 3 Venix partitions, one FAT16 partition. And yes, ELKS boots normally from (MINIX fs) 1.44M floppy, but ELKS (and other OSes) never use compiled in CHS values for the HD, they read it off of the HD itself, thats what the MBR does - provide the partition table and code to load the first sector of the active partition and JMP to that.

Mellvik

closed 2020-02-12 by mfld-fr