ghaerr/elks #324: Cannot boot in BOCHS emulator

mfld-fr, 2020-02-11:
@xrayer wrote:

Hi, I tried to compile the new version but now I don't have correct neither MINIX nor FAT bootbable image. I select 1,44MB HD as always and terted it in BOCHS.
MINIX image says at boot:

**Booting from Floppy...
Boot sector
....3!
Press key**

and repeat when key pressed.
FAT image doesn't warn it's not bootable but just hang. I can see that FAT bootsector is almost empty - the code, only BPB seems to be there "FAT 12" string visible... MINIX bootsectors seems contain its code. I reused directory cross with build tools from my previous ELKS version to save time to download and build everything.

BTW please add some version check of mtools in the Makefile. My old debiad had mtools-4.0.18 and build failed because it didn't know the -R param. It took me some time until I find error message in lot of printings. Then I had to build mtools-4.0.23 from sources as no deb package for my Jessie and then images was built OK.

ghaerr, 2020-02-11:
> Hi, I tried to compile the new version but now I don't have correct neither MINIX nor FAT bootbable image. I select 1,44MB HD as always and terted it in BOCHS.
> MINIX image says at boot:
> 
> **Booting from Floppy... Boot sector ....3! Press key**

@mfld-fr: If I'm not mistaken, @tkchia's latest boot sector rewrote the 'Boot sector' message above to read 'ELKS boot'. This means that this is not the latest FAT boot code being tested.

mfld-fr, 2020-02-11:
@ghaerr : I am not the problem reporter...

xrayer, 2020-02-12:
I have found that the problem was caused by linux image file with zero length placed in image,
I updated sources yesterday (download zip package from github) and I succeed to reproduce build a bootable FAT image by manual way (tested in BOCHS only yet). But there are some problems
1) fd1440-fat.bin was not produced by standard make all
2) I need to go to image and run make images but the FAT bootsector code is empty I had to manually run setboot fd1440-fat.bin ..../fat.bin
3) linux image file in floppy image had zero length. I had to mount the image and rewrite linux file by good one. Then it booted

BTW where the nanox demo has gone? I have checked nanox in kernel config|userland.

mfld-fr, 2020-02-12:
@xrayer : Thank you for your test report !

* '0.3.1' is not released yet : `master` branch requires maturing before we could consider it as a release candidate. Release criteria are defined in #307, including a 'bug free' FAT support.

* `fd1440-fat.bin` is not produced by the standard build. Only from `make images` for now. Point captured in #327.

For other points (empty `linux` and empty boot sector when running `make images`), I let @ghaerr to process them.

ghaerr, 2020-02-12:
@xrayer : Thank you for your testing and verifying that ELKS FAT boot is in fact working on your system! I am interested in more reports as to how ELKS runs on the FAT filesystem.

> I have found that the problem was caused by linux image file with zero length placed in image,
> 2. I need to go to /image and run make images but the FAT bootsector code is empty I had to manually run setboot fd1440-fat.bin ..../fat.bin
> 3. linux image file in floppy image had zero length. I had to mount the image and rewrite linux file by good one. Then it booted

The current "make image" build is broken for unknown reasons as referenced in #326. The problem you're seeing with both /linux being empty and needing to run setboot manually is that the make is failing in the `mcopy` portion when writing the files, so /linux isn't written, and setboot is supposed to run after that. The entire problem is in the Make.packages file at this point.

> BTW2 elks still reports version 0.3.0 instead 0.3.1

That will be changed shortly.

> I updated sources yesterday (download zip package from github) and I succeed to reproduce build a bootable FAT image by manual way (tested in BOCHS only yet). But there are some problems

Is it correct that this issue (Can't boot in BOCHS) is now working, so it can be closed? 

Please report any additional bugs in a separate thread, and thank you!

mfld-fr, 2020-02-12:
Actions before closing this one:
- [ ] add options in configuration menu to select the NanoX demos
- [ ] fix the `make images` issue
- [ ] wait for @xrayer to validate

mfld-fr, 2020-02-12:
After fixing one mistake from me in commit 0fa94757, I ran again a full cleanup with `clean.sh`, then a full build with `build.sh`, added FAT FS in the kernel configuration at the `menuconfig` step, and built images with `nmake images`. It then edited `qemu.sh` to boot on `fd1440-fat.bin` and EKLS came to the login prompt.

ghaerr, 2020-02-12:
> After fixing one mistake from me in commit 0fa9475, I ran again a full cleanup with clean.sh, then a full build with build.sh, added FAT FS in the kernel configuration at the menuconfig step, and built images with nmake images. It then edited qemu.sh to boot on fd1440-fat.bin and EKLS came to the login prompt.

@mfld-fr: Nice!!

With regards to my open PR on make images failing, I now think it’s due to a temperamental design that fails without an appropriate error message when a file is missing. 

I will fix that with another commit to that PR shortly.

xrayer, 2020-02-13:
I downloaded new sources tonight, do make menuconfig (sel. FAT FS + FAT image), make, cd image, make images. Now the bootsector is correctly filled by BPB and code no more need to call setboot but the linux image file is still zero length in FAT image and missing in minix image. After manual mount-copy-umount it boots fine. I understand that some Make.package tuning may be needed...

mfld-fr, 2020-02-13:
Oh 🤔... I am wondering if ever your `mcopy` version fails when overwriting the `linux` placeholder in `Make.package`...

ghaerr, 2020-02-13:
> Oh 🤔... I am wondering if ever your `mcopy` version fails when overwriting the `linux` placeholder in `Make.package`...

No, never - the whole reason that /linux wasn't being written was due to the package manager 'File not found' problem that cropped up as a result of your commit changing the packaging of /bin/ndemo to /bin/nclock, which exposed a weakness in the package manager file assembling code, which I fixed in #326. In that PR, I added better error handling by displaying the missing filename and exiting the build immediately.

mfld-fr, 2020-02-13:
So how do you explain it succeed on my side (https://github.com/elks-org/elks/issues/324#issuecomment-585364785) and failed on @xrayer's side (https://github.com/elks-org/elks/issues/324#issuecomment-585534511) just after fixing my mistake ?..

ghaerr, 2020-02-13:
Not sure - there have been a few out-of-sync steps with the dev tree and user testing over the past couple hectic days, but it's all gotten sorted out now. There is absolutely nothing special about the /linux copy - it is not handled specially at all, it's just another tagged file in the Packages list. The only special handling for /linux is to create its directory entry directly after mformat, something that has never been identified as an issue.

xrayer, 2020-02-14:
After today update I'm unable to build any working image because file target/linux was not created. But there exist elks/arch/i86/boot/Image so I don't know why it was not copied. I have mtools (GNU mtools) 4.0.23
configured with the following options: enable-xdf disable-vold disable-new-vold disable-debug enable-raw-term 
compiled from sources with default config.

root@MR:/elks-2020/image# make images
make -f Make.package minix NAME=fd360-minix OPTS="-i192 -s360" BPB=-B9,2 APPS=":boot|:small"
make[1]: Vstupuje se do adresáře ,,/elks-2020/image"
awk "/:boot|:small/{print}" Packages | cut -f 1 > Filelist
mfs -v /elks-2020/image/fd360-minix.bin mkfs -1 -n14 -i192 -s360
image 1K blocks: boot 1, super 1, inodemap 1, filemap 1, inode 6, data 350
inodes inuse 1, free 190, overhead 1, total 192
blocks inuse 1, free 349, overhead 10, total 360
mfs -v /elks-2020/image/fd360-minix.bin addfs Filelist /elks-2020/target
Adding files from /elks-2020/target
File not found: /elks-2020/target/linux
Make.package:30: návod pro cíl ,,minix" selhal
make[1]: *** [minix] Chyba 255
make[1]: Opouští se adresář ,,/elks-2020/image"
Makefile:94: návod pro cíl ,,fd360-minix" selhal
make: *** [fd360-minix] Chyba 2

ghaerr, 2020-02-14:
Did you do a regular “make” beforehand? That’s the one that copies elks/arch/i86/boot/linux to target/linux. Please send that make output, thanks!

Greg

ghaerr, 2020-02-14:
> So how do you explain it succeed on my side ([#324 (comment)](https://github.com/elks-org/elks/issues/324#issuecomment-585364785)) and failed on @xrayer's side just after fixing my mistake ?..

One explanation for this would be if CONFIG_IMG_BOOT is not set in .config - that would cause all package manager builds to fail. Should I add a check for this?

@xrayer, can you check that this is set in your .config?

ghaerr, 2020-02-14:
> After today update I'm unable to build any working image because file target/linux was not created. But there exist elks/arch/i86/boot/Image so I don't know why it was not copied.

If CONFIG_IMG_BOOT is not set in .config, then target/linux will not be created. Check that
target/linux is created after doing a "make clean; make"

> I have mtools (GNU mtools) 4.0.23
> configured with the following options: enable-xdf disable-vold disable-new-vold disable-debug enable-raw-term

That should be ok.

> compiled from sources with default config.

Strange, CONFIG_IMG_BOOT should be set in default config.

Perhaps you should send your .config file over so I can inspect it. Make sure you run a make clean beforehand also, and start over with a fresh pull, "make clean; make; make images".
> root@MR:/elks-2020/image# make images
> make -f Make.package minix NAME=fd360-minix OPTS="-i192 -s360" BPB=-B9,2 APPS=":boot|:small"
> make[1]: Vstupuje se do adresáře ,,/elks-2020/image"
> awk [...]

In my last commit, I added an error check to stop the build if a file is specified to be included in the package build is not found. That is what is happening here, in your MINIX image build, target/linux is not found.

> Make.package:30: návod pro cíl ,,minix" selhal
> make[1]: *** [minix] Chyba 255
> make[1]: Opouští se adresář ,,/elks-2020/image"
> Makefile:94: návod pro cíl ,,fd360-minix" selhal
> make: *** [fd360-minix] Chyba 2
> 
> If I copy linux image manually to target then it fails bulding 720k image:
> ...
> No free slots in bitmap found
> Make.package:30: nÃ¡vod pro cÃ­l ,,minix" selhal

The packages are created and fairly tight on space... perhaps there is something else in .config that is specified that is causing to run out of space in package build.

You can edit image/Packages to add or subtract files to make them fit a specific image size.

> but when forced make to "make fd1440-fat" it is created ok and it's bootable.

It sounds like there are multiple configuration issues that are affecting the build here.
I can see that there is still some work to do regarding what happens if changes are made to .config, including boot options and applications on/off, the make images build could fail if the Packages file is not also edited.

xrayer, 2020-02-15:
No, CONFIG_IMG_BOOT is not present (neither commented) in config file. Probably because I selected FAT image in menuconfig and there's no boot option - should be now...
When I select Minix + boot, then CONFIG_IMG_BOOT=y is added at the end of config, when switch back to FAT the line is removed. With manually added it made linux file in target directory and with make fd1440-fat produced bootable image.
So it have to be fixed the menuconfig and filelist to fit in lower density images.
I will try to boot it on 286 tomorrow...

xrayer, 2020-02-16:
Today build finally works smooth as expected, FAT bootable image was created and tested also on 286. Good work :) Can be closed now.

mfld-fr, 2020-02-16:
Thank you @xrayer ! Appreciated your serious testing.

closed 2020-02-16 by mfld-fr