ghaerr/elks #327: Roadmap for FAT tuning

mfld-fr, 2020-02-12:
Things are getting messy with 3 active contributors and still some confusion or disagreement on the final FAT or boot design. It is so worth to agree on a roadmap to conclude and keep everybody happy 😃

- move FAT payload out of boot sector to relax space constraint
- always include ELKS BPB in boot sector
- create a new 'boot' file with the existing FAT payload
- drop the FAT BPB duplication in payload
- extract FAT location of ' boot' with 'mtools'
- add generic payload location in ELKS BPB
- option : convert FAT payload from ASM to C
- improve 'boot' to handle 'linux' fragmentation
- tune 'setboot' to preserve existing extended FAT BPB
- tune 'setboot' to update ELKS BPB (geometry and payload)
- add FAT image to image/Makefile
- tune the CI to rename fd1440.bin to fd1440-minix.bin
- tune the CI to generate fd1440-fat.bin

tkchia, 2020-02-12:
Hello @mfld-fr, hello @ghaerr,

Thank you.  Am I correct that this is a road map for the longer term?

From what I see, this is the current situation:
  * For Minix, we have a 512-byte boot sector, which loads a 512-byte Minix-specific loader, which loads `/linux` and splits it into its "setup" and "kernel" portions.
  * Likewise for FAT, except that the 512-byte boot sector tries to directly load and split `/linux` itself --- with some restrictions on where it can be on disk.  But perhaps we might want or need to extend the bootloader to two sectors.
  * `/linux` contains, in its first 512 bytes, a largely unused "bootloader" which we only use to determine how to split the rest of `/linux` into "setup" and "kernel" parts.  (The "bootloader" seems to be largely a holdover from [The Boot Protocol](https://www.kernel.org/doc/html/latest/x86/boot.html) used for Linux for x86-32/-64.  It _can_ _potentially_ be used to make the kernel itself a bootable image, but it seems modern Linux/x86-64 kernels no longer even allow this, and I am not sure how essential this is.)

I am thinking that, inside all of this, there should be opportunities to reduce the complexity of the whole boot setup.

Thank you!

ghaerr, 2020-02-12:
@mfld-fr : Good idea for this roadmap, I assume you'll keep it up by editing it for all to see as issues are debated and code written? It is time for more discussion before more code is needed.

I assume this Roadmap is for 0.3.2 in branch fat_tune, with master for 0.3.1 with no more boot sector changes?

Shall we comment/question/clarify on various roadmap entries in this thread?

@tkchia: what is the free space left on the current FAT boot sector (with EPB in)?

mfld-fr, 2020-02-12:
Yes, you are both right, this is a roadmap for the v0.3.2 release, so after the upcoming v0.3.1.

About the 'raw' boot sector that is the first of the 'linux' image :
1. it is still used today to boot from a 'raw' image without any file system (see #299)
2. after loading, it is used as the `INITSEG` to store some kernel initial parameters

I already started to empty that storage with a cleanup of the unused parameters (see the recent history of `setup.S`). As soon as `INITSEG` is empty, we can remove that first sector and use the 'generic' one instead.

ghaerr, 2020-02-12:
I have a design and can write an update for `setboot` that addresses most of the setboot issues in the roadmap, as well as fixes the bug in 0.3.1 that the fd360-fat.bin and fd720-fat.bin images are broken and won't boot.

Would you like to discuss it, or should I just submit a PR for the master branch to fix 0.3.1?

Changes to `setboot` resulting from roadmap decisions to boot blocks or only for 0.3.2 features would be easily made at the time boot blocks change in 0.3.2, of course.

mfld-fr, 2020-02-12:
PR is a good place to review and discuss some proposed changes, this is how GitHub designed the workflow. 😄

tkchia, 2020-02-12:
Hello @ghaerr,

> @tkchia: what is the free space left on the current FAT boot sector (with EPB in)?

That will be about ... 20 bytes.
```
000001c0  83 89 ec 5d c2 02 00 45  4c 4b 53 20 62 6f 6f 74  |...]...ELKS boot|
000001d0  0d 0a 00 21 0d 0a 00 50  72 65 73 73 20 6b 65 79  |...!...Press key|
000001e0  0d 0a 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
                   ^^ ^^ ^^ ^^ ^^  ^^ ^^ ^^ ^^ ^^ ^^ ^^ ^^
000001f0  00 00 00 00 00 00 00 00  04 3f 10 04 65 4c 55 aa  |.........?..eLU.|
          ^^ ^^ ^^ ^^ ^^ ^^ ^^
```

mfld-fr, 2020-03-09:
We are obviously back to a rolling version, not worth to spend any more time in planning and road mapping activities. So closing this one.

closed 2020-03-09 by mfld-fr